pbatard/rufus #1224: Dead links on rufus.akeo.ie

djoris, 2018-10-09:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [N/A] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [N/A] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
There are a few hyperlinks on the main page for RUFUS (rufus.akeo.ie) that refer to non-existing pages/files. Links checked with Xenu's Link Sleuth, verified by hand.

* USB icon by <a href="http://pcunleashed.com/">PC Unleashed</a><br/>
* http://french.about.com/od/vocabulary/g/mecenat.htm (about.com is no more, uncertain if https://fr.wikipedia.org/wiki/M%C3%A9c%C3%A9nat would cover the topic)
* https://msdn.microsoft.com/en-us/subscriptions/downloads/default.aspx for all Windows versions but Win10 (downloads unavailable since 2015, see: https://www.pcsteps.com/45-download-windows-7-iso-legally-free-digital-river/)
* In the CSS, references to two non-existing images: 
```
.treeView li.collapsibleListOpen{ list-style-image:url('button-open.png'); cursor:pointer; }
.treeView li.collapsibleListClosed{ list-style-image:url('button-closed.png'); cursor:pointer; }
```

ibuprophen1, 2018-10-09:
According to the following links, pcunleashed.com is no longer available and even the domain url is available for purchase. 

https://www.godaddy.com/whois/results.aspx?domain=pcunleashed.com 

https://www.godaddy.com/domains/searchresults.aspx?checkAvail=1&tmskey=dpp_dbs&domainToCheck=Pcunleashed.com

https://dnslytics.com/domain/pcunleashed.com

http://www.isitdownrightnow.com/pcunleashed.com.html

I know that when credits like this are provided on a website, those who are in charge of the website itself (regarding Rufus) doesn't really prioritize this type of change nor doesn't realize it either unless someone (like yourself) brings it to their attention. 

They may just validate your report and typically change/remove the actual link to just the company name itself. 

I don't own nor represent Rufus but, I just want to state that this was a great thing you did! :-) 

~Ibuprophen

ibuprophen1, 2018-10-09:
As you had stated... The following link is only available for Windows 10 downloads. 

https://www.microsoft.com/en-us/software-download/windows10ISO

~Ibuprophen

pbatard, 2018-10-10:
@djoris, many thanks for going through the dead links.

I removed the dead link for PC Unleashed.

I am now using wiktionary rather than French Wikipedia to illustrate what "mécénat" means. With the link being aimed at English speakers, even if the word being used is deliberately a French one instead of its English best approximation, I don't think l want to link to French content, as it will just confuse people.

On the other hand, I am not planning to remove the MSDN links for Windows, as they are the only legal way that I know to obtain these images. Also I am not planning to alter the Windows 10 link (the only non MSDN one), as it does allow __direct__ download of Windows 10 retail ISO __from Microsoft themselves__ as long as you set your browser's User Agent to a non Windows platform. The thing is, depending on the User Agent, Microsoft will either offer to download the retail ISOs, or (if it detects you are running a recent version of Windows), the media creation tool. But if you know about that, you can very much download the ISOs, without having to rely on a non-official third party tool. I definitely want to avoid linking to third party tools on the homepage.

Finally, I removed the unused CSS, which was remnant from a time where I played with collapsible lists.

I can only thank you again for doing the grunt work of validating web content. This is very helpful indeed!

closed 2018-10-10 by pbatard

ibuprophen1, 2018-10-10:
As I had previously commented... 

Not necessarily @pbatard...

The following are just 2 examples of guidance provided by Microsoft representatives reflecting that the downloads provided by HeiDoc are legal.

https://answers.microsoft.com/en-us/windows/forum/all/downloading-windows-isos-legally/9b5b2ca7-8c0a-4ec1-9932-ae2555e41749

https://answers.microsoft.com/en-us/windows/forum/all/using-media-creation-tool-i-cannot-download/bd0d8ef1-c222-4751-ac4c-d96e1028efbe

As a former certified Microsoft OEM System Builder as well as having multiple other certifications with Microsoft I do try to keep up to date with various areas of the certifications.

Microsoft's stance is that only 2 outside websites can legally provide their Software ISO which are "Digital River" and "HeiDoc".

I'm just surprised that this isn't as well known as it should be...

~Ibuprophen

ibuprophen1, 2018-10-11:
My apologies... It won't happen again. :-) 

~Ibuprophen