pbatard/rufus #1225: Version 3.3: write error when writing ISO to USB device (works in version 3.2)

engelhro, 2018-10-12:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I can't write an ISO file to the USB device due to an "Access denied" error (see logfile below).

This problem does **not** occur Rufus version 3.2 with the exact same setup (same PC, same USB device, same port, even the same ISO file).

I don't understand the warning at the end of the logfile ("Controlled folder access"), but as everything still works fine with Rufus 3.2, I *suspect* this message is not related and my problem actually *seems* to come from the update to version 3.3.

However, none of the two entries in the changelog for version 3.3 seems related and indicates any change applicable to my case (simple ISO instead of DD image, and no Windows to Go drive creation).

Thanks for help!

Log
---
```
Rufus version: 3.3.1400 (Portable)
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Using autorun.inf label for drive H: 'System Rescue CD 5.3.1'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0008524F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.0 GB (16105078784 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Using autorun.inf label for drive H: 'System Rescue CD 5.3.1'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0008524F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.0 GB (16105078784 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.06/2012-10-23 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'sysrcd-5.3.1'
  Size: 546.9 MB (Projected)
  Uses: Syslinux/Isolinux v4.06
  Uses: EFI
Using image: systemrescuecd-x86-5.3.1.iso (547.3 MB)
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Zugriff verweigert
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Zugriff verweigert
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Zugriff verweigert
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Zugriff verweigert
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Using autorun.inf label for drive H: 'System Rescue CD 5.3.1'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0008524F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.0 GB (16105078784 bytes)
  Start Sector: 2048, Boot: Yes

WARNING: 'Controlled Folder Access' appears to be enabled on this system
You may need to disable this feature, or add an exception, for Rufus to to work...
```

pbatard, 2018-10-12:
You (or someone else) probably added and exception for Rufus 3.2 in Windows Defender's 'Controlled Folder Access', as 3.3 was a quick bugfix release, and there is literally nothing that can even remotely prevent write access in what was changed between 3.2 and 3.3.

'Controlled Folder Access' (which you could have easily found by googling that term) is a feature of Windows Defender (the default Antivirus on Windows 10) that prevents applications from writing outside of their user directory. As such, if enabled, it obviously prevents Rufus from writing to USB drives, and this is why we added detection for it. If Rufus reports that 'Controlled Folder Access' is enabled, then you can be pretty sure that this __is__ the issue.

You can enable exceptions in 'Controlled Folder Access' for some applications, which, judging by your report, is most likely what you (or whoever controls the system) did for Rufus 3.2, but you seem to have forgotten to do so for Rufus 3.3.

At any rate, you need to go to _Virus & Threat protection settings_ and then select _Manage Controlled folder access_ and either turn it off altogether, or, click _Allow an app through controlled folder access_ and make sure Rufus is there.

If you need more help on how to set up 'Controlled Folder Access', please google it, as you should find plenty of tutorials on that.

engelhro, 2018-10-12:
> You (or someone else) probably added and exception for Rufus 3.2 in Windows Defender's 'Controlled Folder Access' […]

Can't remember that. Indeed "Controlled Folder Access" was activated, but *without any exceptions at all*. Still worked for v3.2 somehow :hushed:

> 'Controlled Folder Access' (which you could have easily found by googling that term) [….]

Yes, my bad, sorry :disappointed_relieved: – was too lazy as I already started to type the issue and discovered the warning message while pasting the log into it.

> If Rufus reports that 'Controlled Folder Access' is enabled, then you can be pretty sure that this **is** the issue.

Okay, thanks for the hint! I now verified that Rufus v3.3 both works with "Controlled Folder Access"completely disabled as well as it being enabled (and the executable put on the whitelist then).

> You can enable exceptions in 'Controlled Folder Access' for some applications, which, judging by your report, is most likely what you (or whoever controls the system) did for Rufus 3.2, but you seem to have forgotten to do so for Rufus 3.3.

I did not forget, I was simply not aware of it :wink: I've no idea why v3.2 (still) works with "Controlled Folder Access" and no special configuration for it (i.e. without an exception!).

> At any rate, you need to go to _Virus & Threat protection settings_ and then select _Manage Controlled folder access_ and either turn it off altogether, or, click _Allow an app through controlled folder access_ and make sure Rufus is there.

In case anyone later stumbles upon this issue, here is the navigation path in a German Windows 10:

Einstellungen → Update und Sicherheit → Windows-Sicherheit → Viren- & Bedrohungsschutz → Ransomware-Schutz → Überwachter Ordnerzugriff.
So easy to find and navigate there :stuck_out_tongue_winking_eye:

Anyway: a **big thanks** for the support and the fast and helpful reply! Issue already solved (within one hour :+1:), I'll thus close it.

closed 2018-10-12 by engelhro