pbatard/rufus #1227: rufus.akeo.ie and rufus.ie are currently down

pbatard, 2018-10-16:
Our web site is experiencing a coordinate DDoS attack since earlier today (Can you imagine there's a low life loser out there who thinks that providing a utility for free should be punished? Apparently they weren't smart enough to get back at us through debate, so they had to resort to dumb script kiddie tactics 😄).

We're working on restoring functionality. More information will follow when we have it.

JDTAY, 2018-10-16:
I was about to ask... I can believe it.

Script kiddies, script kiddies every where.

pbatard, 2018-10-16:
Or a disgruntled security "researcher"...

The thing is, we have started to accumulate a lot of circumstantial evidence against a specific person, who's definitely not as smart or stealthy as they think they are. So much so that I am wondering how much of it would hold in court, if we were to take legal action against them...

ibuprophen1, 2018-10-16:
It's weird that the xda-developers just got done after spending a little over a month with a DDOS attack on the forum server. 

Whom ever it is, i hope they get what they deserve! 

Your hard work @pbatard is greatly appreciated and I truly wish you nothing but, the best of luck in getting up and running! :-) 

~Ibuprophen

bkkgbkjb, 2018-10-16:
Hi

sorry for the troubles you're having

Currently I'm able to get all the websites and downloads of rufus .exe work

However, my rufus promts to download Syslinux which as you mentioned, is temporarily down

What should I do to get the files downloaded? 
Otherwise my .exe doesn't work...

pbatard, 2018-10-16:
@bkkgbkjb, the following:

> Additional file download (Syslinux, GRUB) has not been restored yet

is why you are experiencing the download issue. I'm in the process of fixing this, but __please__ be patient, okay.

Also, https://rufus.ie should be working again for the main web page, so no need to use HTTP.

I'll keep you updated here.

pbatard, 2018-10-16:
Okay, file downloads for GRUB and Syslinux should work again now.

I also set up a redirect for rufus.akeo.ie & rufus.ie which should help people accessing the older domain, but SSL hasn't kicked in on that, so you're probably going to have issues with that redirection until it does (probably in a few hours, if it's like SSL for rufus.ie).

bkkgbkjb, 2018-10-16:
Hi pbatard

Yes, I just tried your awesome rufus and it works perfectly

Sorry for pushing you here on Github

It's just this morning(GMT +8) I decided to reinstall my whole computer

After removing the existing OS, I found your rufus cannot work and had no choice 
but to wait for roughly 3 hours

I've tried a lot of similar solutions 
none of them works with my Debian netboot mini.iso

Really appreciate your hardwork and contribution on maintaining such a 
great utility

pbatard, 2018-10-16:
> It's just this morning(GMT +8) I decided to reinstall my whole computer

If you really wanna have fun, try reinstalling server content __twice__, during the same day, whilst being throttled to 80 KB/s by your ISP... 😄 

All in all, that's still a lot of fun though, coz I was able to validate that my disaster plan, for just this situation, could indeed work, and right now, the person who attacked me has no more cowardly place to run. They're going to have to do __much__ better than this next time around...

Anyway, glad to hear you sorted out your installation now that the files are back up. And don't worry, people can __try__ to attack us, but, as long as they clearly aren't able to use more than 2 brain cells to do so, there's no much risk that they'll amount to much of anything in terms of preventing you guys from continuing to rely on Rufus!

pbatard, 2018-10-16:
* https://rufus.akeo.ie should now be fully available again (if your DNS servers have been refreshed).

luckman212, 2018-10-16:
Thanks for everything @pbatard 
Can't believe these lowlifes have nothing better to do.

pbatard, 2018-10-17:
Since the sites should be accessible again, I'm going to close this issue.

closed 2018-10-17 by pbatard

vn7kr, 2018-10-17:
While [https://rufus.ie]( https://rufus.ie) is working fine (uses SSL), [https://rufus.ie/downloads](https://rufus.ie/downloads) is redirected to non-SSL  [http://rufus.ie/downloads](http://rufus.ie/downloads). Not sure if it's because of something on my end.

@pbatard thanks for taking care of this whole issue so promptly.

vn7kr, 2018-10-17:
Re: [the above issue](https://github.com/pbatard/rufus/issues/1227#issuecomment-430687880) it's working fine now. Thanks!

vn7kr, 2018-10-17:
@pbatard thanks for looking into the issue and the explanation. It's weird that GitHub does the https to http redirect for a sub-URL.

pbatard, 2018-10-20:
Hmmm, I have now enforced both rufus.ie and rufus.akeo.ie to use SSL always from the github interface (which means that any request to `http://...` should be converted to `https://...`) but that doesn't seem to change anything for https://rufus.ie/downloads.

This is definitely a github issue, as reported in isaacs/github#289. According to the last entry there (from 2 days go), GitHub support reported:

> This redirection quirk that you've discovered is a known issue, and relates to trailing slashes and how our redirects work at the moment. Making a request to a URL without a trailing slash will cause redirections to run in this order:
> 
> * request to original URL
> * 301 to HTTPS version of original URL
> * 301 to HTTP version with added trailing slash
> * 301 to HTTPS version with added trailing slash
> 
> Always including the trailing slash will allow you to skip the extra redirects:
> 
> * request to URL with trailing slash
> * 301 to HTTPS version of original URL
> 
> We have an issue open internally to hopefully improve this situation, but investigations have demonstrates that it is a non-trivial issue to resolve.

ibuprophen1, 2018-10-20:
It's a weird thing... 

URL:
https://rufus.ie/downloads

Results:
https://rufus.ie/downloads
301 Moved Permanently
http://rufus.ie/downloads/
301 Moved Permanently
https://rufus.ie/downloads/
200 OK

[ToolBot Headers Log.txt](https://github.com/pbatard/rufus/files/2498581/ToolBot.Headers.Log.txt)

~Ibuprophen