pbatard/rufus #1228: Rufus website is down

cyfrost, 2018-10-16:
Hi, I don't know where else to say this but looks like it'd be noticed here. The rufus website is down. Is this a maintenance chore?

Thanks!

blackcrack, 2018-10-16:
you meant this ?

```
Your connection is not secure

The owner of rufus.akeo.ie has configured their website improperly. To protect your information from being stolen, Basilisk has not connected to this website.

rufus.akeo.ie uses an invalid security certificate.

The certificate is only valid for the following names:
  www.github.com, *.github.io, *.githubusercontent.com, *.github.com, github.com, github.io, githubusercontent.com  

(Error code: SSL_ERROR_BAD_CERT_DOMAIN)

[Add Exception...]
```

well, this is at my domain the same, because self signed Certificate ..
you trust or you trust not this page.. i trust and add a Exception..

best regards
blacky

cyfrost, 2018-10-16:
@blackcrack take a look at OP, edited with screen capture.

yiijam, 2018-10-16:
use this site https://rufus.ie

cyfrost, 2018-10-16:
@yiijam Thank you, that does it for now. Although it still remains to be seen why the github pages version is down. Also, I was wondering, why not take a https://www.rufus.com domain ? I don't mean to come across as rude or abraisive, just genuinely wondering :)

pbatard, 2018-10-16:
> Also, I was wondering, why not take a https://www.rufus.com domain ?

It was already taken. And `.com` is for commercial. I'm not selling anything...

Also, next time you open an issue, please check the existing ones. There literally was an issue than answered your questions right before the one you opened (yes, the checklist about looking for prior issues does apply even if it's not a Rufus application problem). See #1227.

As such, since this is a duplicate, I will close this one.

closed 2018-10-16 by pbatard