pbatard/rufus #1230: The window of Rufus disappears

Lukaszwiatkowski, 2018-10-17:
After choose an ISO image (differents images) and click START - the window of Rufus disappears.
In disk management - USB disk is without letter.  After assign a letter, it turns out that the USB disk has not been formatted.
Tested on few USB disks.

pbatard, 2018-10-17:
I'm going to need __WAY__ more information than that. First of all, you chose to completely ignore the checklist (How about a log from before the window disappears?), especially as you the the __ONLY__ person to ever report such an issue, whereas more than 3 million people use Rufus each month.

In other words, this clearly looks like an environmental issue, and, unless you provide details about what in your environment is different from the millions of other people who use Rufus without issue, it's going to be __impossible__ to troubleshoot this issue.

At the very least, please try to test on a different machine. And please indicate what version of Windows you are using, what arch, what locale, etc.

Lukaszwiatkowski, 2018-10-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer. - on a different computer Rufus works
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm using a Windows 7 Sp1 64-bit PL.  Previously, Rufus worked without a problem on this computer. Nothing has changed.

Log
---
```
<FULL LOG>
Rufus version: 3.3.1400 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0415
Will use default UI locale 0x0415
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
0 devices found
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
Found USB 2.0 device 'General USB Flash Disk USB Device' (8640:800F)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 975, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001475CB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8021606400 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'General USB Flash Disk USB Device' (8640:800F)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 975, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001475CB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8021606400 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2018-10-17:
> on a different computer Rufus works

Which confirms that this is an environmental issue.

> Nothing has changed.

Something __must__ have changed, even if you didn't consciously initiate it. It could be an update, or maybe you changed something in your security solution that now freezes applications when they are trying to access a drive at low level (which would explain why the window disappears after you press Start)

Now, one thing that could help, since Rufus sends its logging data to the global Windows debugging facility, is if you installed DebugView (A Microsoft utility that you can get from [here](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview)) and then ran Rufus, as we will then see what Rufus is doing, even if the window has disappeared, which may give us some clue as to what it is in your environment that makes the application disappear.

Also, since my bet is on yet another "security" solution overexerting its mandate, and trying to freeze an application that it deems "unsafe", I would strongly encourage you to check your security logs...

closed 2018-10-17 by Lukaszwiatkowski