pbatard/rufus #1232: Can't burn Ubuntu 18.10 ISO because a missing DLLs

castillo92, 2018-10-19:
Issue description
-----------------
I want to burn the last ISO of Ubuntu 18.10, but Rufus need some files to do it and cant get it.

Log
---

> Rufus version: 3.3.1400 (Portable)
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02
> System locale ID: 0x0C0A
> Will use default UI locale 0x0C0A
> SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
> Localization set to 'es-ES'
> Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
> 1 device found
> Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
> Cylinders: 941, Tracks per cylinder: 255, Sectors per track: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0xA179A452
> Drive has a Windows 7 Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 7.2 GB (7742685184 bytes)
>   Start Sector: 2048, Boot: Yes
> Scanning image...
> ISO analysis:
>   Image is an ISO9660 image
>   Will use '/isolinux/isolinux.cfg' for Syslinux
>   Detected Syslinux version: 6.04/20180907 (from '/isolinux/isolinux.bin')
> Disk image analysis:
>   Image has an unknown Master Boot Record
>   Image is a bootable disk image
> ISO label: 'Ubuntu 18.10 amd64'
>   Size: 1.9 GB (Projected)
>   Uses: Syslinux/Isolinux v6.04
>   Uses: EFI
>   Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
>   Because of this, some features from this image may not work...
> Using image: ubuntu-18.10-desktop-amd64.iso (1.9 GB)
> Downloading https://rufus.ie/files/syslinux-6.04/20180907/ldlinux.sys
> Unable to send request: This system's SSL library is too old to be able to access this website.
> Extended version was not found, trying main version...
> Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
> Unable to send request: This system's SSL library is too old to be able to access this website.
> Could not download the file - cancelling
> Downloading https://rufus.ie/files/syslinux-6.04/20180907/ldlinux.sys
> Unable to send request: This system's SSL library is too old to be able to access this website.
> Extended version was not found, trying main version...
> Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
> Unable to send request: This system's SSL library is too old to be able to access this website.
> Could not download the file - cancelling

jjboy32, 2018-10-20:
Ubuntu-18.04.1-desktop-amd64, Rufus requires two necessary files, ldlinux.bss and ldlinux.sys, to be downloaded online when carved into a Usb disk. This will enable you to install the PC into the system before you can succeed. Rufus is very responsible. Just like when the Microsoft WIM file is larger than 4G, the EFI boot and engrave into the U disk also requires Rufus to add the necessary partitions to enable the successful EFI boot and install the PC as the GPT partition.

jjboy32, 2018-10-20:
If Rufus does not download the necessary files, it can also be obtained online.

castillo92, 2018-10-21:
> You need to make sure that your Windows 7 system has all the recommended system updates. If you do that, then support for TLS 1.2 (which is what is missing here) should get added to your system automatically.
> 
> Alternatively, you may want to see [this](https://support.microsoft.com/en-us/help/3140245/update-to-enable-tls-1-1-and-tls-1-2-as-a-default-secure-protocols-in) or [this](https://github.com/rust-lang/cargo/issues/5066), that relate to the problem you are seeing.
> 
> I tested that on Windows 7 system with all the recommended updates, this issue does not occur.

Thanks you very much! It's works great!

I access the first link and tried the "Easy fix", which install the TLS capabilities.

Regards! =)

closed 2018-10-21 by castillo92

kevin-olbrich, 2018-11-23:
This also happens on Windows 10 1803 and Rufus 3.3.1400.
In general, this is always a problem, when TLS-Support in IE has been limited.