pbatard/rufus #1233: Missing first character of image name in Boot selection

chazste, 2018-10-19:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When using the -i command line option the name of the image file as displayed on the status bar and in the boot selection control is missing the first character. In the log below notice that the line "Image provided" is correct and the last line is missing the leading C.

I'm assuming that the issue is line 1073 of rufus.c where the search for the last backslash does not allow for the case where there are none. (i.e. it was provided as a relative path in the current directory from the command line.)

Log
---
Rufus version: 3.3.1400 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TEAC USB   HS-CF Card USB Device' (0644:0200)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'TEAC USB   HS-MS Card USB Device' (0644:0200)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'TEAC USB   HS-SD Card USB Device' (0644:0200)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'TEAC USB   HS-xD/SM USB Device' (0644:0200)
Device eliminated because it appears to contain no media
0 devices found

Image provided: 'CatUsbInstaller.iso'
Scanning image...
ISO analysis:
  Image is an UDF image
  Will use '/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/2014-10-06 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'cat usb image tool'
  Size: 4.4 GB (Projected)
  Has a >4GB file
  Uses: Syslinux/Isolinux v6.03
Using image: atUsbInstaller.iso (4.4 GB)

<FULL LOG>
```

ibuprophen1, 2018-10-19:
I noticed that it states that the image is UDF but, also states that it does not have an x86 Master Boot Record. 

If the image is UDF, did you select that as your partition scheme? 

Or am I off base... :-) 

~Ibuprophen

pbatard, 2018-10-20:
@ibuprophen1, you are way off mark here.
UDF is not a partition scheme, (classic) x86 boot records are not being used on ISO images, and the issue has literally nothing to do with anything you mentioned, as it has pretty much already been identified by OP.

You don't have to try to comment on every issue... ;)

ibuprophen1, 2018-10-20:
My apologies... LMAO!

commit 7757cab3 (2018-10-22): [ui] fix possible truncation of short image path

* Closes #1233
* Also fix MinGW builds due to missing PROCESSOR_ARCHITECTURE_ARM64 define
closed 2018-10-22 by pbatard