pbatard/rufus #1237: Can't access or format my usb anymore [Write error at sector 0: [0x00000079]] error

John-NoobPilot, 2018-10-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I was using rufus for a long time for many distros , Yesterday i left rufus installing ubuntu on my usb and forgot to lock my pc to find out that my little sister was playing with my pc and canceled the operation , So the first (Bright idea BTW) was to format the usb before using rufus again and it was by the clean command in diskpart and it took a long time so without any thinking i cancel the clean operation and tried to format it by clicking on format on the device and same result (taking a long time and fail) i tried KDE partition manager and it stuck at 0.0% present disable and enable it from device manager didn't help, Tried clean and then clean all from diskpart but it fails with the error [DiskPart has encountered an error: The semaphore timeout period has expired.] , And the last thing i tried was rufus again but i'm getting [Write error at sector 0: [0x00000079] The semaphore timeout period has expired.] .

The usb is still active at the disk management
[Status on disk managgement](https://imgur.com/a/LWPo0Af)
[Trying to format it](https://imgur.com/a/eU7RZOr)
[Fail](https://imgur.com/a/I4Ux2hP)
[On my computer](https://imgur.com/a/Tez0jme) 

Microsoft DiskPart version 6.1.7601
Copyright (C) 1999-2008 Microsoft Corporation.
On computer: STRIKER-PC

DISKPART> list disk

  Disk---###---Status---Size---Free---Dyn---Gpt
  Disk---0---Online---465 GB---0 B
  Disk---1---Online---3700 MB---0 B

DISKPART> select disk 1

Disk 1 is now the selected disk.

DISKPART> detail disk

ADATA USB Flash Drive USB Device
Disk ID: 0004CBF5
Type   : USB
Status : Online
Path   : 0
Target : 0
LUN ID : 0
Location Path : UNAVAILABLE
Current Read-only State : No
Read-only  : No
Boot Disk  : No
Pagefile Disk  : No
Hibernation File Disk  : No
Crashdump Disk  : No
Clustered Disk  : No

  Volume---###---Ltr---Label---Fs-------Type-------Size-------Status---Info
  Volume---4------F------------RAW----Removable--3699 MB-Healthy

DISKPART> clean

DiskPart has encountered an error: The semaphore timeout period has expired.
See the System Event Log for more information.

DISKPART> clean

DiskPart has encountered an error: The semaphore timeout period has expired.
See the System Event Log for more information.

DISKPART> clean all

DiskPart has encountered an error: The semaphore timeout period has expired.
See the System Event Log for more information.

Log
---
```
<Rufus version: 3.3.1400 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:C96A)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004CBF5
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.6 GB (3878682624 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-WS-dvd-x86_64-28'
  Size: 654.5 MB (Projected)
  Note: File on disk is larger than reported ISO size by 768 KB...
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-Workstation-netinst-x86_64-28-1.1.iso (583 MB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/' for Syslinux 

installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
Requesting lock...
Writing Image...
Write error at sector 0: [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Write error at sector 0: [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Write error at sector 0: [0x00000079] The semaphore timeout period has expired.
Retrying in 5 seconds...
Write error at sector 0: [0x00000079] The semaphore timeout period has expired.
Drive has a Syslinux Master Boot Record
Remounted \\?\Volume{c427c09b-c2b6-11e8-828e-94de80ea31bc}\ on F:\
\\?\Volume{c427c09b-c2b6-11e8-828e-94de80ea31bc}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:C96A)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004CBF5
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.6 GB (3878682624 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt>
```

pbatard, 2018-10-28:
Well, this looks like a hardware or environmental issue. And if the Windows native tools can't partition or reformat the drives, there's very advice I can provide. Not sure what happened as your little sister was using the PC (maybe she yanked the drive out and then plugged it back in again), but I doubt the fact that Rufus was running when that happened had any incidence on the issue. You could have been copying a bunch of files using Windows File Explorer (which is pretty much exactly what Rufus was doing at the time), and you would have run into the same issue.

Best advice I can give is to try with a different drive. You may also try the bad blocks check as suggested by the check list but it will probably produce the same error...

JonnyTech, 2018-10-28:
Does the drive have a physical lock switch on it? It may not report back to the operating system that it is in read-only mode, hence the problem.
[image][image]

ibuprophen1, 2018-10-28:
@John-NoobPilot, based upon the information you had provided, your thumb drive may have software within it (as with many of the thumb drives out there) that might have an active lock you may have to turn off/disable first. 

The following link should be helpful for this specific to your thumb drives manufacturer. 

http://www.adata.com/en/ss/software-2/

Good Luck! :-) 

~Ibuprophen

John-NoobPilot, 2018-10-29:
Thank you all for your help and sorry for the delay .
@pbatard 
>  (maybe she yanked the drive out and then plugged it back in again)

i don't think that she manged to unplug the usb [especially from that place for a kid](https://imgur.com/a/9tIlX5s)

> but I doubt the fact that Rufus was running when that happened 

no actually it was running and installing ubuntu iso on it

> You could have been copying a bunch of files using Windows File Explorer

no 

> You may also try the bad blocks check as suggested by the check list but it will probably produce the same error...

yes same error

@JonnyTech  Unfortunately no

@ibuprophen1  I did download the software but it returned with [this](https://imgur.com/a/A2ZX74R) 

There is another thing i would like to mention , when i try to format my usb (and fail) windows wont be able to show it [again](https://imgur.com/a/efInr6Y) even if i unplug/re-plug it again and the only way to solve it is to disable it from device manger unplug/re-plug it again and then  [enable](https://imgur.com/a/KHw4ABm).
Disable/enable alone wont help i have to unplug it after i disable and then re-plug again then enable it.

I may have went deep for for this issue and turn it into a hardware issue instead of rufus issue but is it possible that my usb is dead now

pbatard, 2018-10-29:
>  >  but I doubt the fact that Rufus was running when that happened
>
> no actually it was running and installing ubuntu iso on it

You misunderstand what I said. I didn't say that I doubted you were running Rufus when that happened. I'm just indicating that it doesn't matter if it was Rufus doing the file copy or File Explorer, because, behind the scenes, there is __no difference__ at this stage of the process.

> >   You could have been copying a bunch of files using Windows File Explorer
>
> no

Not sure what you are trying to refute, but I'm afraid that if you are implying that Rufus and File Explorer are not using the __exact__ same APIs for copying files, and therefore that, as I indicated, it really doesn't matter what application was copying files when the issue occurs, then I am going to ask for your credentials as a developer.

I can understand that you would __like__ the fact that Rufus does something special compared to File Explorer when copying files to the USB. Unfortunately, the reality of how Windows APIs work is that there are no fundamental differences. So, as much as you would like to blame Rufus, you're not going to be able to do that unless you can point to something in the Rufus code that refutes what I just asserted. But of course, if you spend time looking at the code, you will see that the only thing Rufus does when copying the files is calling `WriteFile()`, which is also what File Explorer does. __Especially__, and this is something that people who are unfamiliar with how Rufus actually works, we're not sending any low level USB commands on the bus, or even altering disk sectors during the file copy operation. So, __again__, I am 100% positive that whatever behaviour you are trying to blame on Rufus would have happened exactly the same if you had been manually copying the content of your ISO manually using File Explorer. As such, if you are waiting for a miracle solution to what looks like a hardware problem, you're only going to be disappointed. 

Most likely your USB Flash Drive is dead and you will have to replace it.

John-NoobPilot, 2018-10-29:
> You misunderstand what I said. I didn't say that I doubted you were running Rufus when that happened. I'm just indicating that it doesn't matter if it was Rufus doing the file copy or File Explorer, because, behind the scenes, there is no difference at this stage of the process.

Sorry for my misunderstanding here 

> Not sure what you are trying to refute, but I'm afraid that if you are implying that Rufus and File Explorer are not using the exact same APIs for copying files, and therefore that, as I indicated, it really doesn't matter what application was copying files when the issue occurs

Again sorry i should have provided a clear answer instead of just saying no what i meant is that i wasn't using anything(copying files to the usb or anything else) when Rufus was running 

I'm really sorry , I did start an issue here because i thought i might have a solution from(Write error at sector 0: [0x00000079]) error i wasn't trying to blame Rufus for this at all

pbatard, 2018-10-29:
No worries. I think I overreacted a little as well, and I can understand how what I said could be misinterpreted.

I wish I had a solution for you, but the type of Windows error you are receiving is usually a clear indication of a hardware issue.

The only thing I can suggest is, if you have a Linux machine, to try to repartition and reformat your drive there, just in case you're running into a weird Windows behaviour, but apart from that, it looks like you will have to get a new drive.

closed 2018-10-30 by John-NoobPilot