pbatard/rufus #1238: List Hard drives in the System

kdpatino, 2018-10-29:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Hello, I want to ask if it is possible to list and install FreeDos with RUFUS in one of the Hard Drives in the system. In my case I have 3 Hard Drives, and I want to use one of them with FreeDOS. So I want to know if rufus could list the hard drives in the system.

Thank You.

Log
---
```
<FULL LOG>
```

pbatard, 2018-10-29:
FreeDOS should be installed in any drive that is listed by Rufus.

On the other hand, internal system hard drives (such as SATA or SSD drives) are __not__ listed in Rufus, by design choice.

So your option is to either use an USB external drive (which you can enable the listing of as per https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected), or (but this is __100% UNSUPPORTED___ and I take __NO RESPONSIBILITY IF YOU END UP ERASING CRITICAL SYSTEM DATA__), if your drive has the "removable" set in your BIOS (up to you to figure that out), you may try the <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> cheat mode.

That is all the extent of the help I will provide on this matter, since the current behaviour of Rufus will not change, regardless of whether you manage to achieve what you want, and unsupported cheat modes mean I don't provide support on them. As such, I will close this issue.

closed 2018-10-29 by pbatard

ibuprophen1, 2018-10-29:
I don't believe that Rufus was even designed for this but, why not go to http://www.freedos.org and grab a copy to put on your HDD... 

~Ibuprophen