pbatard/rufus #1239: Rufus have problems to detect a USB stick

Sopor-, 2018-11-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
I have a USB stick that Rufus have problems to detect. It will show up in Windows but Rufus will still show 0 devices until i enable `List USB Hard Drives` and when it show up i can disable `List USB Hard Drives` and it will still show 1 device so it is not a Hard drive device.

Log
---
```
Rufus version: 3.3.1400
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8F02F9D9
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16021192704 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8F02F9D9
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16021192704 bytes)
  Start Sector: 2048, Boot: No
```

pbatard, 2018-11-01:
Try <kbd>Alt</kbd>-<kbd>.</kbd> to enable USB enumeration debugging and post the new log. It should tell us what the story is.

Sopor-, 2018-11-01:
Before i press the `List all USB hard drives`

```
Rufus version: 3.3.1400
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
  Found ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing IDs belonging to 'USBSTOR':
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
```

When i press `List all USB hard drives`

```
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
  Found ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing IDs belonging to 'USBSTOR':
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\Disk________USB_DISK_Pro____PMAP'
  Matched with ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
  Matched with Hub[4]: '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {FD58018A-EC3D-4A99-BD30-B3BD3AB3A306}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4CDEFD9A-90CA-447B-8FBB-81FAF02544A4}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2018-11-01:
Hmm, that is super weird...

We don't get `USBSTOR` enumerated when USB HDDs are disabled, even though the option does literally nothing to prevent it (we're just [calling `SetupDiEnumDeviceInfo()` here with basic options](https://github.com/pbatard/rufus/blob/c2ba8fc0d2e72d9c9c4417da0c46e1a1e6778cec/src/dev.c#L503), that are exactly the same in either case and which __should__ always give enumerate `USBSTOR` devices.

The fact that we don't even get the line:
```
Processing 'USBSTOR' device:
```
means that, as far was as `SetupDiEnumDeviceInfo()` is concerned, there are no `USBSTOR` devices at all, which is a very strange behaviour that should be impervious to any bug we might have in our code (since there is no code that can make us break out of the `for` loop that goes over the `SetupDiEnumDeviceInfo()` enumerated devices).

Add these weird `Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'` errors that are reported when USB HDDs are disabled (NB: the reason they seem to occur when processing an unrelated device is simply because, for each new device reported by `SetupDiEnumDeviceInfo()`, we are always looping through all the stuff we've enumerated in the first pass to look for children, since [we don't keep a lookup table](https://github.com/pbatard/rufus/blob/c2ba8fc0d2e72d9c9c4417da0c46e1a1e6778cec/src/dev.c#L574-L577)), and I'm starting to believe this could very well be a Windows enumeration bug, where stuff fails the first time you try to enumerate them. That would certainly explain why your device remains enumerated even after you disable USB HDDs.

I need to think about this some more, but right now, I fail to see how this could be explained by a bug in Rufus. Especially, I really don't see how `CM_Locate_DevNodeA() + CM_Get_Child()` seem to start to work all of a sudden, when all we feed these API calls is very basic static data, that should not change from one enumeration request to the next.

One thing I would like you to answer (since you skipped that part of the checklist) is:
- What happens if you try with the same USB drive on a different Windows machine? Especially, since you are using Windows 8.1, I would really like to know if you see the same on Windows 10...

Sopor-, 2018-11-04:
Exact the same thing happens on Windows 10 on a complete different machine :smiley: 

```
Rufus version: 3.3.1400
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x041D
Will use default UI locale 0x041D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'sv-SE'
0 devices found
0 devices found
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {FD58018A-EC3D-4A99-BD30-B3BD3AB3A306}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4CDEFD9A-90CA-447B-8FBB-81FAF02544A4}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {FD58018A-EC3D-4A99-BD30-B3BD3AB3A306}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4CDEFD9A-90CA-447B-8FBB-81FAF02544A4}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2018-11-05:
Thanks for the test.

Still can't think about anything that Rufus could do that might be the cause of this issue. Right now, I am still more inclined that it's a firmware issue with that specific drive.

As to the reason why Windows would detect the drive whereas Rufus eliminates it the first time around, this can be explained if the firmware does not properly implement one of the USB queries Rufus relies on during its enumeration phase (whereas Windows would not send such a query natively when listing the device in File Explorer for instance), and, since Rufus does rely on that query to be successful before it lists the device, the device will be eliminated if the firmware cannot properly answer to it.

Then I suspect the second time this query is sent, the firmware is able to answer it (it could be a bad or slow initialization issue, and the second time around, the necessary elements have been initialized to honour the query), which is why it then seems to be okay. There might also be some combining factors, such as what other USB queries have been issued before, that could impact the detection.

At any rate, while it is still very possible that it's a bug in Rufus (but one that I'm not sure I want to invest that much time trying to troubleshoot, especially when I don't have the means to replicate it and when it only seems to affect a single device), bugs in Flash Drive firmwares are also not unheard of, because these devices have to answer a whole slew of potentially complex USB queries, and, if you only ever test with File Explorer, you may miss bugs in your firmware code because only a subset of the queries that your firmware must answer will be stressed. This might be even more true if it's a variable initialization problem that only happens the first time a query is sent after a device is plugged, as, even if the manufacturer happens to have a comprehensive test, unless they replug the device before each query, they might miss that issue...

All of this means that, for the time being, I'm going to set this issue to deferred, as I'm not planning to look into further unless I'm starting to get a few more reports, from people using a different Flash Drive, that would seem to indicate that the issue is in Rufus itself, and not the drive's firmware...

Sopor-, 2018-11-05:
If i instead run Rufus 2.1.649 it works

```
Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
0 devices found
0 devices found
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1947, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {FD58018A-EC3D-4A99-BD30-B3BD3AB3A306}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4CDEFD9A-90CA-447B-8FBB-81FAF02544A4}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2018-11-05:
Hmm, okay. Then that changes things.

Still not a high priority for me though (I have other stuff to keep me busy ATM), so if you can find out which version of Rufus, between 2.1 and 3.4 started to have the problem, that would help.

Sopor-, 2018-11-05:
Rufus 3.1.1320 - working
Rufus 3.2.1397 - not working

pbatard, 2018-11-05:
Many thanks for testing this. I'll look into this further when I get a chance, but it will probably be a week or two before I do that.

pbatard, 2018-11-23:
Okay, I'm coming back to this issue.

@Sopor, can you please download [`rufus-3.4_TEST2.exe`](https://rufus-web.akeo.ie/testing/rufus-3.4_TEST2.exe) and run it with your USB then copy/paste the log (no need to have <kbd>Alt</kbd>-<kbd>.</kbd> enabled, but you can try with it if you want)?

This adds a couple `GetDevices:` messages to the log that I hope will shed some light on why your device can't get detected in first pass.

Sopor-, 2018-11-23:
You should use `Sopor-` and not `SopoR` when you mention me :)

I run Rufus, i plugin the USB stick, not detected by Rufus, i enable `List USB Hard Drives`, the USB stick will get detected, i disable `List USB Hard Drives`. I will only explain what i do so you can follow it in the log file. Here is the log:
```
Rufus x86 v3.4.1426 (Test2)
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
GetDevices: CurRefresh = 1224461.921, LastRefresh = 0.000
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
GetDevices: CurRefresh = 1224492.203, LastRefresh = 1224461.921
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
  Found ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing IDs belonging to 'USBSTOR':
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
GetDevices: CurRefresh = 1224492.375, LastRefresh = 1224492.203
GetDevices: Less than 1 sec elapsed
GetDevices: CurRefresh = 1224583.296, LastRefresh = 1224492.203
Processing Hub 1:
Processing Hub 2:
[... 59 lines cut ...]
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\Disk________USB_DISK_Pro____PMAP'
  Matched with ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
  Matched with Hub[4]: '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {BF989238-4DFC-44E5-B8E0-1E2894FCFB4E}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {0D83A302-2A47-4C24-8D01-10817D689C0D}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2018-11-25:
I don't get it. The __ONLY__ change that could remotely apply to your issue between 3.1 and 3.2 is [a472e96e87ca563d174a36729742a8d952a07716], which is what I tried to instrument.

But the log indicates that this change is not the cause of your issue.

Are you absolutely sure that the difference of behaviour between 3.1 and 3.2 was not a fluke during your testing?

For good measure, I have just uploaded [`rufus-3.4_TEST3.exe`](https://rufus-web.akeo.ie/testing/rufus-3.4_TEST3.exe) that reverts the changes from [a472e96e87ca563d174a36729742a8d952a07716]. If that does not revert the behaviour you observe for this device, then I'm really not sure where else to look...

Sopor-, 2018-11-26:
With `rufus-3.4_TEST3.exe` it is working perfect ...

```
Rufus x86 v3.4.1426 (Test3)
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
  Found ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing IDs belonging to 'USBSTOR':
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&318a7e80&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[213]: USB\VID_05E3&PID_0745\000000000903
  Found ID[242]: USB\VID_1A40&PID_0101\5&DC4A972&0&14
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&144fac0d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[177]: USB\VID_8087&PID_8001\5&D1AA17F&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&f432ec7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_8087&PID_8009\5&182A3301&0&1
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[025]: USB\VID_046D&PID_C077\6&344E08ED&0&4
  Found ID[170]: USB\VID_060B&PID_6220\6&344E08ED&0&3
  Found ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing IDs belonging to 'USBSTOR':
  USB\VID_13FE&PID_4123\070B2B0BB421EE77
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_840_EVO_250GEXT0'
Found non-USB non-removable device 'Samsung SSD 840 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000NM0033-9ZMSN04'
Found non-USB non-removable device 'ST2000NM0033-9ZM175' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\Disk________USB_DISK_Pro____PMAP'
  Matched with ID[193]: USB\VID_13FE&PID_4123\070B2B0BB421EE77
  Matched with Hub[4]: '\\?\usb#vid_1a40&pid_0101#5&dc4a972&0&14#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {BF989238-4DFC-44E5-B8E0-1E2894FCFB4E}
Max parts: 128, Start Offset: 17408, Usable = 16022206976 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {0D83A302-2A47-4C24-8D01-10817D689C0D}
  Size: 14.9 GB (16021175808 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2018-11-26:
I don't get it. There's no way [these 4 lines of code](https://github.com/pbatard/rufus/blob/master/src/dev.c#L323-L327) should have any incidence on the `Could not get children of 'USB\VID_13FE&PID_4123\070B2B0BB421EE77'` messages you observe. The code was added to decide whether to run the device detection __as a whole__ or not, and instrumentation demonstrated that, even when it is not commented out, the initial device detection for your device does run up to the end (albeit with the `Could not get children` messages), so it's not a matter of the added code actually preventing something that used to run from running..

Also, this is not a section of code I can revert, first of all, because there's no logic between reversing it and not getting the `Could not get children` messages and second, because it will reintroduce the refresh problems that this code is meant to fix.

So I'm really tempted to just give up on this issue (unless I get my hands on a device that can replicate it on my side), because it defies all programming logic that I know of and I really don't have the time to try to pull everything apart to figure why a subsystem (for which I don't have the source and that I cannot instrument, since it's Windows') is being impacted by innocuous and completely unrelated changes. If it were memory leaks (due to an increase of code segment size or something, from the added code), I would expect to see side-effects ripple across more than one device and on other parts of the application. But that is not the case...

I may try another TEST release or two, in the blind, to see if it helps. But I'm afraid I don't have the drive of a _Fox Mulder_ to spend weeks on end investigating and inexplicable phenomenon, especially when I have plenty of other things lined up. So, if it's only a single device of yours that's not being properly detected, I can live with that...

pbatard, 2018-11-28:
Okay, because I was running into other issues due to the new refresh prevention code, I have now reverted it for the 3.4 release. I therefore expect this issue to be closed.

There's a new v3.4 BETA 2 you can use to test [here](https://rufus.ie/downloads/).

Feel free to comment on the issue if you find that the problem is still present in 3.4 BETA 2.

closed 2018-11-28 by pbatard

Sopor-, 2018-11-28:
Tested several times with 3.4 BETA 2 and it is working every time i insert the USB stick.