pbatard/rufus #1241: Consider changing the ui

ThiloL, 2018-11-06:
Hi Pete,

[image]

Please consider whether the text fields should not differ more clearly from the buttons.

pbatard, 2018-11-06:
Can you explain what you mean?

I literally have no idea what you are suggesting here...
Also, please try to use a more descriptive issue title than "changing the UI" which is rather vague, and also doesn't really help em understand what you want.

pbatard, 2018-11-06:
What do you call "text fields" in the image you attached? 'FERTIG'? Coz that is not a text field, it's a progress bar with overlaid text.

At any rate, if your request is that 'FERTIG' should somehow be made looking different from 'START' and 'SCHLIESSEN', I'm going to have to reject it. Even if people are unfamiliar with the look and fell of the modern Windows flat UI, clicking on 'FERTIG' will produce nothing and, as soon as they press 'START' and see the progress bar filling, even if they might somehow have gotten the idea that this may be a __very large button__ (which should give them a clear hint that it's not) they will realize that it's s simple status report.

&rarr; Rejected.

closed 2018-11-06 by pbatard

ThiloL, 2018-11-06:
Don't disappoint your German translator. You have no idea how to make different design for buttons? :-|

pbatard, 2018-11-06:
I'm simply reusing the default UI and colour scheme from Windows.

If you are unhappy that Windows 10 "flat" UI makes UI elements hard to distinguish between themselves, you will need to take up that complaint to Microsoft and not me.

For instance, in Windows 7, the buttons and progress bar have "3D" shadowing applied that clearly distinguishes them but Microsoft decided that shadows were too early 2000, so they decided to use a different UI style ("flat") starting with Windows 8.

So, if you are unhappy that this style may confuse users, please log a complaint with Microsoft. As a developer, I am only __re-using__ whatever UI style of the day Microsoft came up with, and the last thing I want is modify the defaults for that style, such as the colour scheme used by button and progress bar face, to make the UI look __different from what Microsoft decided is best for users__ (which you may or may not agree with).

UI matters are __way too subjective__ for me to want to override the defaults. It all boils down to a matter of personal taste, and going one specific way or another is exactly like saying "my taste is better than yours!", which is a ridiculous statement to make. Therefore for better or worse, I will stick with the default colours and shapes for button face and progress bar, which, in Windows 10, happen to be exactly the same, as I am not going to get into a flame war about what scheme would be more preferable for users. It's Microsoft's job to decide that, and I genuinely have better things to do.

ThiloL, 2018-11-07:
This is a default (WPF)-UI. Textbox, Labels and Buttons clearly differ.

[image]

pbatard, 2018-11-07:
Again what you assume is a text box is in fact a progress bar. The control that displays 'FERTIG' is __not__ a text box.

ThiloL, 2018-11-07:
OK. But if I start rufus it says FERTIG. This makes no sense, because there is nothing finished yet.

pbatard, 2018-11-07:
Well, I will advise changing your translation then.

In English it says 'READY' which means "I'm ready to perform an action". If, in German, 'FERTIG' is closer to "done" than "ready", I will advise you to change it to something closer to the meaning of the English translation.