pbatard/rufus #1242: Fake website?

da2x, 2018-11-07:
`http://rufususb.com/`

Appears as the first result for "rufus usb" on Bing search.

Says it's the official website, but that doesn't seem to be the case. Links to a download on box.net. Malware? VirusTotal says no, but someone has gone to a lot of work to setup this site.

pbatard, 2018-11-07:
Not the first "non-official" web site, and probably won't be the the last.

Can't say I care that much about it when they don't appear on google search. Also, if you want to avoid spreading these fake websites, please don't provide a working link to them in this issue tracker.

closed 2018-11-07 by pbatard

da2x, 2018-11-07:
(The link had a zero-width space in the middle. It wouldn't work unless you typed it in manually.)