pbatard/rufus #1245: [Feature Request] Automatically split install.wim to .swm if bigger than 4GB and FAT32

o-l-a-v, 2018-11-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Would be awesome if Rufus could utilize Dism.exe to split Windows Install.img to span  / .SWM, if target is FAT32 and Install.wim is >=4gb.

Log
---
```
<FULL LOG>
```

pbatard, 2018-11-15:
Rufus already has [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) for that. There really is __no need whatsoever__ to use FAT32 for a Windows installation media, when Rufus provides you with the ability to boot NTFS on __ANY__ UEFI machine.

Did you try UEFI:NTFS (which is what Rufus will apply by default if your target is UEFI and it detects a >4GB `install.wim`)? Are you experiencing issues with it?

o-l-a-v, 2018-11-15:
> Rufus already has [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) for that. There really is **no need whatsoever** to use FAT32 for a Windows installation media, when Rufus provides you with the ability to boot NTFS on **ANY** UEFI machine.
> 
> Did you try UEFI:NTFS (which is what Rufus will apply by default if your target is UEFI and it detects a >4GB `install.wim`)? Are you experiencing issues with it?
> 
> If not I will just close this request, as it's a duplicate for a feature that Rufus already provides.

You have to disable secure boot in order for that function to work. So yes, there is a real use case for this.

pbatard, 2018-11-15:
Not really. If you understand what Secure Boot does, then you'll also understand that, as opposed to what __many__ people irrationally think (simply because disabling something that says "secure" in its title has to be "bad", right?) and as explained in the second part of [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Blah-UEFI-Blah-FAT32-therefore-Rufus-should-Blah), there is no risk associated with temporarily disabling Secure Boot during Windows installation.

Especially, if you want to have certainty that your Windows installer will not install anything malicious while Secure Boot is temporarily disabled, then you can rely on Rufus being digitally signed (therefore trusting that the creation process did not install malcious content) and your ISO having been validated to be official (by comparing its SHA-1 against official values, which you can easily do through [adguard.net](https://msdn.rg-adguard.net/public.php?seach=Windows+10+consumer+x64&str=100) for instance), then, unless you have strong reasons to believe that the machine you created the media from is infected, there really is no reason to baulk at the idea that Secure Boot must be temporarily disabled.

Still, if you still want to raise the Secure Boot issue, __PLEASE__ complain to Microsoft, because it's only their __plain abuse of power__ that prevents us from signing UEFI:NTFS for Secure Boot.

Since I still see no real-case for this feature, I will close it.

closed 2018-11-15 by pbatard

o-l-a-v, 2018-11-15:
> Not really. If you understand what Secure Boot does, then you'll also understand that, as opposed to what **many** people irrationally think (simply because disabling something that says "secure" in [...]

I'm not here to judge weather secure boot is a good thing.

The fact is that many businesses require secure boot to be enabled. It's also a requirement for modern client management, like Intune MDM. It would make the lifes of many administrators easier if Rufus could do wim to span / swm on the fly. Our options right now is either do the spanning manually (or convert to .ESD) and use fat32, or disable / re-enable Secure Boot with UEFI:NTFS for every PC you reinstall. Which adds up to a considerable amount of time when talking multiple PCs.

I get if you don't see/ agree  to the potentially added value, and I won't nag more about it. Just thought it would be a nice feature to add to Rufus.

pbatard, 2018-11-15:
Then bring it up to Microsoft.

The __ONLY__ reason you have to disable Secure Boot is because Microsoft have arbitrarily decided, on a very irrational basis (since, strangely, they do allow Linux shims) that anything GPLv3 cannot be signed for Secure Boot.

I hope you can understand that this is a __VERY DANGEROUS PRECEDENT__. What if, in the near future, Microsoft decide that the only way you'll be able to install applications on Windows is through thei App Store, and then apply the same policy of no GPLv3 there (because, after all, if they can justify it for Secure Boot, they can use the exact same arguments to prevent any GPLv3 being delivered by the App Store). Then, because Rufus is GPLv3, that means all you will have gained by not complaining to them is no more Rufus on Windows.

So, by playing Microsoft's game __for short term convenience__, you might very well be reducing freedoms, for everyone, in the long run.

I hope you can appreciate that, as a Free Software developer, I do have to balance user convenience with user freedoms, especially as I deliberately chose GPLv3 for my software. Therefore, where some people may bark at seeing their immediate convenience being reduced, they also need to appreciate that there are principles at stake, that need to be fought for, otherwise __everybody looses__. As such, a developer may have to try to uphold some principles even if it means reducing user convenience.

But if you have corporate power behind you, then, __by all means__, please complain to Microsoft and let them know that their "no GPLv3 for Secure Boot" is hurting you and may lead you to reflect as to whether you want to continue to use Microsoft OSes since their installation is becoming a problem as a result. That's the only way they may start to appreciate that their blatant attempt at abusing their power is not something that people are going to let slide under the carpet...

ThiloL, 2019-01-14:
Hi Pete,

not all users are allowed (able) to disable "Secure Boot". For this users, the "Split-WIM" feature would be very fine.

-Thilo

pbatard, 2019-01-14:
__All__ x86 motherboards do allow disabling of Secure Boot.

If you have other information, I'd really like to hear it, because that would be huge news.

Microsoft actually tried to force manufacturers to produce systems where Secure Boot could not be disabled for ARM, because, since it was a new platform where they could introduce new rules, they thought they'd be able to get away with it, but there was such a backlash at the time that the end result is that since then manufacturers of motherboard make sure that Secure Boot __can__ always be disabled on x86.

Now, obviously, if the issue is that Secure Boot is enabled, but you don't know the BIOS administrative password to disable it, then __clearly__ you should not be booting something on a machine you aren't entitled to administer, let alone install Windows on it...

ThiloL, 2019-01-14:
Hi Pete,

why should users disable and enable secure boot on EVERY installtion, if you could add the Split-Feature?
I'm starting to lose track of your reasoning. You're mixing politics with tech stuff. Microsoft will not change his strategy, even if you change it or not. Or do you believe that?

-Thilo

pbatard, 2019-01-14:
> why should users disable and enable secure boot on EVERY installtion

For the same reason why users may have to enter their BIOS /UEFI to select their USB installation media when installing Windows. Are you seriously suggesting that users should never be asked to enter their BIOS/UEFI settings ever, even when installing an OS, which is the one occasion where they probably want to double check how they are set-up for OS installation.

You're making this a much bigger deal than it really is. __ALL__ x86 users __can__ disable Secure Boot. So there is no blocking issue there. Sure, it __might__ be _inconvenient_ to do so, but then again, it might also help users know how they can proceed to install an OS that hasn't been Secure Boot signed, which may be important for their freedom, so I fail to see what that issue is. For instance, (Free)DOS is certainly not Secure Boot signed so if you don't disable Secure Boot (and enable CSM if needed), then you can't boot DOS. Yet Rufus does create DOS bootable drives, so if we follow your reasoning, should Secure Boot users also not be required to disable Secure Boot EVERY time they want to boot DOS?

IMO, and this is the part you fail to grasp, by pointing users to Secure Boot matters and making them understand that it is perfectly acceptable to disable Secure Boot when the truth of the matter is that, as long as Microsoft has the ability to __arbitrarily__ exclude whichever software they don't like, then users should be educated with regards to their __freedom__ to bypass Secure Boot if they choose so, and boot software that someone, who is not them, has decided they should __not__ have the freedom to boot.

If you think that I am only starting to mix politics with tech stuff, you must not have looked very closely at what I have been doing all these years. I have been a GPLv3 and FSF proponent for a very long time (heck, my GitHub BIO couldn't make this any more clearer), so of course I will fight for users freedoms wherever I see fit. And not letting an entity with dubious motives control what can and cannot be booted on hardware that you own is a freedom that is worth pointing users to, especially if the only price they have to pay (since again Secure Boot can always be disabled on x86) is a slight inconvenience.

Also, I hope you do realize that Windows 7, Windows 8 and pre 1803 Windows 10 users cannot mount 2 partitions at the same time from removable media, which makes your _"(simple) technical feature" not that simple to implement, since you'd have to hide a partition for the other to be mounted, and, if you are a fan of convenience and user-friendliness, __users will be very puzzled on why they can only see a 512MB partition on a drive that is supposed to be much larger and contain a bunch of other files from their Windows ISO__. So please be mindful of not restricting your view to post 1803 Windows 10 usage only. Rufus still has to support Windows 7 and Windows 8 users.

pbatard, 2019-01-14:
That's even worse. Splitting a wim is not a zero-cost operation. It will increase the time it takes to create the media, whereas copying the content wholesale will always be much faster. I though you were talking about the other approach as suggested [here](https://github.com/pbatard/uefi-ntfs/issues/18#issuecomment-451452454), which is much better in terms of media creation time.

One of the established goals of Rufus is to create a bootable media as fast as it can possibly be created. Adding a costly wim splitting operation to the media creation does not meet that goal.

pbatard, 2019-01-14:
Also, just to point to the _"(simple)"_ technical aspect of splitting a wim, you'd have to either __mount__ the ISO to be able to access the wim to split (Sorry Windows 7 users, that means you're out!) or extract the 4GB wim somewhere and then split it (Ouch!).

So we're getting into the _"Why is Windows To Go not available for my system in Rufus?"_ headache_, that aren't so great for users...

ThiloL, 2019-01-16:
There is a arcticle around this issue: https://www.osdeploy.com/blog/windows-setup-fat32-usb-with-+4gb-install.wim 
If I patch the source ISO like mentioned, will this work?

pbatard, 2019-01-16:
Not sure what you're still trying to push. Have you looked at what the script does? It just automates one of the two alternate methods (besides UEIF:NTFS) without bringing anything new to the equation. i.e. __ALL__ the drawbacks are there.

So.,the first thing you can see is that it needs to copy a >4GB to a temp directory. That's a __costly__ operation (and requires mounting of the ISO, which cannot be easily achieved on Windows 7). Even with a fast system and copying onto an SSD, you can count at least 2 mins extra time copying that file. And if you are using an HDD based system, multiply this by 3 or 4. Don't think many users will be happy with that, do you? Especially the part about needing >4GB of space to be taken somewhere on their system (since, if you're going to use FAT32 on the USB, you couldn't even use the USB as a scratch drive). Why should 4GB of system space need to be used when all you want to do is move some data from an ISO on an USB?

And that's just the first part, because once you have the `install.wim` copy you still need to split it, and, even if you copy the parts straight onto the USB, I quote _"The process to split the Install.wim takes a little longer than a simple copy since the SWM files need to be generated, so give it some time"_.

So there again, you're going to be adding an lot of extra time. And all this because some people are being irrational at the idea of __temporarily__ disabling Secure Boot.

I'm sorry, but in terms of speed, space wastage and compatibility (i.e. keeping Windows 7 users happy), UEFI:NTFS is by far the better solution. There's absolutely zero reduction in the maximum achievable copy speed (well, unless you count the writing of the 512KB FAT partition, but since we're applying a pre-existing image wholesale, it's a couple of seconds lost at worst), and the only space wastage you get in 512 KB, on the target USB.

So you may still try to push for lesser solutions to be applied, but as long as they come at great cost for users, you're going to have a lot of difficulty to convince anyone that the inconvenience of __temporarily__ having to disable Secure Boot (which again, you can __ALWAYS__ disable on an x86 based system) vastly outweighs  the inconvenience of suddenly having to require people to waste a lot of extra time and disk space on costly unwarranted operations.