pbatard/rufus #1246: Rufus can't download core.img files for grub 2.03

ybt28, 2018-11-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<i can't download grub 2.03.1 files for writing manjaro 18 iso >

Log
---
```
<Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General USB Flash Disk USB Device' (8644:800B)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001715E9
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (4008706048 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.03.1
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MJRO180'
  Size: 1.8 GB (Projected)
  Uses: GRUB2
  Uses: EFI
Using image: manjaro-gnome-18.0-stable-x86_64.iso (1.8 GB)
Downloading 'core.img' from http://rufus.akeo.ie/files/grub-2.03.1/core.img
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'core.img' from http://rufus.akeo.ie/files/grub-2.03.1/core.img
Unable to access file: 404
http://rufus.akeo.ie/files/grub-2.03.1/core.img was not found - will use embedded version

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Writing Image...
Drive has an unknown Master Boot Record
\\?\Volume{fe380846-e2ed-11e8-a141-8ca982a33f0e}\ was already mounted as F:\
Remounted \\?\Volume{fe380846-e2ed-11e8-a141-8ca982a33f0e}\ on F:\

Found USB 2.0 device 'General USB Flash Disk USB Device' (8644:800B)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x2A1B08CE
Drive has an unknown Master Boot Record
Partition 1:
  Type: Unknown (0xcd)
  Size: 1.8 GB (1973440512 bytes)
  Start Sector: 64, Boot: Yes
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 3854440, Boot: No
Checking for Rufus updates...
Updates: Unable to connect to the internet>
```

pbatard, 2018-11-15:
Please try with Rufus 3.3 (which you can download [here](https://rufus.ie). Because we have decommissioned the rufus.akeo.ie server, Rufus 2.18 and earlier cannot access the server.

But anyway, since you are writing in DD mode, the download failure is irrelevant anyway.

For the record, we only have files for GRUB 2.03 for the time being, and not GRUB 2.03.1. On a side note, the Manjaro people are __SERIOUSLY__ starting to get on my nerve, first by ignoring [an easy to fix bug that I opened 6 months ago](https://bugs.manjaro.org/#ticket/zoom/17934), that would allow Manjaro users to run Rufus in ISO mode, and second, by once again by breaking the common rules of versioning, and slapping a `.1` to a GRUB version, just so they can do their own thing.

If 2.03.1 is a __real__ thing, and not something that Manjaro pulled out of thin air __AND__ it turns out that the default GRUB 2.x provided by Rufus is incompatible with what Manjaro needs, I may try to add a GRUB 2-03.1 binary on the server. But I can only do that once Manjaro have fixed support for ISO mode, which they really don't seem eager to do, so it looks like that's not going to happen for a while...

ybt28, 2018-11-16:
I first tried WITH Rufus 3.3 both iso and dd mode but still to no avail.. I
guess the problem is with manjaro. Thanks though for replying

But can I please get the files for GRUB 2.03

pbatard, 2018-11-16:
Again, if you are writing in DD mode, which is what your log shows, you don't need these files.
And Manjaro does not work in ISO mode (the alternative option from DD mode), so even if I gave you GRUB 2.03.1 files (which I don't have), it would be pointless, because __these files are only used in ISO mode, which is INCOMPATIBLE with the current Manjaro__.

In other words, you don't need these files.

I will therefore close this issue.

closed 2018-11-16 by pbatard

ybt28, 2018-11-16:
Oh OK Thanks