pbatard/rufus #1247: Access to the device is denied

Hoho4227, 2018-11-17:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

JonnyTech, 2018-11-17:
#1071

pbatard, 2018-11-17:
> `ABORTED: Cannot use an image that is located on the target drive!`

That's your problem.

You can __NOT__ create a bootable USB from an ISO that resides on the same USB. It's exactly like sawing the branch you are sitting on, because, __of course__ Rufus will erase __ALL__ data from that drive during the creation process, so, if it was to let you proceed, it would have nothing to create your image from.

You need to move your ISO somewhere else than on the USB.

closed 2018-11-17 by pbatard