pbatard/rufus #1248: A problem of downloading ldlinux.sys and in updates

NoviXPT, 2018-11-17:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Here im leave the log of the program:
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - cancelling

Log
---
```
<FULL LOG>
```

JonnyTech, 2018-11-17:
Please supply a full log, not just what you think is necessary. Are you running Windows XP?

ibuprophen1, 2018-11-17:
I just selected the link and it worked instantly... 

~Ibuprophen

pbatard, 2018-11-17:
Duplicate of #1232. Nothing to do with Rufus, you need to make sure your version of Windows is up to date when it comes to accessing HTTPS sites.

For instance, if you can't download https://rufus.ie/files/syslinux-6.04/ldlinux.sys __when using Internet Explorer__, then it means this is an OS issue.

Follow the advice from #1232, and next time, please actually search the issue tracker with the error message you got (if you do that #1323 is the first issue you'll get) and paste a full log.

Closing.

pbatard, 2018-11-17:
Also, just in case you wonder, because we had to switch web servers, the check for update is going to remain broken until we release Rufus 3.4, which fixes compatibility with the new server. But this is unrelated to your issue.

closed 2018-11-17 by pbatard

sheddup, 2019-01-08:
>Nothing to do with Rufus

You could always detect the download issue then take actions from there(like alerting the user or using an alternative download method)  
My connection has no problem downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys but the rufus application just sits there 'downloading ldlinux.sys' very frustrating to experience.

pbatard, 2019-01-08:
> You could always detect the download issue

I do. It's mentioned explicitly in the log with a message that states that the SSL library being used by the system is too old.

> then take actions from there(like alerting the user or using an alternative download method)

I'm sorry but I'm not going to spend a lot of time on platforms (Windows 7) that is going to reach End of Life soon (Jan 2020) and that simply happen not to be able to download files because important system updates, such as the ability to access __modern__ SSL sites were not applied. Spending time fixing Windows 7 obsolete behaviours is just not a good use of anybody's development time.

Also, if this is not your issue, then please provide your full log, because, as far as I'm concerned, since you don't provide anything, I'm going to assume that you are running into the Windows 7 HTTPS issue like OP.

sheddup, 2019-01-08:
This is on Windows 10
Log - https://privatebin.net/?bb2f535d5db3b17f#MJSNk5zuwSOXIN8jgquePnZ6cqy4+EHaYQnbBI8rpDs=

From what I can see the log shows two URLs the first URL is a 404 - https://rufus.ie/files/syslinux-6.04/20180907/ldlinux.sys the second URL works for me in Firefox I never tried in Internet Explorer/Edge because I removed that when setting up Windows.

pbatard, 2019-01-08:
> This is on Windows 10

Then this is not the same issue. Unless you were seeing ```Unable to send request: This system's SSL library is too old to be able to access this website.``` in the log, you shouldn't have piggybacked on this issue as your problem is completely different.

> the first URL is a 404 

Yes, that is normal. Rufus always tries two URLs, the first one, which is constructed according to the version used by the ISO, is in case there is a need for a custom version of Syslinux to be used, which isn't the case here, so, since we didn't place a custom `ldlinux.sys`, this will result in a 404. Then, if Rufus finds out there's no custom `ldlinux.sys` required, it attempts to download the vanilla version, as released by the Syslinux developers.

> `Unable to send request: The attempt to connect to the server failed.`

__This is a pure environmental issue on your side__

Something in your environment is preventing Rufus from connecting to the server. It doesn't matter whether Firefox or other browsers work, because it is perfectly possible, through (mis)configuration, to prevent an application from being able to download from the internet while letting others do so. Especially, granting a download exception for browsers and preventing all other applications from being able to connect to the internet is commonly used in corporate firewalls.

I hope you can understand that if you can download the file through Firefox but Rufus tells you it cannot, whereas people using vanilla versions of Windows 10 don't have this problem at all, then it's entirely up to you to figure out the issue and fix it. 

Especially, considering that you indicate that you removed Internet Explorer, which I assume also includes the STANDARD LIBRARIES that application can use to download files (WinInet and so on), I think your problem can be explained very easily and I hope you can realize that it would be crazy to ask a developer to embed custom internet libraries in their app, just so that the handful of people who go heavy handed on removing system tools and libraries, without fully measuring the consequences of what they are doing, have an easier life.

You screwed up your system, you fix it.

sheddup, 2019-01-08:
I didn't screw my system up I forgot to add rufus to my firewall whitelist, my bad. 
I still think you should ALERT the user that a download failed, instead of just passively logging.

pbatard, 2019-01-08:
Did you just remove your earlier comment with the log? Please __don't do that__! I had to post it back with your entry, so that __OTHERS__ running into the same error message can find this issue and maybe realize that they needed to whitelist Rufus, if that is their issue.

Romarain, 2019-02-14:
> Duplicate of #1232. Nothing to do with Rufus, you need to make sure your version of Windows is up to date when it comes to accessing HTTPS sites.

We're not supposed to know what technology you use in your software. And never an operating sofwtare has to be updated to run https : it's only a browser/software matter (protocole).

> For instance, if you can't download https://rufus.ie/files/syslinux-6.04/ldlinux.sys **when using Internet Explorer**, then it means this is an OS issue.

I can access it on my Thinkpad R400 with Windows 7 on Chrome,but still, Rufus can't access. 

> Closing.

No, you won't close anything like that because you just want to avoid thinking.

> I'm sorry but I'm not going to spend a lot of time on platforms (Windows 7) that is going to reach End of Life soon (Jan 2020) and that simply happen not to be able to download files because [...]

And we won't waste our time with an arrogant developer who has a superiority complex and think that everybody on the planet has the minimal hardware required to run Windows 10. You' clearly lack of IT skills and experience, and seem unable to adopt a retro-compatibility philosophy.
Just imagine that when upgrading to USB3.0, manufacturers had say "Now we won't waste our time with people still having too old hardwares, go buy a new PC or stfu".

You're completly megalomaniac and you won't excuse anything by assuming that everyone had deleted some IE and obscure libraries. I don't know about HTTPS libraries, but if you suppose that Chrome and Firefox developers had act like you do and assumed that Windows would proved them all the libraries they needed, you're completly out of the reality. **Developers HAS to embed technologies they want to use, if OS don't.**
W7 doesn't offers HTTPS protocols/libraries ? **Add it into your soft, or don't use HTTPS. Period.**

Now if you think I'm gonna upgrade all my laptops to Windows 10, not only you're dreaming, but I'm gonna use an other software that Rufus. Bye bye mister arrogant.

pbatard, 2019-02-14:
> I can access it on my Thinkpad R400 with Windows 7 on Chrome,but still, Rufus can't access.

Try the same with Internet Explorer on Windows 7.

Chrome and Firefox embed their own SSL libraries, which is why they can access the site. But IE, like Rufus, relies on the OS ones, so it can't.

And again, once people get their Windows 7 up to date, they can access the web site through IE or Rufus.

> No, you won't close anything like that because you just want to avoid thinking.

I don't want to avoid thinking. What I want to avoid is have to develop a full blown browser library, that adds __custom__ support for up to date SSL, just so that the people __who have not kept their Windows 7 up to date__ can continue to use an outdated OS.

> and think that everybody on the planet has the minimal hardware required to run Windows 10.

That's not at all what I am advising. Windows 7 users can absolutely access the SSL downloads in Rufus, __as long as they have applied all the current Windows updates__. I tested it, others confirmed it, and, if you read what I stated above as _"Windows 7 users should just upgrade to Windows 10"_, then you are interpreting things wrong.

The _" because important system updates, such as the ability to access **modern** SSL sites were not applied"_ is meant to convey exactly that: Provided you have applied all the Windows system update available, you should have no trouble downloading files from the Rufus servers in Rufus.

So, instead of going on a rant about "arrogant developers", you may first want to validate whether your claim ("Developer is asking everyone to upgrade to Windows 10") has any shred of truth.

> you won't excuse anything by assuming that everyone had deleted some IE and obscure libraries.

Because I have __ACTUALLY TESTED__ my software on fully up to date vanilla Windows 7 machines (not on VM but real hardware), and found no issue whatsoever when downloading Syslnux files from the current Rufus servers, and I also know that the vast majority of Rufus users on Windows 7 also have no issue, then I can only logically assume that, if someone running Windows 7, and who has applied all the system updates from Microsoft, is still unable to download in Rufus, it means that they must have altered their system.

Especially, the error reported by one of the users above (`Unable to send request: The attempt to connect to the server failed.`) has nothing to do with the SSL issue, since the SSL issue generates a very explicit message, and is clearly something that does not apply to every Windows 7 users. So I am rightfully treating it as a pure environmental issue, as in all my testing on vanilla Windows 7 machines, I have never gotten that issue and have always been able to download files from the Rufus servers.

>  **Developers HAS to embed technologies they want to use, if OS don't.**

And in this case, if you __DO__ have an updated Windows 7 system, the OS does. Which is why Rufus will continue to rely on the OS to provide SSL functionality, instead of embedding its own, especially with all the risks and drawbacks it would create. Ergo, while your assertion is correct, your premise is completely wrong.

> W7 doesn't offers HTTPS protocols/libraries ?

W7 does, and they work just fine for accessing the Rufus servers unless you failed to apply expected Windows updates. Again, I have __VALIDATED__ this repeatedly.

Next time, before rushing into a completely erroneous conclusion about what developers expect their users to do (because, ALL I am expecting here is for Windows 7 users to have an up to date version with all the major Microsoft updates applied, in which case there is no issue at all), you may want to spend a bit more time trying to understand what the developer is actually stating, and maybe give rest to the idea that all developers are arrogant megalomaniac pricks that don't give a damn about users running on older but still supported OSes.

> Bye bye mister arrogant.

And a good day to you, mister _incorrect assumption_.

Hopefully, the next people who comment on this thread will take some time to actually read what's being said, and not rush themselves into apoplexy from preconceived and utterly false ideas about what a developer is actually requesting users to do, which is only to use an up to date OS, be it Windows 10, Windows 8.x __or Windows 7__.

Romarain, 2019-02-23:
So, let me clear one thing :
- You come here with absolutely no other explaination about YOUR assertions on W7 environments blablabla, and just say that it's a W7 related problem to the people BEFORE my arrival here.
- Given what you said, we can only understand that Wè would have some problem, and that youdon't want to do ANYTHING.
- I corrected you about this lazy behavior, and spotted your arrogance.
- And now you come with a full set of explaination that seem, about W7 updates, to be coherent and understandable.

So you won't make me believe that I was in fault during the communication, nor that I made any logical mistake. YOU seem to lack of any logical chaining comprehension, assuming that everybody is in your head and that you can just make authoritarian assertion with no more explaination than "I won't wast my time with people staying on too old system" or "you should have made something".

Your explanation about updates are right : if people experience HTTP connection problem with your software, it should be or that updates are missing (I believe you, while not verifying it actually) or that updates are installed but that the only cause can be local modification.

That being said, you didn't provided that explanation at any moment before I spotted your arrogance, **wich by the way is still more valid even if you try to cover your trace with retroactive justifications.**

Now that's a time for me to do a fancy conclusion like yours in your last comment : hopefully, people will carefully read what has been said on this present thread, and understand that I'm fully right about yourself and that I wasn't rushing. Now if you have problem in communication, just check if you're not actually the one who is rushing and lacking of pedagogy and details because of a certain lack of patience.

And please, don't be sophist by using your last comment as a so-called "proof" of your patience,when there is some paragraphs above that prove that you weren't. Don't reverse the facts chronology.

Bye.

PS : next time, be sure not to enter a discussion with someone by using that tone, without verifying that your speaker has the cognitive ability to notice contradictions and sophism. You made the mistake of hidding your fault and assuming that your speaker was dumb, and **that's** a wrong premisse ;)

pbatard, 2019-02-23:
>  about YOUR **assertions** on W7 _environment_, and just said that it's a W7 related problem (period) to the people BEFORE my arrival here.

Assertion? Nope.

OP reported `Unable to send request: This system's SSL library is too old to be able to access this website.` which isn't something that can really only happen on Windows 7.

So, while replying to that, __AND THIS IS THE IMPORTANT PART__, I pointed to #1232, which provides further details about the issue and indicates, clearly, (but feel free to dispute that if you want) that you can very much get W7 to work if you just install system updates.

>  Given what you said, we can only understand that W7 would have some problem with HTTPS/SSL libraries, and that you don't want to do ANYTHING.

Bullshit. Don't confuse _"can only"_ and _"might, but only if one __chooses__ not to look at the explicitly linked issue, as they should (especially if they plan to go on a tangent and attack the developer)"_.

If a developer links to an issue and says this is a duplicate of this one, you are expected to look at the duplicated issue, since, __OF COURSE__, you should assert that said developer will not want to repeat themselves over and over again when they already provided details. This is even more true if the first thing you're gonna settle on is to attack the developer for being arrogant. If you have an __accusation__ to make, you'd better make doubly sure that this accusation stands by doing your research.

So, clearly, now you're trying to save face about the fact, because you seemingly decided that looking at the original duplicate issue was beneath you even as you were readying yourself to throw epithets at a developer, you interpreted this whole thread wrong.

Nice attempt, but that's not gonna fly.

>  I corrected you about this lazy behavior, and spotted your arrogance.

LOL. Who's arrogant here? The person who, more than one week after I indicated that they completely misread the situation, are still trying to save face by trying to pretend that it wasn't their fault they misread it, and are accusing the other party of behaviour that simply cannot apply? Or the person who indicated at length that they didn't need to fix an issue that has been demonstrated not to require any fix besides applying system updates? How's that arrogant? Or even lazy?

Be very very mindful of trying to bullshit your way into continuing to want to accuse someone of behaviour, simply because you don't want to have to retract an original assumption that has been demonstrated wrong (and which you also now agree was wrong, even if you are trying to use the all too classic excuse of _"Well, even if I was wrong, it's someone else's fault"_, which I will dispute by simply indicating that, the only way you could ever have a chance of pushing that kind of excuse is if I hadn't clearly pointed to #1232, which I did, in the very first sentence of my very first reply.

> And now you come with a full set of explaination that seem, about W7 updates, to be coherent and understandable. But you try to use it to reverse the situation as if the future was replacing the past.

Why do I get the feeling that you still haven't read #1232. If you had, then I fail to see how you could write yet another bullshit statement.

Remember, here's how things went.

1. OP reports an issue and indicates that they are getting the error `This system's SSL library is too old to be able to access this website.`
2. In m first reply, because I am aware that this is something that can only occur on W7, is to indicate that this is a duplicate of #1232 and tell them that they need to make sure that _"their version of Windows is up to date when it comes to accessing HTTPS sites."
3. Issue is closed.
4. Months later another completely different poster decides to piggyback on this issue (__which__, by the way, is the precise reason why I always frown on this), to report on a __completely different issue__ that they are observing on Windows 10. 
5. After I investigate and comment on this other issue (with the quirp on not wanting to go out of my way for Windows 7 when we people who keep that version of Windows up to date will not run into it and that platform will cease to be officially supported in 2020 anyway) __the poster for that second unrelated issue decide to remove their posts__ (which is why you see me quoting stuff that doesn't appear to follow, and that should clearly indicate that some content was removed even as I had no involvement in removing it).
6. You now pick on this thread, and decide to interpret it as you wish (those arrogant developers!), instead of looking at it as it is.

> So you won't make me believe that I was in fault during the communication, nor that I made any logical mistake.

Doesn't matter what you believe. People can look at facts and decide for themselves.

As far as I am concerned, you completely failed to pick on 2 elements you should have picked (#1232) and quoting of a post that no longer seems to be there, that should have let you know that your initial interpretation of what I was saying might not be what you assumed it was.

And now you are trying to raise hell in a feeble attempt to point that how it wasn't really your fault (but somehow mine) and how it was natural that you'd failed to pick on the elements I pointed to, rather than admit your mistakes. That's not very mature behaviour, if you ask me.

> YOU seem to lack of any logical chaining comprehension

[... cut ...] ), even if it goes in the complete opposite direction of what I did mean.

Personally, I'd much rather let people judge for themselves, by pointing them to clear evidence, rather than fabricate a bold statement of what I'd like other people to agree with, especially if that statement is meant to include disparaging blanket qualifiers about the general behaviour of another person (which is never a good sign when someone is trying to others that they have both the logical and moral high ground)...

Romarain, 2019-02-24:
> Assertion? Nope.
> 
> OP reported `Unable to send request: This system's SSL library is too old to be able to access this website.` which is something that can really only happen on Windows 7.
We're not supposed to know that. You may know that, but we're NOT in your head.
That's why I ponted out a pedagogic problem, and an arrogant mentality of yours.

> So, while replying to that, **AND THIS IS THE IMPORTANT PART**, I pointed to #1232, which provides further details about the issue and indicates, clearly, (but feel free to dispute that if you want) that you can very much get W7 to work if you just install system updates.

No, as I said, you didn't specified if we have to update from W7 to W10 (wich can be seen as an update) or just update W7 to its full set (wich can be also seen as an update) : and as I said, it constitues a semantic ambiguous meaning that you used for your sophism.

**You won't force me to repeat that one more time.**

> Bullshit. Don't confuse _"can only"_ and _"might, but only if one **chooses** not to look at the explicitly linked issue, as they should (especially if they plan to go on a tangent and attack the developer)"_.
I'm sorry but you don't mean anything here, and just get bogged down into your sophism about "us having to be in your head and to have your insights". I perfectly spotted that your answer were NOT giving any clear information BEFORE I spotted your arrogance, and you keep trying to say that we made a mistake, and that I made one. You play with words like "can only" or "might", but what makes you believe there is a difference that I should have know, is PRECISELY what I explained you we cannot know.

So, YOU have ot do some efforts to make you be understood.

> If a developer links to an issue and says this is a duplicate of this one, you are expected to look at the duplicated issue, since, **OF COURSE**, you should assert that said developer will not want to repeat themselves over and over again when they already provided details in the original issue. This is even more true if the first thing you're gonna settle on is to attack the developer for being arrogant. If you have an **accusation** to make, you'd better make doubly sure that this accusation stands by doing your research.

Wrong : I don't have to suspect that there would be any more information in the so-called duplicated issue, nor to stop reading the actual thread. **However, I read something making me understand that W7 was guilty of being too old, and that a developer won't do any effort to adapt his software.**
Thus, this is enough to not go on a previous thread.

> So, clearly, now you're trying to save face about the fact, because you seemingly decided that looking at the original duplicate issue was beneath you even as you were readying yourself to throw epithets at a developer, you interpreted this whole thread wrong.

And you won't save the face by refusing to see how you're acting like an arrogant individual on this thread, **by having clearly given us enough informations to conclude things.** We don't have to go for further informations when you're clearly saying us to update W7 to W10 with semantic ambiguity and insinuations.

> Nice attempt, but that's not gonna fly.
Nice try, but you won't reverse the situation, Mr sophist !

> LOL. Who's arrogant here? The person who, more than one week after I indicated that they completely misread the situation

No, you, who are completly irresponsible and childish, and clumsily try to hide his lazyness when developping, wich is being transffered on his answers on forums. 

> ... blablabla... Be very very mindful of trying to bullshit your way... blablabla

> Why do I get the feeling that you still haven't read #1232. If you had, then I fail to see how you could write yet another bullshit statement.
As I said, we don't have, given what you said in this actual thread. You keep failing in logical chaining understanding, and chronology or even "mind theory" (knowing what others can know).

The rest of your comment is a repetition of what has been previously said, as a loop.
You fail understanding that we don't have to look an other thread BEFORE reading the actual one and your answers + your answers are giving us informations that can make us to conclude that W7 is lacking HPPTS/SSL libraries and that we have to update it to W10 (your insinuations).

**I think people will see more clearly what kind of individual and developer you are, and that your tens of paragraphs of repetitive information and constant looping like a child trying desesperatly to defend himself will play against you.** I don't have anything more to add to the conversation wich is constant ad-hominem coming from you (assuming that we're too fool and incompetent to read the present thread + have to read an other one before reading this actual one + have not to read your comments here or have to interpret them the way you want), and chronology reversion.

Next time, stop being arrogant by assuming that people are too stupid to see your lazyness in development sweating in your hints about updates. Just admit you failed to develop correctly your software **or** to make you understood (wich can happen when there is a psychological slip). I will make the discussion shorter and you won't pass for being arrogant.

Romarain, 2019-02-24:
Not mentioning that you forced all original thread titles like _"impossible to download ldlinux.sys"_, wich a are correct bug tracking title related to what happen to users, to be renamed _"This system's SSL library is too old to be able to access this website"_ (like the thread [#1232](https://github.com/pbatard/rufus/issues/1232)), **wich are more oriented titles corresponding to YOUR opinion on the problem.**

That's a sort of controlling I saw on many forums, coming from people who can't accept that **users aren't technicians**, and who can't accept **their own fault** in the bug occurence.

pbatard, 2019-02-24:
Congratulations.

I think people reading your last reply will be able to make their mind indeed.

Saves me a lot of time demonstrating once again that, before you start to throw epithets at someone, on account that you wrongly assumed they were stating something that they weren't going to do, as well as accusing the other party of exactly the kind of behaviour you are displaying (ad hominem much?), you might want to cool your head down a bit and actually peruse through facts first.

The only consolation I have is that, even if you don't want to admit it here, you'll probably learn from your mistake and thread a bit more carefully next time you want start going on a tangent about how someone is lazy, arrogant, childish or any other nice ad hominem qualifier you feel like slinging around, and maybe pay a bit more attention to what the person you have decided to clash against actually wrote (especially prior to the statement that appears to irk you). This way, you will avoid having to make the problematic choice of either admitting that mistake as your own, or trying to pin the blame of your misinterpretation onto somebody else...

Romarain, 2019-02-24:
Thanks ! And I will keep being sure to spot arrogance from developers in their attempt to say that we should migrate to the last version of an OS, and then go backwards to try to cover their traces. I will also keep working on redirecting people on **right entitled threads** like this _present one_, and encourage them to **not** be ashamed of just naming things in relationship with their own problem just because a lazy dev wants we know the origin of a problem before its real diagnostic.

> PS: Renaming an tracker issue title to the exact error message users see in the log when they encounter that issue is the logical way to ensure that, when... _blablabla_

Hahaha, this is amazing ! :D You keep trying to reverse the chronology and run off your mind-theory deficiency.

pbatard, 2019-02-24:
Anything else you want to add? The more you do, the more interesting it becomes, and I would indeed very much like for other users to be pointed to this issue as a very insightful cautionary tale of how not to behave, once you realise you have misinterpreted something.

Romarain, 2019-02-24:
Do you assume you're forcfully right, and that we have to go on a public trial just because people will necessarily be on your side ? I also think that your behavior is not a good example. Do I call for a popular revenge so far ?

pbatard, 2019-02-24:
Please don't stop now. Keep going.

Romarain, 2019-02-24:
I won't, I'm not like that ;) This is not supposed to be an harrasment. 
Everything as been said.

pbatard, 2019-02-24:
Oh, but I insist. If only to figure out who you believe is harassing whom, and even more so as I think I can turn this into a very educational exercise when it comes to letting people understand what an ad hominem is, and how to spot it.

Besides, I don't have much of anything better to entertain myself with right now and I sure wouldn't mind seeing you demonstrate once again that you don't appear to understand what an ad hominem is as well as how much further you are willing to prop your _"Even if I was completely wrong to assume a developer was saying Windows 7 users should just update to Windows 10, not only it wasn't my fault at all to have missed the obvious hints, but furthermore despite all __evidence__ (which I will oh-so-conveniently qualify as 'retroactive whatever', by asserting that the developer is able to completely turn the meaning of what they did or wrote previously to mean something completely different) I will still stand by my earlier misinterpretation and feel entitled to throw more ad hominems."_

Ergo,  just in case this needs to be spelled out to you:
* Telling someone that something specific they are doing, after pointing exactly what, is _"no very mature behaviour"_: Not an ad hominem (since this is wholly different from stating "X is circumstantially immature". Likewise _"first foray into mature arguing territory"_ is not an ad hominem in case you wonder.
* Qualifying incorrect statements, as "bullshit", not an ad hominem (remember, for this to be an ad hominem, it would need to be interpretable as X circumstantially suffers from general flaw Y).
* Stating that you misinterpreted a statement, clearly nothing to do with ad hominem (It would be one if I was saying _"You are, in the whole circumstance of this issue, incapable to properly interpret most of what you read"_).
* Telling someone that they are throwing a lot of epithets at another person, when this can be easily verified as fact (see below), not an ad hominem.
* Now, the only thing that should look close to an ad hominem in what I wrote is when I did retort to someone who repeatedly called me arrogant (now __that__ is an ad hominem, though, realistically, it is probably closer to name calling, but I will give you the benefit of the doubt by assuming that you are intending it to apply only to my qualities as a developer) by returning the compliment while pointing to a very specific behaviour of them that looked quite arrogant to me (this could be considered as a _tu quoque_ ad hominem). But then again, this was formulated as "Who's arrogant?" "How's that arrogant?" rather than an explicit assertion, so I'll leave others to debate whether this can technically be considered as one.

Of course, it's hard to say exactly which statements of mine you chose to interpret as ad hominem, but these are the most obvious candidates I can think of, so I'm highlighting those.

Unfortunately however, and from my experience, it looks like a lot of people confuse disparaging qualifiers applied to statements they made, after (or before) they are also separately deconstructed for their intrinsic flaws through other means, and where the qualifier comes as a shortcut, as ad hominem.

Now, if I wanted to, I could go over all the statements you made, that were devoid of any specific context that can only be qualified as either name calling or ad hominem. But there are simply too many of them so, as an exercise to the interested reader, I'll let them search for "arrogant", "lazy", "child/childish" and a bunch of terms which, when repeated in the volume you seem to be willing to use them, I would assert could also qualify as harassment.

Thus, with the above said, I can at last indicate that I a good chuckle at your _"constant ad-hominem coming from you"_, from earlier today, especially as it is itself an obvious ad hominem...

Therefore, I can only hope that you will feel inclined to post some more replies, for everybody's enjoyment. After all, you wouldn't want to let me have the last word, would you?

Romarain, 2019-02-28:
>Oh, but I insist.
FAIL ^^. You were the guy insinuating, with _"Please don't stop now. Keep going"_. that I was trolling and that I should stop. Now you see ? You're the one restarting the process.

>If only to figure out who you believe is harassing whom,

You are actually the one harrassing the other because of the previous point. 

> and even more so as I think I can turn this into a very educational exercise when it comes to letting people understand what an ad hominem is,

You always assert that people will understand X, and just repeat the last word used be your speaker (here, the word "ad-hominem"). I bet you don't understand the concept you think people will understand at your place, and this to keep your mental domination on them and keep patronizing them like you did at the very begining of this thread (by assuming that people were stupid not to know what was the cause of the problem + insinuating things that you would blame after).

>Besides, I don't have much of anything better to entertain myself with right now

Arrogant sarcastic troll spotted ^^. You're clearly saying "I don't care of what people say, I will just shut their mouth by ignoring anything they say, this collapsing their reasoning in order to win".
That's basically how sophists an assholes proceed.

> and I sure wouldn't mind seeing you demonstrate once again that... _blablabla_

LOL. You say that you don't care of what others say, and then you assume that we will read you ?? Hahaha.

As I said before, I won't enter your loop again. You're constantly ingnoring what we say, and repeat the same mistakes with badly syntaxed phrases as :
>Qualifying incorrect statements, as "bullshit", not an ad hominem (remember, for this to be an ad hominem, it would need to be interpretable as X circumstantially suffers from general flaw Y).

... wich are unreadable. Not mentioning that anyway, Nobody has to read you, given your level of contempt.

pbatard, 2019-02-28:
> You were the guy insinuating, with _"Please don't stop now. Keep going"_. that I was trolling and that I should stop.

Once again, you're interpreting things wrong. Trolling was never mentioned, and the fact that you interpret this sentence as an accusation of trolling tells a lot.

At this stage, and this appears to have been the case for some time, you have to __fabricate__ hidden (and unprovable) explanations, or throw character traits, to address everything that has now been presented to disprove your point and especially your continued misinterpretation of my initial statement. Of course, I have no doubt that you firmly believe that you are in the right to venture these __hypotheses__ out of the blue but, unfortunately, that is not how arguing works. But I'll go into more details about this below.

So, no, I was not accusing you of trolling (especially as the behaviour I was inviting you to pursue doesn't match my definition of "trolling"). But I suppose that, at this stage, it isn't much of an ad hominem to state that you'd continue to insist that, regardless of what the other party indicates they meant (or didn't mean), you will continue to insist that only your _interpretation_ should be taken as gospel,

> You are actually the one harrassing the other because of the previous point.

Thanks for answering that question.

> as soon as you think you obtained the silence, you steal the last word :D

If I really wanted to obtain the last word, wouldn't you agree that the much easier thing to do would be to first ban you, and then post one last time? After all, it is in my power to ban whoever I want from this issue tracker.

> But responding to someone on a forum can't be perceived as an harrassment, just based on that last-word effect. Or else, I could accuse you to harrass me as well.

Agreed. On the other hand, continuously accusing someone of being "arrogant", "lazy", "pretentious" and so on __is__ harassment.

> Moreover, that last-word fear strategy is a childish method

I don't know. It seems to work exactly as I wanted so far... 😄 

> You always assert that people will understand X

Yes. Sue me for asserting that people are generally smart and can read what is presented to them... rather than jump to the wrong conclusion.

> Arrogant sarcastic troll spotted ^^. **You're clearly saying _"I not in a reasonable process, I just don't care of what people say, I will just shut their mouth by ignoring anything they say, this collapsing their reasoning in order to win. And that's gonna be a game for me, giving me much pleasure."_.**

And here we have exactly the kind of reply I was hoping to see from you again.

As I stated earlier, your lennthy reply from a few posts prior demonstrates everything one needs to know about your "arguing" process. So there's not much else for me but to just enjoy watching just how far down you are willing to go. I am both genuinely interested and entertained.

> You're constantly ingnoring what we say

Yet I am quoting you extensively...

> This guy is amazing. What a pretentious !

Name calling. Again, never a good way to try to demonstrate that your arguing is the soundest.

> nobody has to read you, given your level of contempt.

Here is is. A perfect example of ad hominem

First, let's start with the definition from [Wikipedia](https://en.wikipedia.org/wiki/Ad_hominem):

* _Ad hominem (...) is a fallacious argumentative strategy whereby genuine discussion of the topic at hand is avoided by instead attacking the character, motive, or other attribute of the person making the argument, or persons associated with the argument, rather than attacking the substance of the argument itself_

If you know your latin, the term (and the Wikipedia definition) makes it pretty clear that, as opposed to your restricted definition, an ad hominem is when you resort to attacking the person formulating a statement, rather than the statement itself, usually by attributing them a disparaging general characteristic, with the goal to present that (usually also alleged) negative trait as a motive to refute the statement wholesale.

Now, let's say for a minute that I was contemptuous. A contemptuous person can still assert that `1+1=2` or `1+1=3`. Thus, unless you actually go for rebuttal of the point they are making, should you believe that it is incorrect, you are not doing yourself any favour with regards to winning the argument by trying to get the audience agree with the fallacious _"This person is contemptuous therefore whatever they state is wrong and should be ignored"_

See how that works? Or do I need to break it down further?

Any time you throw in a "arrogant", "sarcastic", "trolling", "childish", and so on (i.e. any time you are attacking the other party's character), as your main/sole argument to "disprove" my statement, you are not doing yourself any favour.

> First,it lacks a verb in the first phrase ^^ ("clearly nothing to do" isn't syntaxically valid).

Well, it's kind of weird you would pick on the 3rd item of a list that all display the same syntactic behaviour as all the others. The list is organized as "quote: clarification as to whether it qualifies as an ad-hominem or not". Please ask a English native speaker whether they think that a verb was required where you say it was, coz, while I'm pretty sure I may have been using weird phrasing elsewhere (which again, if that's what you need to bring to try to refute an argument won't carry much weight), this isn't re [... cut ...]
And when you realized that it wasn't working too well, you had to resort to pulling a completely unverifiable reason out of thin air (a.k.a. an __insinuation__), as to why the developer would be angry with users of that platform.

> Now that's me who is enjoying this debate ☺

I'm sorry, but when someone has to call on name calling, ad hominem, hypothetical "intuitive" ideas or insinuations of what the other party __might__ have secretly been doing, (alleged) syntax errors and other argumentative fallacies, there's not much of a debate.

But please don't let that refrain you from posting further.

Romarain, 2019-02-28:
>Once again, you're interpreting things wrong. Trolling was never mentioned, and the fact that you interpret this sentence as an accusation of trolling tells a lot.

Yeah, it tells that you constantly use insinuations and as usual, run of them afterward. :dagger: 

>At this stage, and this appears to have been the case for some time, you have to **fabricate** hidden (and unprovable) explanations

No no no, YOU don't answer to what I write. I perfectly explained that you reversed chronology and mixed ad-hominem with other things. 

So all these paragraphs from :

>or throw character traits,

to 

>you will continue to insist that only your _interpretation_ should be taken as gospel,

are null and have no value.

>If I really wanted to obtain the last word, wouldn't you agree that the much easier thing to do would be to first ban you, and then post one last time?

You have no right to ban anyone, and you're trying to obtain it by forcing others to be perceived as trolls. I also perfectly spotted that in my last comment.

>On the other hand, continuously accusing someone of being "arrogant", "lazy", "pretentious" and so on **is** harassment.

No. It's a critic. And you seem to have a problem to accept it. Moreover, you also provide critics, and use adjectives too.

>Yes. Sue me for asserting that people are generally smart and can read what is presented to them... rather than jump to the wrong conclusion.

They didn't, given all what I explained to spot your constant sophism and reversion ;)

>And here we have exactly the kind of reply I was hoping to see from you again.
As I stated earlier, your lengthy reply from a few posts prior demonstrates everything one needs to know about your "arguing" process. So there's not much else for me but to just enjoy watching just how far down you are willing to go. I am both genuinely interested and entertained.

So, you're happy to see me spotting the arrogant troll you are ? :D That's perfect, I didn't need more !

>Yet I am quoting you extensively...

And ignoring what is said...

>Name calling. Again, never a good way to try to demonstrate that your arguing is the soundest.

We have the right to name you and to use adjectives to describe you, especially if we make the effort to prove it. And you have no right to ignore what we say, then to answer "name calling, blablabla that's a good demonstration of...", **because we can just ignore that demonstration as you ignore ones of others.** So, keep fapping on those "name calling".

>Here is is. A perfect example of ad hominem

That can't be an ad-hominem, given that if you're using contemps by using hints and insinuations and then balming others to udnerstand this, this childish behavior has to be described and the word "contempt" is one of the word that correspond to that behavior.

Plus, I use argumentation to support that word, and all the demonstration covered the sophism and reveersion technics you constantly use to cover your trace. So you can't use the ad-hominem reversion, given that you previously used it with your contempt about people "being unable to read".

>I'm afraid, from the elements I pointed above, which I will further reiterate, it should become clear who is the person who is in denial as well as who is insinuating things.

Blablabla. I'm affraid that people will see, by looking just two comments above, that I gave a definition of ad-hominem and a chronological map of your reversion. You're screwed and those "Im affraid" moment you keep repeating are useless against that. :D

>First, let's start with the definition from [Wikipedia](https://en.wikipedia.org/wiki/Ad_hominem)

No. :D  First, let see your speaker definition (me, I'm your speaker), instead of runing of what he perfectly spotted. You constantly ignorewhat others say and I don't have to enter your game.

>you are not doing yourself any favour with regards to winning the argument by trying to get the audience agree with the fallacious _"This person is contemptuous therefore whatever they state is wrong and should be ignored"

Wrong ! I firstly proved that you were doing insinuations and then reversing the chronology to cover your first thoughts, and THEN conclude with the contempt adjective. I never looped into a "he's contemptuous so all he says is wrong".

> See how that works? Or do I need to break it down further?

Hahaha, keep going ! :D This is crazy funny. You're gonna pass for a fool in front of every developper coming here. Rufus will be assimilated to a kid who cover himself by inventing things we never said, and trying to make his readers to forget the insinuations he committed.

pbatard, 2019-02-28:
> Yeah, it tells that you constantly use insinuations and as usual, run of them afterward.

You __choose__ how you want to interpret things.

You are pretending I am saying one thing (__your__ misinterpreted version), and then, when I point that your interpretation is incorrect, with evidence, you simply state that you were correct and I am running off, even though I am still there, demonstrating how your interpretation was, and still is, wrong.

How convenient.

> So all these paragraphs from (...) to (...) are null and have no value.

Also exceedingly convenient.

> You have no right to ban anyone

Feel free to appeal to GitHub if I do ban you. I'll be more than happy to point to all the name calling I have received from you on this issue, and let them decide whether they think the ban was legitimate.

> We have the right to name you and to use adjectives to describe you, especially if we make the effort to prove it. And you have no right to ignore what we say

Prove it? With what? Your insinuation that I am supposed to despise Win7 users because of a fictitious issue you invented to further your narrative? Or that I am arrogant for pointing to your obvious mistakes, be it in interpreting a benign statement, repeatedly using ad hominem and all the other argumentative fallacies you've been trying to throw?

You still haven't provided a single fact that could remotely back up your theory that I despise Windows 7 users, which is the crux of the matter here. All we're seeing here is you stating _"I correctly spotted your initial disparaging behaviour against Windows 7 users"_, with no evidence to back up this central claim of your arguing (since everything else you state depends on it).

On the contrary, the one thing you did do with regards to that assertion of yours, is tell everyone that the context, which can be easily surmised from the prior indication that this is a duplicate issue with a direct easily browsable link to said issue, doesn't count and that one should absolutely be entitled to ignore __context__ before they form their __opinion__ on the meaning of a sentence.

If you want this debate to get somewhere, please try to address that for a change.

Or to put it very explicitly:

__What FACTS can you put forward to back up the idea that, even as I pointed to an issue that seems to disprove both your assertions that there is something that should be fixed in Rufus, and that I am willing to help Windows 7 users (which, as the original issue will demonstrate I am still actively testing on), I am despising Windows 7 users and will take the high road on them at the first chance I get?__

How about you start by not ignoring this? How about enlightening us on the thought process that made you construct your initial reply from the statements of mine you saw before, and how you were absolutely entitled to call me arrogant and lazy then.

> That can't be an ad-hominem

Sadly, even if you don't like it, it is. And a prime example at that. You can huff and puff all you want, and pull your hair to try to pretend it isn't _really_ and ad hominem (or, if it is, it's okay because you should be entitled to use ad hominem), but that doesn't change facts.

You have explicitly been inviting people to ignore what I might state on account of a posited character flaw (quote: _"nobody has to read you, given your level of contempt"_). 

No matter how you want to present it, it __is__ an of ad hominem.

Now, if you want to go into context with regards to that statement, how it wasn't meant to be an ad hominem, and how people should not __misinterpret__ what you write, I am all hears... 😄 

> chronological map of your reversion

Yes, as I stated over and over again, you seem to be convinced that I am really really despising Windows 7 users, and then reverted my stance when you, oh-so-correctly, ignored all the evidence that pointed otherwise.

Come on now: You can use the word _lying_, since telling that I am lying is pretty much what you are stating in everything but actual words.

> let see your speaker definition

Wow. I really gotta use that one some day:
_"It doesn't matter if an established definition contradicts mine, it's my definition that should prevail, even if inaccurate"_

> You constantly ignore what others say and I don't have to enter your game.

More convenience. No need to disprove an argument with facts when you state that the "game" should not apply to you.

> You're gonna pass for a fool in front of every developper coming here. Rufus will be assimilated to a kid who cover himself by inventing things we never said, and trying to make his readers to forget the insinuations he committed

I am more than willing to take that chance.

I'm pretty sure anyone who reads that first statement of mine above will agree that, at the very least, there has to be a different way to interpret my _"Nothing to do with Rufus, you need to make sure your version of Windows is up to date when it comes to accessing HTTPS sites."_ besides a contemptuous _"Rufus is really at fault here, but I'm too lazy to do anything about it. Instead, all you Windows 7 idiots should upgrade to Windows 10 already..."_.

Finally, since this was an addon of yours I obviously couldn't address, let's talk about about the full chronological map, rather than your version that also _conveniently_ brushes over very important facts:

* A user on comes with the `You version of the SSL library is too old` issue.
[... cut ...]  trying to allude I manifested then, as well as how the way I explained the statement should actually be construed is _obviously_ wrong (I mean, if only your interpretation of it is meant to stand, then it should be rather easy to demonstrate how mine is flawed). If I reversed chronology, as you allege, then it should be all too easy to do so, since you can drop anything that comes after that statement.

Care to give it a go? Or are you still going to find some convenient excuse not to go there?

Romarain, 2019-03-01:
>You **choose** how you want to interpret things.

No, as I proved it, you ignore what people say, in order to keep saying the same thing.
I don't interpret, I prove : as when you used hints, and I demonstrated that you later blamed us for reading what you said.

>Also exceedingly convenient.

As you, ignoring others. You don't have more rights than us.

>Prove it? With what? Your insinuation that I am supposed to despise Win7 users because of a fictitious issue you invented to further your narrative?

My insinuations ? No, yours. As I proved it, yeah, by typing things your simply ignore.

>You still haven't provided a single fact that could remotely back up your theory that I despise Windows 7 users,

Yes I dit, by quoting you. There's even a series of quoting, a few scrolls up, where people canc clearly see that you accused me to "not be able to read and understand most of what is written", and at the same time, saying that you never said it to me. xD You're screwed a long time ago.

>On the contrary, the one thing you did do with regards to that assertion of yours, is tell everyone that the context, which can be easily surmised from the prior indication that this...

No, seriously : did you see the lenght of that phrase guys ?? Hahaha, this is syntaxically killing eyes.

>Or to put it very explicitly:
>
>**What FACTS can you put forward to back up the idea that,**...

xD Hahahaha. 

>even as I pointed to an issue that seems to disprove both your assertions that there is something that should be fixed in Rufus

You didn't disprove anything I said. See ? That's short, hmm ? That's how you proceed, now enjoy it !

>Sadly, even if you don't like it, it is.
> You have explicitly been inviting people to ignore what I might state
>No matter how you want to present it, it **is** an of ad hominem.

^^ Where are the proofs ? You think you can just repeat what I say to you, about you ignoring what I said ?

>Yes, as I stated over and over again,

Yeah, and you don't seem able to stop ! xD

>Wow. I really gotta use that one some day:
>_"It doesn't matter if an established definition contradicts mine, it's my definition that should prevail, even if inaccurate"_

No, it was an invitation to NOT IGNORE what I said and steal the right to define things, backuped with Wikipedia. You simply ignored my definition, and skipped the discussion to YOUR definition, and I recalled you that you had NO RIGHT to monopolize the attention, by remembering that I provided a definition that you ignored.
And now, you try to **reverse** the situation to make me pass for a guy who would make his explanation to preval on others ! xD

You're a constant joke, accusing others to do what you do yourself. You ignore people, steal the attention, want to be read when you skipped what people has written, and then reverse the situation. You have a real problem of chronology.

>* You **misinterpret** a short statement that makes no mention whatsoever about upgrading Windows

No. As I proved it sevral time, you used hints, anbiguous terms, stated clear thoughts about W7 being too old and not maintained in 2020, and then blame us for understanding what you mean, and try to cover your trace.

>So, just to bring the point home, I did not "reverse (the) chronology" (I could quip about that, but that would only distract from the point)

OH YEAH you could, hahahaha.

>Care to give it a go? Or are you still going to find some convenient excuse not to go there?

Giv it a go for what ?

pbatard, 2019-03-01:
_Whereby we find the other party completely skirting a very direct question about the what fact they can bring forward to corroborate the initial assertion they made..._

> I demonstrated that you later blamed us for reading what you said.

How I like your constant use of "us" and "we", as an artifice to indicate that everybody "should" be on your side in this matter. Sadly, you it is just that, an artifice. And I don't think I blamed __you__ (unless blaming means something different for your). I stated that you completely misinterpreted multiple statements of mine (__every single one__ you quoted in your initial reply), and then ran with this complete misinterpretation.

> you accused me to "not be able to read and understand most of what is written", and at the same time, **saying that you never said it to me. xD** You're screwed a long time ago.

Yet, strangely, you don't appear to want to quote that thing I am supposed to have said.

I will surmise that this is your attempted rebuttal at my indication of how this:

> I'm sorry but I'm not going to spend a lot of time on platforms (Windows 7) that is going to reach End of Life soon (Jan 2020) and that simply happen not to be able to download files because [...]

is meant to be read, which, I as __already__ tried to explain to you but which you don't appear to want to hear, when __not__ isolated from the context (since it is a direct reply to _"You (could) take actions from there (like alerting the user or using an alternative download method)_", __which I quoted__) does not even remotely mean what you __insinuate__ it means.

So let me go over this logically, first by indicating, if needed, that at this stage of the discussion (as I __also__ indicate in the post you quoted from) because the second poster didn't provide enough data, I have to assume that they are running into the same SSL issue as OP and are using Windows 7:
* New user, with allegedly similar issue, appears to ask that I do more for the __limited number__ of Windows 7 users that might run into the SSL issue even if (since this directly derives from the context of the original issue), for one reason or another, they haven't kept their system up to date, and especially they bring forward the issue of using an alternate download method (which I will assert, means adding a custom SSL library and furthermore, assert that this is really what they are after because between getting an error or having the app succeed without issue, they will of course prefer the latter).
* I reply that doing so, especially the part suggested that would require me add a cumbersome and custom SSL library __ONLY__ to help this __limited subset__ of Windows 7 users is not a good investment of my development time, especially considering that Windows 7 is about to reach end of life (meaning, if it was Windows 10, and I had evidence that there exist a lot of users that can't update and are going to be affected, I may do something, but for a platform with just one year of official shelf life, it simply doesn't make logical sense to invest that level of effort), and even more so as, if you are no using an _obsolete_ version of Windows 7, (meaning one that you have kept up to date in terms of system updates) you should not see this issue.

As far as I am concerned, and I will assert, as far as others who are able to read a statement in context, this is the very logical thing to state on account that:
* It is more beneficial for a developer to spend time adding features for platforms that aren't reaching end of life, as this logically where they will benefit the most users.
* As far as I have seen, this only affects a very limited number of users and can be easily remediated (for instance, I have yet to receive feedback from a any Windows 7 user affected by this issue who still weren't able to access the Rufus download server using SSL after applying the relevant update). 
* And I will add, considering that the problem occurs when a user tries to download files, it means that they are using a Windows 7 platform in a connected fashion, which becomes a big risk if that platform is missing critical system updates (such as ones that fix important security vulnerabilities). So adding a custom SSL library to get non up to date Windows 7 download files might actually end up being a __worse proposal__ for the user than asking them to bring their platform up to date, as it may end up leaving them more vulnerable to threats.

So, that's all what that quote meant. If you want to interpret it as being a disparaging comment about Windows 7 users, or even me stating that _"Windows 7 is obsolete, move on already"_, it's of course your choice. But you need to realise, if you still can, that if you consider the context, your interpretation that this is a disparaging comment about Windows 7 (or its users) is not even remotely close to was a non biased person is likely to read from it, especially in context.

Ergo, with this as well as the elements I pointed out in my last reply, I believe I have provided enough evidence to demonstrate that __NONE__ of the points you quoted from me, in your initial reply, __when read in context__, can even remotely be interpreted in the way you appear to have read them. If you read it carefully, none of what you are trying to assert is there, and therefore none of what you __insinuate__ (since you seem to like that word) about my behaviour does stand the test of scrutiny.

[... cut ...] ut instead always strive to read statements in context, next time you want to send disparaging comments towards someone.

Now, since I could do without yet another _"No, it's you who are wrong! And it's you who are not reading what I say!"_ I will now also lock this thread. And I will also warn you that, if you feel like pushing this matter further (by trying to add yet another retort like _"You're a constant joke"_ in another issue for instance), I will also have no problem banning you as user.