pbatard/rufus #1251: Rufus on rpi3! cant find my usb

Askmewho, 2018-12-10:
I know, there its a lot of post than reffers to this problem.
the meaning of the post its to tell you and the community than I succesfully run rufus-arm on rpi3 with wine natively.
[image]

i will keep trying to fix the problem with wine. its happen on the x86 wine so, it was expected to appears on arm too.

pbatard, 2018-12-10:
Yes, I saw your earlier post. Please don't create an issue if it's to report that WINE can display the graphical interface of a UI application. That's the very least one should expect from WINE, and I am __not__ interested in the troubleshooting of WINE issues.

Rufus was designed to be used on Windows, period. If WINE actually manages to emulate enough of the Windows platform for Rufus to work on a non Windows platform, great! If not, then it's a pure WINE problem and I'm afraid I have more than enough other stuff to take care off (such as [adding the rpi3 UEFI firmware to EDK2](https://lists.01.org/pipermail/edk2-devel/2018-December/033686.html), so that people can run the __full UI version of Windows 10__ on the rpi3... and use Rufus ARM64 there without having to emulate anything).

As such I will close this issue since it's out of scope.

closed 2018-12-10 by pbatard

Askmewho, 2018-12-10:
I understand your possition, i will no bother you any more. besides, wine its not an emulator. and I am running native rufus arm version for rt.