pbatard/rufus #1253: [0x00000002] Rufus try to load en-US file on German Operating System

fuba82, 2018-12-20:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
Using a German Operating System but Rufus try to load **en-US** file.

Log
---
```
Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: FIXED, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7628, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8E38209B-5488-432D-85B7-430ABF21FE2B}
Max parts: 128, Start Offset: 17408, Usable = 62742757888 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {8C9A6961-57DB-4394-B0DF-09C48B6A861C}
  Size: 58.4 GB (62741726720 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_DE-DE_DV9'
  Size: 4.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1809Oct_German_x64.iso (4.7 GB)
Found potentially blocking process(es) against I:
● C:\Windows\System32\svchost.exe (rwx)
Could not load 'C:\WINDOWS\system32\en-US\shell32.dll.mui': [0x00000002] Das System kann die angegebene Datei nicht finden.
```

pbatard, 2018-12-20:
This has already been fixed [f94d2f479ffd40046fc279a0a3be108f19d4b8ad] but that fix is part of the 3.5 release which is not out yet. Basically the problem is that Rufus is trying to display a localized version (in your case German) version of the _"This drive is in use. Do you want to format it anyway?"_ message, and, prior to the fix, if we couldn't locate the string for that message in the Windows MUI libraries for German or English, then we would just cancel formatting altogether, which of course is not what we want. So Rufus 3.5 will at least display its own English version for that message if it can't find the localized one (which it usually can). For the record, I have only ever seen that issue happen when trying to run Rufus on Windows Pro Workstation (and even there, it happened a couple of times, and then I never saw it again). Can you tell what type of Windows 10 you are using? Is it Pro, Pro Workstation, Home, Enterprise, or something else?

Also, if you make sure that no other service is accessing the drive, you should be able to get Rufus to create it, since the root of the problem is that if there is something accessing the drive, and that when Rufus tries to report a message to alert the user about it, it automatically cancels the whole operation (which again, will be fixed in 3.5). So if you make sure no other service is accessing the drive, you will be good.

Finally, as opposed to what you assume, the loading of the `en-US` version of `shell32.dll.mui` is actually a [fall back that happens if the `de-DE` version of `shell32.dll.mui` could not be located](https://github.com/pbatard/rufus/blob/master/src/stdfn.c#L962), so the problem is not that Rufus is trying to use the `en-US` version by default, but instead that it couldn't locate the `de-DE` and then the `en-US` version of what it was looking for.

fuba82, 2018-12-20:
Im using Windows 10 Pro Version 1809 Build 17763.195.  
I've also tried to run a **/sfc scannow** and installed en-US Language Pack but it did not work.  
Maybe something went wrong with the new Windows Version after update or somthing.  
  
The Drive is not used by anything, i've also tried to re-mount it physically and also totaly cleared and re-assigned it with diskpart.  
  
The German shell32.dll.mui is in the right place becasue i found it there.  
  
I temporary *fixed* the problem by copy the German shell32.dll.mui from the de-DE folder to then en-US folder (it was not present there for some reason?).  
Rufus is now working was always, but i really dont know why the problem happens becasue until now Rufus **always** worked fine.  
  
Thanks for your fast Response!  
Greetings :)

pbatard, 2018-12-20:
Thanks for the update. And likewise, when I got the error on my system, I found that `shell32.dll.mui` existed in the system's locale directory (in my case `en-GB`) and not in `en-US` (which is fine: up until now I had assumed that if a `.mui` existed in the current locale then it must also exist for `en-US`, which is why I never suspected that the fall back could fail, but as I found out on my `en-GB` version of Windows, that is not always the case).

At this stage, I'm not sure why Rufus wasn't able to locate `shell32.dll.mui` in the locale folder. When I tried to test the code that failed, on the same machine as I saw the issue, it wouldn't fail and was able to open `shell32.dll.mui` (and get the localized message to display) without any issue. At any rate, Rufus 3.5 will contain code to display a message even if `shell32.dll.mui` could not be opened, which will fix this issue.

> The Drive is not used by anything,

if Rufus tells you it was, then it was. Specifically, Rufus did report that a Windows Service was accessing the drive, and I am 100% positive that, even if you think your drive was not being accessed, it was actually being accessed at the time:
```
Found potentially blocking process(es) against I:
● C:\Windows\System32\svchost.exe (rwx)
```

Be mindful that, it's not because you cant' see an actual application accessing the drive that it idn't accessed. If Rufus reports that something is accessing the drive, you should believe what Rufus tells you.

> I temporary fixed the problem by copy the German shell32.dll.mui from the de-DE folder to then en-US folder

You shouldn't do that. As opposed to what I also thought when I originally wrote the code, `shell32.dll.mui` does not have to be present in the `en-US` folder. Instead, you should have either waited for whatever Windows service was accessing the drive to be done with it (perhaps a reboot could have helped), knowing that the next version of Rufus will let you continue instead of cancelling the whole format process.

fuba82, 2018-12-20:
> Be mindful that, it's not because you cant' see an actual application accessing the drive that it idn't > > accessed. If Rufus reports that something is accessing the drive, you should believe what Rufus tells you.
Hmm very interesting...

-  I can eject the Drive by Windows without any problems and messages.  
- Also i restarted the System  
- Also i Re-Assigned the Drive physically  

But Rufus < 3.5 always say:  
Found potentially blocking process(es) against I:
● C:\Windows\System32\svchost.exe (rwx)
Also directly after Windows restart!

I compiled the github master by VS2017 and with v3.5.1433 it works but not with v3.4.1430.  
Anyway i got my ISO now to my USB Drive and it works 👍 
  
Thanks!

pbatard, 2018-12-20:
> I compiled the github master by VS2017 and with v3.5.1433 it works but not with v3.4.1430.

Yup, I told you this would be fixed in next version.

If that's okay with you, I will close this issue then. But if you want to help, and have some time, you may still try to identify what service might be the culprit, so that other people running into the same issue can find out what they should try to disable.

closed 2018-12-20 by pbatard

fuba82, 2018-12-20:
Yeah you can close it.  
  
The service what Locks the Drive is:
**Distributed Link Tracking Client**

>**Service Name:**  TrkWks  
>**German Localized Name:**  Überwachung verteilter Verknüpfungen (Client)  
>**English Localized Name:**  Distributed Link Tracking Client (TrkWks) 
>**Command Line:**  C:\WINDOWS\System32\svchost.exe -k LocalSystemNetworkRestricted -p -s TrkWks
>**Registry:** HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\TrkWks

pbatard, 2018-12-20:
Thanks for that info. It looks like this service is running by default (and in the vast majority of cases it should not be the cause of any conflict).

I'll see if I can cook something in Rufus to also report the parameters that were used with `svchost.exe` so that problematic services can be identified more easily.