pbatard/rufus #1256: Built-in updater doesn't work to update from 3.3 to 3.4 : Unable to access version data

fanfan54, 2018-12-29:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I've been using Rufus Portable (the portable version downloaded from your website, with rufus.ini in its folder) for decades, every time I have been able to use the built-in updater.
I'm still on Rufus 3.3.1400 and the built-in updater won't see the update to 3.4.
In fact, when I try to manually update by clicking "Check Now" in settings, I get a misleading default message box that says "No new version of Rufus was found", while in the logs and in the status bar I see "Updates: Unable to access version data".

On this screen I tried with "Include beta versions" enabled, but I have the same issue when it's disabled.
[image]

Maybe it's a GitHub/server issue? I hope you will be able to fix it, because I guess that only few people also check for updates manually on your website, when Rufus says that no new version was found.

**I'm not the only one with this issue, because when I search for "Updates: Unable to access version data" on the issue tracker on GitHub, I see that many logs from other issues, such as #1230 and #1213 for the most recent ones, have the same error line in the log.**

I tried on another computer, I've the same error message and the same issue. Both had a working Internet connection.

I did not try to use Rufus for its function before exporting the log, I just tried to manually check for stable updates.

****

Log
---
```
Rufus version: 3.3.1400 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
Localization set to 'en-US'
0 devices found
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data

```

fanfan54, 2018-12-29:
The version 3.2 has the same issue, so maybe it's a server error.

fanfan54, 2018-12-29:
Here is what I see on Rufus 3.3 when I set "VerboseUpdateCheck" to 2 in the .ini config file:
```
Rufus version: 3.3.1400 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Checking for Rufus updates...
Checking release channel...
Base update check: Rufus_win_x64_10.0.ver
Trying Rufus_win_x64_10.0.ver
Trying Rufus_win_x64_10.ver
Trying Rufus_win_x64.ver
Trying Rufus_win.ver
Found match for Rufus_win.ver on server https://rufus.ie/
Updates: Unable to access version data
```

pbatard, 2018-12-29:
Next time, please look at the ChangeLog for 3.4:

> * Fix broken update check due to server switch

We switched server and because we don't have full control of it, we can't set the mime type of `.ver` types to `text/plain` which means that all versions of Rufus prior to 3.4 cannot read the version data since they check that the `.ver` they receive is `text/plain`  (and not `application/data` as GitHub sets it by default now).

The __only__ way we can fix, short of swicthing servers and being submitted to the major DDoS attacks that forced us to move to GitHub, since GitHub only applies official mime types, is if we create an RFC to have the whole internet accept that all `.ver` files should be of mime `text/plain`, which isn't going to happen.

So, this is an issue that:
* Was mentioned in the ChangeLog
* Has been fixed in 3.4
* Cannot be fixed for earlier versions and therefore if you are using a version earlier than 3.4, you will need to upgrade manually.

Closing.

pbatard, 2018-12-29:
By the way, if you want to look at the code, [bc6e23ec119c19e02f6c97d4c2ccd88c4300007b] is what you are looking for.

closed 2018-12-29 by pbatard