pbatard/rufus #1261: Burning ISO resaults in ubootable USB

PhoebeAnn, 2019-01-09:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log. [No error message]
- [x ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image. [Having a problem finding the official image value -- it's in the image somewhere, and they say to use an app  to find it. working on that.]

Issue description
-----------------
After using RUFUS (or ETCHER, but I figure RUFUS support will get me a better answer faster), my USB stick winds up as an unusable, unbootable GPT volume. This only happens (so far) with this particular ISO (FreeNAS-11.2-RELEASE-U1). I can use DISKPART to reformat as MBR FAT32 OK; no problems doing a long format; issue has been replicated on 2 PCs and 3 USB sticks.
Log
---
```
Rufus x86 v3.4.1430
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Canon MX510 series USB Device' (04A9:175C)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk Cruzer Glide 3.0 USB Device' (0781:5597)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x795F59FA
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32009879552 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Canon MX510 series USB Device' (04A9:175C)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk Cruzer Glide 3.0 USB Device' (0781:5597)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x795F59FA
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32009879552 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'FREENAS'
  Size: 572.0 MB (Projected)
  Note: File on disk is larger than reported ISO size by 300 KB...
  Has a >64 chars filename
This ISOHybrid is not compatible with any of the ISO boot methods we support
Using image: FreeNAS-11.2-RELEASE-U1.iso (575.5 MB)

Computing checksum for 'E:\Downloads\FreeNAS\FreeNAS-11.2-RELEASE-U1.iso'...
  MD5:	 6fbc3b9c83143b19faadac97f99eee39
  SHA1:	 36e9b3e9f378d39227c9a0a3a0ccf16e4343576e
  SHA256: 894fd364871995e620492334f9d713ec91623523dc0aa33f627191b41e221cf9

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 29.8 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Writing Image...

Found USB 2.0 device 'Canon MX510 series USB Device' (04A9:175C)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk Cruzer Glide 3.0 USB Device' (0781:5597)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x795F5979
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 575.5 MB (603411968 bytes)
  Start Sector: 1, Boot: No

```

pbatard, 2019-01-09:
> my USB stick winds up as an unusable, unbootable GPT volume.

If it's unbootable, then you have to take it to the people who created the image. Because both Rufus and Etcher wrote your image in DD mode, then whether it boots or not is entirely up to the people who created the ISO.

So the "bootability" factor is not something that Etcher or Rufus have any means to alter in this specific case.

Now, with regards to "unusable", you will have to define what you mean.

Clearly, if you were able to repartition and reformat the drive, then it is not unusable. And you could probably have done the same in Rufus (select "Non bootable" in the first dropdown) to restore your drive to a state where you can access its content on Windows.

However, if your complaint is that Windows is not able to access the content of the drive once you have written the ISO onto it, then I think you need to be aware that Windows only support a __very limited__ number of file systems by default, and typically, the file systems used by Linux or *BSD distributions (such as the ones on which FreeNAS is based) will not be accessible from a Windows. But there's nothing fancy to it. Microsoft have decided that they did not need to add support for "foreign" file systems in Windows, so of course, after you write an image in DD mode that contains such a foreign system (which tools like Rufus and Etcher can do, even if Windows does not support said file system, since all that's needed it copy the image bit for bit onto the USB), then the USB will no longer be accessible from Windows, and may lead people to think that it is "unusable". But that is just the way things are if you are trying to access a device that contains content on a file system that your OS does not support: you won't be able to see anything on that drive.

Does this mean that the drive is unbootable? If the people who created the image did their job properly, then no, it doesn't?
And does it mean that the drive is unusable? Not at all. You can use it, but just not for storing data from Windows, and if you want to do just that, then you will need to reformat it with a file system that Windows can handle.

Now I tried to test `FreeNAS-11.2-RELEASE-U1.iso` on an UEFI machine, and it didn't seem to boot for me, but I suspect this may be because it requires BIOS/CSM which I hadn't enabled, and it's not my job to figure out what FreeNAS requires when I could validate that the image was written exactly as it should from Rufus (i.e. if there is any issue, then it's a pure FreeNAS one, and it's up to the FreeNAS people to sort it out).

I found however that I had no issue restoring the drive to a file system that Windows can handle in Rufus (again, by selecting "Non bootable" in the "Boot selection" dropdown, so I believe your issue only stems from the misunderstanding that Windows can read all file systems out there, whereas, natively, it only supports up to 6 file systems, and has no support whatsoever for the ones used by Linux/NetBSD/FeeBSD/FreeNAS.

As I believe I have answered your question, I will close this issue.

closed 2019-01-09 by pbatard

PhoebeAnn, 2019-01-09:
Thanks, Pete. Exactly the conclusion I came to. I should have mentioned, I suppose, that the USB is also unbootable on the FreeNAS machine for which is was intended, so it's not completely a Windows-related problem. I'm off to the FreeNAS forums.

PhoebeAnn, 2019-01-14:
I just tried to use the " Windows USB/DVD Download Tool" to write the USB drive, and it claims that FreeNAS-11.2-RELEASE-U1.iso is not a valid ISO file. How much checking does RUFUS do to verify that the target file is a "valid ISO file", and what might constitute an invalid ISO file?

pbatard, 2019-01-14:
```
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'FREENAS'
  Size: 572.0 Mo (Projected)
  Note: File on disk is larger than reported ISO size by 300 Ko...
  Has a >64 chars filename
This ISOHybrid is not compatible with any of the ISO boot methods we support
Using image: FreeNAS-11.2-RELEASE-U1.iso (575.5 Mo)
```
FreeNAS is an ISOHybrid image, and therefore it can be written in DD image mode, which Rufus supports but which Microsoft's _Windows USB/DVD Download Tool_ doesn't.

Please be mindful that different applications developed by completely different groups of people, that may happen to __look__ like they share some functionality, might still have completely different features.

I also hope that you understand that this is not a support forum for Microsoft tools, and you shouldn't try to use a roundabout way ("Why doesn't the Microsoft utility not support my ISO?" &rarr; "Why does some other utility support it?") to get somebody else but the original developers of the utility to answer a question about the utility's limitation.

PhoebeAnn, 2019-01-15:
I do understand that "...this is not a support forum for Microsoft tools...". I didn't know there were multiple ISO image flavors, either, Thanks for the clarification, and my apologies for not actually reading the log I sent you.