pbatard/rufus #1263: FAT32 for MBR windows install drives

DanTheMan827, 2019-01-14:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Feature Request
-----------

Allow formatting USB drives for Windows installer as FAT32 when MBR is selected.

If this is possible, it would allow the same USB drive to be used for both MBR and UEFI systems

Alternatively, it may be possible to have another small FAT32 partition before the NTFS one containing the EFI boot files.

pbatard, 2019-01-14:
I don't thing your properly looked at the FAQ. You can already created BIOS + UEFI bootable Windows installation media with the <kbd>Akt</kbd>-<kbd>E</kbd> cheat mode.

Relevant entries [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode) and [here](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes)

closed 2019-01-14 by pbatard

DanTheMan827, 2019-01-14:
> BIOS + UEFI bootable Windows installation media with the Akt-E cheat mode.

😳 sorry about that...

pbatard, 2019-01-14:
Yeah, there is a reason why the checklist is __NOT__ optional. If you mark an item as checked, I expect you to have performed the relevant step. It's only there to help __YOU__!