pbatard/rufus #1265: Windows 10 (to go) 32bit missing bootx64.efi (needs bootia32.efi)

escape75, 2019-01-17:
Issue description
-----------------
When you create a Windows to Go USB flash disk using a Windows 10 ISO 32-bit,
the resulting USB boots via BIOS but not EFI since I'm assuming the Rufus EFI loader
is looking for /EFI/Boot/bootx64.efi and there will only be /EFI/Boot/bootia32.efi

Additionally you cannot rename bootia32.efi to bootx64.efi to make it work.
Boot process says that bootx64.efi cannot be found and aborts.

*Windows 10 using 64-bit ISO works fine in both BIOS and EFI modes.

pbatard, 2019-01-17:
Also, it very important that you clarify whether the UEFI firmware you are trying to boot your media on is actually 32-bit (please provide the full model and manufacturer of your motherboard or PC). Unless you are using a cheap Chinese low-powered all-in-one PC, it is exceedingly unlikely that you have a 32-bit UEFI firmware in which case, and as opposed to what used to be the case for BIOS, the __ONLY__ type of Windows you can boot and install is 64-bit. 32-bit compatibility on 64-bit UEFI firmware does not exist, so you can't simply pick a 32-bit ISO and expect it to work on 64-bit UEFI (unless of course you are booting in CSM mode, but then you're really using BIOS rather than UEFI).

escape75, 2019-01-17:
I was trying to boot on my HP Pavilion 300 Mini.

Here's the log file of when I create the USB:

`Rufus x86 v3.4.1430
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
Using autorun.inf label for drive E: 'USB'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1971, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004B69A
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.1 GB (16218324992 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X86FRE_EN-US_DV9'
  Size: 3.2 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: En_Windows_10_Consumer_Editions_Version_1803_X86_Dvd_12063380.iso (3.2 GB)
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0\sources\install.wim:[0] (API)
Extracting: C:\Users\Anna\AppData\Local\Temp\Ruf1AD6.tmp (From [1].xml)
Closing: \\.\CDROM0\sources\install.wim
Will use 'Windows 10 Pro' (Build: 17134, Index 6) for Windows To Go

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 15.1 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Windows selected the UEFI:NTFS partition for formatting - Retry needed
Format error: [0xC00304D5] The operation could not be completed. A retry should be performed.
Volume is already mounted, but as D: instead of E: - Unmounting...
Re-mounted volume as 'E:' after error

Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1971, Tracks per cylinder: 255, Sectors per track: 63
[... 90 lines cut ...]
     86816 KB in use by the system.
     23392 KB occupied by the log file.
  10267796 KB available on disk.
      4096 bytes in each allocation unit.
   3959290 total allocation units on disk.
   2566949 allocation units available on disk.
NTFS Fixup completed.

Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
Using autorun.inf label for drive E: 'CCCOMA_X86FRE_EN-US_DV9'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1971, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000BE2B9
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.1 GB (16217252352 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 31676369, Boot: No
`

pbatard, 2019-01-17:
Well, considering that the HP Pavilion 300 Mini comes with Windows 10 x64 pre-installed, then your UEFI firmware has to be 64-bit (else, HP would have to force their UEFI firmware to boot in CSM mode always, which would make the provision of an UEFI firmware completely pointless). This means that the only version of Windows you can ever install, in UEFI mode, has to be the 64-bit one.

Was there any reason you chose to install Windows 32-bit instead of Windows 64-bit? What are you trying to achieve by installing the 32-bit version that you don't think you can accomplish with the 64-bit one?

Also, from what I can see in the log:
* Rufus had no issue detecting the `bootia32.efi` from your ISO (else it wouldn't report `Uses: EFI` and `Uses: Bootmgr (BIOS and UEFI)` since those are the `ia32` versions on 32-bit ISOs
* `bcdboot.exe` was invoked without error for all firmware types (that's what the `/f ALL` option does) so, if it wasn't able to set boot for 32-bit UEFI (since the 32-bit bootloaders is what the 32-bit version of the Windows 10 ISO provides), it would have reported an error.

Therefore, the only issue here is that you are trying to boot 32-bit UEFI bootloaders with a 64-bit UEFI firmware, and as I explained, unlike what is the case in the BIOS/CSM world, UEFI treats the x86_32 and x86_64 as incompatible processors, so can't do that.

escape75, 2019-01-17:
Thank you for the explanation,- the issue can now be closed!

I was trying to use Windows 32-bit because:
1) it takes up less space on a smaller USB drive
2) performs and boots faster than the x64 editions

There was a strange error when I was formatting the drive (as seen in the log above)
"Windows selected the UEFI:NTFS partition for formatting - Retry needed"
but I retried and it went fine the second time.

pbatard, 2019-01-17:
> There was a strange error when I was formatting the drive (as seen in the log above)

Yes I saw it. This is something that can happen because we write a small partition for UEFI:NTFS and then ask Windows to format the remaining partition, which is fine most of the time, but, occasionally, Windows will pick the smaller UEFI:NTFS for formatting on its own, which is not what we want. I don't have a good solution for that at the moment, so the best I can do is detect the issue and ask the user to try again, since usually things will work as expected the second time around. This is not indicative of any specific problem, it's just one of Windows' weird behaviours that need to be accounted for...

I will close this issue then.

closed 2019-01-17 by pbatard