pbatard/rufus #1267: [question] linux live filesystem doesn't use full flash drive space

matcool, 2019-01-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When trying Linux Mint on my new 8gb flash drive I noticed the flash drive still had 5gb free, and when using it I only had 1.1gb free instead of my expected ~5gb.
Is there any way to increase the "live filesystem" size?

pbatard, 2019-01-20:
This is because you are hitting the limits of the FAT32 file system (which doesn't allow the creation of files larger than 4 GB). So the larger your "virtual live partition" can be is 4GB, leaving ~1Gb unused.

Depending on the distro you use, you should be able to overcome this issue by choosing NTFS instead of FAT32 when creating the drive in Rufus, as you should then be able to create a "virtual live partition" that occupies all the remaining free space.

Note however that the use of NTFS for Linux should be considered experimental at this stage. Last time I checked, I have had some good success with Debian and Ubuntu, but the distro __must__ come with native support for NTFS (especially with regards to its bootloaders), and not all distros do.

Alternatively you can try writing the ISO in DD mode. If you are using an ISOHybrdid, Rufus shouold have prompted you whether you wanted to write it in ISO mode or DD mode, but since you chose not to attach your log (__Please don't do that, it makes it a lot harder to try to help YOU!__), I can't tell if Mint was an ISOHybrid. Not sure how live partitions are handled in DD mode though, so you'll have to check with the Mint support forums.

DaveBlack, 2019-01-22:
In case helpful, Ubuntu based Linux Mint supports NTFS on USB devices.

@pbatard, Have you had thoughts on offering exFAT as an alternative to NTFS, as that should work for most OS? Apologies if been asked previously,

pbatard, 2019-01-22:
Yes I have thought about it, and it's the other way round: NTFS is what most OS will pick over exFAT. For instance, not so long ago, Syslinux, which is one of the main Linux bootloaders, added support for NTFS, but (as far as I know) they have yet to add support for exFAT. Likewise, you'll fine that a lot of modern UEFI firmwares have support for native NTFS boot (though this is not that much of an issue, since Rufus can install an NTFS driver of its own if needed), whereas I am not aware of __any__ UEFI firmware that provides native support for exFAT.

So, whereas a lot of people seem to think that exFAT would be preferred over NTFS, it's the other way round: De facto, NTFS has won over exFAT for some time, and therefore it makes little sense spending time on trying to promote exFAT as a viable alternative to FAT32, when nobody else (bootloaders, UEFI firmwares) is going to follow suit at this stage.

Also, I really fail to understand (this has happened quite a few times in this issue tracker) why every time someone seems to propose NTFS as a good alternative to the FAT32 limitations, someone will feel the need to chime in with: _"But what about exFAT? Can't you please consider exFAT instead?"_

There's really no need for 2 competing solution to work around the same issue here. The one argument I have heard for preferring exFAT over NTFS is that, allegedly, it's supposed to be easier when it comes to wear and tear of flash memory, but there again, I have yet to see hard evidence to corroborate this assertion, especially given that modern flash drives are designed from the get go to support well known and widespread file system such as NTFS.

So, really, if you want to help with the FAT32 issues, please help improve NTFS support where needed, as NTFS has pretty much but won the battle of becoming the de-facto file system of choice where FAT32 won't do. This means reminding distro maintainers that they should not forget to include NTFS module support in their GRUB or Syslinux bootloaders, and of course devise their init scripts to also be NTFS friendly, just like Debian and Ubuntu have been doing for some time now.

pbatard, 2019-01-31:
Closing issue.

closed 2019-01-31 by pbatard