pbatard/rufus #1270: mutiple iso bootable

pedramlavasani, 2019-01-24:
hi
Rufus is the best but i would like to say one point 
add multiple iso bootable option to next update please

pbatard, 2019-01-24:
Next time, follow the checklist. There's an entry for your question in the FAQ!

closed 2019-01-24 by pbatard

pedramlavasani, 2019-01-24:
> Next time, follow the checklist. There's an entry for your question in the FAQ!

where is FAQ?

pbatard, 2019-01-24:
There was a link to it in the checklist __you chose to ignore__ when you created the issue:

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

Again, if you __choose__ to ignore what people advise you to do, you forego any right to ask for help.