pbatard/rufus #1271: PID request for Device in use by another process

shakydd, 2019-01-26:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Thanks again for the great work over the years on Rufus. 

This is really a feature request and apologies if there's a better place to put it. I have the problem that others have had with the 'Device in use by another process' warning. Would it be possible in a future version to output the PID of the process as well as the name? For example, in my scenario, I have svchost.exe using the device but of course there are quite a lot of them - narrowing it down to the exact PID would be helpful in troubleshooting what is locking the device.

Thanks again for an amazing product.

Log
---
```
<FULL LOG>
```

pbatard, 2019-01-26:
This has already been added in [7b2edbfd6fef8c5bd8cdc7f2c12e0fd0c6f0a0cf] along with something that will be of a lot more of interest: the full commandline that Windows used to launch the service (which will help tell precisely what service that is) [5309dc88e202044d0a49e6bd1aed51753b1670a2].

It'll be part of Rufus 3.5.

closed 2019-01-26 by pbatard

shakydd, 2019-01-26:
Fab, top stuff Pete

Sent from BlueMail<http://www.bluemail.me/r?b=14344>