pbatard/rufus #1273: NTFS is only available when using MBR

sasatefa2009, 2019-02-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am trying to burn a rescue disk called "WinPE 10-8 Sergei Strelec". the disk has to be in MBR/FAT32.
When using GPT i had the two file systems availabe (NTFS-FAT32), but when using MBR i only have NTFS available.

Log
---
```
<Rufus x86 v3.4.1430
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x286917DB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.3 GB (7847542784 bytes)
  Start Sector: 8192, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'USB_STRELEC'
  Size: 3.7 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS only)
Using image: WinPE10_8_Sergei_Strelec_x86_x64_2019.02.04_English.iso (3.7 GB)
>
```

pbatard, 2019-02-10:
Yes, that is by design. There is no point in providing FAT32 for MBR when __I have yet to hear of a single BIOS system that cannot boot Windows in MBR/NTFS mode__.

Providing the option to use FAT32 would actually be problematic as FAT32 cannot be used with some ISOs that contain a file that is larger than 4 GB and, again, since I have yet to have had a report of a single BIOS system that is unable to boot a MBR/NTFS Windows image, it makes zero sense to enable FAT32 to be selected.

> the disk has to be in MBR/FAT32.

I'll be very curious to hear the justification behind that, since I don't know of any valid reason to want to use FAT32 when booting a Windows image in BIOS mode (UEFI mode is slightly different, but then you can use GPT/FAT32 so I have to assume that, since you think you "need" MBR/FAT32, you are trying to boot in BIOS mode). 

So why __exactly__ do you think you need MBR/FAT32? And what is the error you get when you try MBR/NTFS?

sasatefa2009, 2019-02-10:
> >Yes, that is by design. There is no point in providing FAT32 for MBR when **I have yet to hear of a single BIOS system that cannot boot Windows in MBR/NTFS mode**.

The ISO actually booted in MBR/NTFS but the CD Maker says it has to be in FAT32. and thats because some dos programs inside didn't work in NTFS.

> > the disk has to be in MBR/FAT32.
 I'll be very curious to hear the justification behind that, since I don't know of any valid reason to want to use FAT32 when booting a Windows image in BIOS mode (UEFI mode is slightly different, but then you can use GPT/FAT32 so I have to assume that, since you think you "need" MBR/FAT32, you are trying to boot in BIOS mode).

> > So why **exactly** do you think you need MBR/FAT32? And what is the error you get when you try MBR/NTFS?

GPR/FAT32 didnt boot it. so it has to be in MBR.

pbatard, 2019-02-10:
> The ISO actually booted in MBR/NTFS but the CD Maker says it has to be in FAT32. and thats because some dos programs inside didn't work in NTFS.

If you are running DOS then you're not booting or installing Windows. There is NO version of Windows since Windows 98 that requires you to start in DOS mode.

I will need to need __a lot__ more details about these DOS applications that need to be run as part of a Windows installation, as well as whether they are really DOS applications and not regular win32 commandline applications that will run just as well from the installation media in NTFS after you press <kbd>Shift</kbd>-<kbd>F10</kbd> to open a command prompt.

Again, please be very weary about anybody saying that an installation media "has to be" FAT32, without providing a very detailed explanation about why that should be. You still haven't provided those details.

> GPR/FAT32 didnt boot it. so it has to be in MBR.

Ans again, since you have chosen not to answer this very direct question, what is the problem you face exactly when you write Strelec in MBR/NTFS mode. Do you see an error during installation? If so, what it is.

I hope that you can appreciate that choosing not to answer very direct questions makes me very suspicious about whether there exists a real problem or not. If there is a problem when using MBR/NTFS, then you should have no trouble describing precisely what it is.

Oh, and I'll conclude by saying that, since anybody can take a properly working official Windows ISO, and screw it up, the only ISOs I will only ever officially support in Rufus are the official Microsoft one, which clearly, Strelec isn't. For instance, are you aware that you can __EASILY & FREELY__ get an official Windows installation media, that will work just fine in Rufus and that you can be confident has not been altered to contain any malware (which, I will assert, definitely has to account as one of the main reasons why anyone would ever want to push their own custom installation Windows media, and try to declare it as "better" than the official Microsoft one) once you have validated its SHA-1, by simply going to https://tb.rg-adguard.net/public.php? If Strelec needs to run other applications besides the official Windows installer, as part of its installation process, I can't help but think that it's probably trying to do something very fishy there...

sasatefa2009, 2019-02-10:
I think that you think that this is a Windows OS, it's a rescue disk that only contains a WinPE inside.

[image]

the USB booted in MBR/NTFS but "HDD Regenerator" didn't work.

pbatard, 2019-02-10:
In that cases, it uses a custom ` Bootmgr`, which, of course, is not supported, in the same manner as older `Bootmgr` (such as the ones from Windows XP or Windows Vista are not officially supported &mdash; If they work, great, but if they don't, you're 100% on your own). IMO, there are much better methods to fill a bunch of boot options on an image, including DOS & WinPE, than hijacking `BootMgr`, one of which being the much more flexible Grub4DOS, which should be supported just fine in Rufus.

In other words, if you want to use this image, and complain that it doesn't work with Rufus, then I'll direct you to use another application, as there's no way I'm going to add MBR/FAT32 support for `BootMgr` based images just so that people who hijack `BootMgr` to do something it's not designed to to (run DOS applications) can get multiboot support (something that Rufus is also not designed to support) for a handful of DOS applications.

Closed as _NeverFix_.

closed 2019-02-10 by pbatard

sasatefa2009, 2019-02-10:
[image]