pbatard/rufus #1274: Make image from disc

emwu88, 2019-02-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
[image]

When make image from USB drive, rufus make VHD, not a DD image.
I read in FAQ:
"Alt-V (v2.3 to v2.18, removed from v3.0) - Save selected device to UNCOMPRESSED VHD."

Image have extra 512byte on footer, (Your answer):
https://superuser.com/questions/1367044/why-does-rufus-call-a-vhd-file-a-dd-image

Log
---
Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0415
Will use default UI locale 0x0415
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
0 devices found

<FULL LOG>
```

pbatard, 2019-02-11:
Aha, then this is my cue to enlighten you:

An __uncompressed__ is just a DD image with a 512 byte footer added.

Therefore, because writing that footer (or not) is inconsequential, you can absolutely use the ` .vhd`  that was created by Rufus as a DD image.

Now, in case you have doubts about those extra 512 bytes potentially causing an issue let me explain that:
- If you are restoring the image to a drive of the exact same size as the one you used to create the VHD, then these 512 bytes will happily be dropped by `dd` (At least the Linux version of `dd` doesn't care much if you try to write more data that the drive can hold - it'll report it, but it will not have any kind of impact on the data written)
- If you are using a drive that is larger than the one you used to create the VHD, then, even if you are unfortunate enough to have the 51-byte VHD footer fall at the precise location where the GPT backup is located (a few sectors before the end of the drive), then because this is a backup, any modern OS will detect that the backup is corrupted and fix it by duplicating the master, which is located at the beginning of the disk. Oh, and, even when you restore a pure GPT based DD image onto a larger drive, this kind of GPT backup fix needs to happen anyway, since, obviously, the backup GPT from the original image is at the wrong location, and a new backup GPT needs to be created at the right one. In other words, the writing of that 512-byte VHD footer is __no different__ than the case of expecting whatever garbled data one might find at the expected location for the backup GPT. And really, even as this is a complete non-issue, the possibility that the VHD footer will fall at the precise location where the backup GPT should go is exceedingly small. Oh, and if your image is using a MBR partition table, then it's even less of an issue, because none of the data up to (size of VHD - 512 bytes) will __EVER__ be seen as accessible from the original image.

In summary, and I really have to stress this out, __for all purposes, an uncompressed VHD image is exactly the same as a DD image__. Therefore you can use the Rufus created ` .vhd` exactly as you would use a standard DD image. There's __REALLY__ no _"What if...?"_ or _"But surely..."_ associated to using an umcompressed `.vhd` as you would a DD image.

And in case you still want to dispute that statement, then __PLEASE TRY FIRST__, and the come with a verifiable counter-argument that you have validated, as it will save time for everybody. 

Since I believe I have answered your question if full, I will close this issue.

closed 2019-02-11 by pbatard

DI555, 2019-03-06:
Pardon me, my 50 cents.
And sorry, but 'MultiBoot' issue is closed for commenting(((
''MultiBoot'' is not number one awaiting, is much more..., it's numero uno!
Especially in the way where images will be used like only .iso files... And booting from them will be emulated from that .iso's !!!

And another numero uno is the boot manager functions! Why only flash or sd drive.... Whould be great to install and manage any drive with multiboot function!

pbatard, 2019-03-06:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support