pbatard/rufus #1275: 404 - syslinux

4val0v, 2019-03-03:
Issue description
-----------------
Kali-Linux 2019.1 >USB Flash drive
[image]
Rufus get file from [files/syslinux-6.04/20181203/ldlinux.sys](https://rufus.ie/files/syslinux-6.04/20181203/ldlinux.sys)
Respons: **404** Not Found !

Log
---
```
Rufus x86 v3.4.1430
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x07F33FE5
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15470690304 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20181203 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.2 Гб (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: kali-linux-2019.1-amd64.iso (3.2 Гб)
Downloading https://rufus.ie/files/syslinux-6.04/20181203/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data
```

pbatard, 2019-03-03:
Okay, there are 2 parts to your issue:

1. That Rufus produces an error when attempting to download https://rufus.ie/files/syslinux-6.04/20181203/ldlinux.sys (i.e. a potential CUSTOM version of Syslinux 6.04) is __normal__. These custom versions are only provided on our servers if we know that the regular version (in this case https://rufus.ie/files/syslinux-6.04/ldlinux.sys) won't do. As far as Kali is concerned, there is no such issue (that we know of), so, since vanilla Syslinux 6.04 should work, no "20181203" custom needs to be provided. In other words, again, the check on whether https://rufus.ie/files/syslinux-6.04/20181203/ldlinux.sys exists is supposed to fail. And that is why you see the message `Extended version was not found, trying main version...`

2. On the other hand, that Rufus produces an error when attempting to download https://rufus.ie/files/syslinux-6.04/ldlinux.sys is not normal (the file does exist) and appear to indicate an issue with your environment. This is further corroborated that even as you are using Rufus 3.4, you also can't seem to be able to access the update data.

So, here is what you need to do:

1. Open Internet Explorer (it has to be Internet Explorer, as it is the only browser that uses the Windows system libraries for internet access)
2. Paste the https://rufus.ie/files/syslinux-6.04/ldlinux.sys there and see if it can access the file.

If you get an error, then I'm afraid that, considering that the file is accessible (it is really hosted on GitHub, and if the GitHub servers were down that would be known) and considering that other Windows 7 users are not reporting that error (some may report a _your SSL library is too old_ error, but that one can easily be fixed by making sure that you have applied all the Windows 7 system updates from Microsoft, as you are supposed to, and this is not the error we see in your log), then you will have to figure out what it is in your environment that is preventing access.

Again, just in case you wonder, I __do__ test Rufus regularly on Windows 7 (for instance I have just confirmed that, if you use Rufus 3.4, the check for update does work), so I am 100% positive that the issue you are seeing is not a general problem that Windows 7 users are running into, and that would need to be addressed in Rufus, but something that appears to be specific to your environment, which you need to identify.

Maybe you are using a security solution that is preventing access to certain sites, or that is preventing applications (but not browser) to connect to the internet (I know that quite a few corporations will configure firewalls that way). Or maybe, and especially if you can't seem to access the file with IE, you misconfigured your system or failed to apply all the recent system updates (because GitHub uses modern SSL, you won't be able to download GitHub content unless your version of Windows 7 is up to date).

Oh, and if you do find what in your environment is causing the issue, please report it here, so that it may help other users.

4val0v, 2019-03-03:
I had to manually download ldlinux.sys and ldlinux.bss, put them in a folder created Rufus
I download files from links from [487 issues](https://github.com/pbatard/rufus/issues/487#issuecomment-459591975)

pbatard, 2019-03-03:
Yes, that is what you can use as a __workaround__. But you will admit that having to do that manually is inconvenient, and you may want to try to figure out why your Windows system can't download files or access the application update data, when other Windows systems don't have a problem doing so.

nesch52, 2019-03-04:
20190304 same error as above , had to download "ldlinux.sys" manually. Filestructure : c:\Users\Username\Downloads\RUFUS\rufus_files\syslinux-6.04\pre1-41-g5e426532\
has been created but is empty. Downloading manually to that location and running RUFUS as Admin, did not create any bootable USB from Win7*.iso with rufus-3.4p.exe (0.98MB). Later it says writing...sucsessful but nothing on the stick.

pbatard, 2019-03-04:
@nesch52 

Please attach your full log. I can't help you without a log.

pbatard, 2019-03-04:
Also, I seriously doubt you got that issue when writing an __official__ Win7*.iso, as Microsoft official ISOs do not use Syslinux at all. And I can only support official Microsoft ISOs.

nesch52, 2019-03-04:
You are right, I used a "systemrescuecd-6.0.2.iso" (871MB), because i have a LenovoW530 with preinstalled & licensed Windows 7 prof /de x64...  thought it might work too. Could not find anything else... Here ist the log:
Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'Intenso Business Line USB Device' (058F:6387)
Found USB 2.0 device 'SAMSUNG HD153WI USB Device' (067B:2773)
Device eliminated because it was detected as a Hard Drive (score 25 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SAMSUNG HD204UI USB Device' (1BCF:0C31)
Device eliminated because it was detected as a Hard Drive (score 28 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'TOSHIBA USB 3.5"-HDD' (0930:0B09)
Device eliminated because it was detected as a Hard Drive (score 10 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1011, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xF632D819
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.7 GB (8316764160 bytes)
  Start Sector: 1056, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre1-41-g5e426532 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'SYSRCD602'
  Size: 869.9 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: systemrescuecd-6.0.2.iso (871 MB)
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data
Found USB 2.0 device 'Intenso Business Line USB Device' (058F:6387)
Found USB 2.0 device 'SAMSUNG HD153WI USB Device' (067B:2773)
Found USB 2.0 device 'SAMSUNG HD204UI USB Device' (1BCF:0C31)
Found USB 2.0 device 'TOSHIBA USB 3.5"-HDD' (0930:0B09)
4 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1011, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xF632D819
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.7 GB (8316764160 bytes)
  Start Sector: 1056, Boot: No
Downloading https://rufus.ie/files/syslinux-6.04/pre1-41-g5e426532/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

Computing checksum for 'C:\Users\esch\Downloads\Windows 7 Wiederherstellung\systemrescuecd-6.0.2.iso'...
  MD5:	 faf22d057c79f9903b60639046663e44
  SHA1:	 3b51a5e98cb4461d2d9af56663208b71f46495a3
  SHA256: 580378905cb1aac41125167cde936148b2b8e37646192cfbc85bfbeef6d58b85
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data

Save to VHD operation started
Requesting disk access...
Will use a buffer size of 64 KB
Saving to image 'C:\Users\esch\Downloads\Windows 7 Wiederherstellung\Intenso.vhd'...
Appending VHD footer...
Operation complete (Wrote 7.7 GB).

Found USB 2.0 device 'Intenso Business Line USB Device' (058F:6387)
Found USB 2.0 device 'SAMSUNG HD153WI USB Device' (067B:2773)
Found USB 2.0 device 'SAMSUNG HD204UI USB Device' (1BCF:0C31)
Found USB 2.0 device 'TOSHIBA USB 3.5"-HDD' (0930:0B09)
4 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1011, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xF632D819
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.7 GB (8316764160 bytes)
  Start Sector: 1056, Boot: No

pbatard, 2019-03-04:
> Could not find anything else...

If you have a license then you can get the official ISOs here:
* https://www.microsoft.com/en-us/software-download/windows7

Alternatively you can get them here (even if it is no affiliated with Microsoft, this site does provide official ISO downloads from the Microsoft servers):
* https://tb.rg-adguard.net/public.php

```
Downloading https://rufus.ie/files/syslinux-6.04/pre1-41-g5e426532/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling
```

Looks like the same error as OP. have you tried accessing https://rufus.ie/files/syslinux-6.04/ldlinux.sys from Internet Explorer? I'd really like to find out the result of that test.

At any rate, I'm afraid the only advice I can provide is the one I gave above: Something in your environment (could be a security solution, could be something else) appears to be preventing downloads, as I can confirm that a fully up to date Windows 7 system has no trouble downloading data from the Rufus servers. You will therefore have to find out what, on your system, might be preventing access.

nesch52, 2019-03-04:
Any hint whre to download an official Win 7 professional de? I have only the 4 rescue disks bought from Lenovo officially.

nesch52, 2019-03-04:
Thanks for the d.l.-hint, I Use Chrome X64 latest ver: Version 72.0.3626.119 (Offizieller Build) (64-Bit)

pbatard, 2019-03-04:
I'm afraid you __must__ use Internet Explorer for the test. Chrome and Firefox use their own libraries, so they can't provide the information I want with regards to whether the issue is with your Windows system libraries or something else.

nesch52, 2019-03-04:
OK will try that in IE soon. But next problem , i only can find a : Product ID: AND an according CD Key: ...don't find the Product key???  .System came preinstalled and online activated . So IwWill first try to download  the Win 7 ISO you mentioned before. Microsoft needs a Product key to download the ISO , which i can't find and does not accept the above mentioned...

pbatard, 2019-03-04:
Then use the second link I provided. That's why I provided 2. At any rate, if you need a product key or something else, you should contact your laptop's manufacturer. I can only help you with Rufus usage, __not__ Windows reinstallation.

nesch52, 2019-03-04:
Second Link says: "File is not available for download" when selecting: Windows (Final), Windows 7SP1 (build 7601), Windows7 Professional SP1,Deutsch, Win7_pro_de_x64_ISO...

nesch52, 2019-03-04:
OK Thanks ....

pbatard, 2019-03-04:
You can also try this then: https://www.heidoc.net/joomla/technology-science/microsoft/67-microsoft-windows-and-office-iso-download-tool

But, really, if the OS was preinstalled, then your laptop manufacturer is required provide you with the files you need to reinstall your OS.

ibuprophen1, 2019-03-04:
@4val0v, When you contact the PC manufacturer, they may state that your PC had came with a recovery partition of the Windows 7 OS but, if for "any" reason an error is provided when trying to recover the OS, they are required by law to provide you with a set of recovery disks. 

If the manufacturer states/mentions anything about Windows 7 has reached/surpassed it's EOL/EOS (EOL = End of Life/EOS = End of Support) date, then you can refer to them the following Microsoft Announcement link/information. 

https://www.microsoft.com/en-us/windowsforbusiness/end-of-windows-7-support

Sometimes, depending on the individual you speak to, circles may be attempted. When you see that it's going nowhere, always ask to speak to the supervisor without arguing with them or you could easily get nowhere fast. 

I wish you the best of luck? :-) 

~Ibuprophen

pbatard, 2019-03-05:
Since I don't think there's anything more that can be done on this issue, I will close it.

closed 2019-03-05 by pbatard