pbatard/rufus #1277: Error C2001: Newline in constant

kuh3h3, 2019-03-10:
I know this is known problem and is not rufus`s problem.
but the error occurs always when build rufus on korean, chinese, japanese,russian locale.

yes.
it`s my side(vs 2017 with win 2010( korea locale))`s problem.

i didn`t want to change rufus code`s encoding.

so add directly /source-charset:utf-8 to Additional Options in C/C++ Command Line
in two project( ms-sys and  rufus)

now build ok (though some weird character remain).

pbatard, 2019-03-10:
Thanks for the report.

I think I will use [`/utf-8`](https://docs.microsoft.com/en-us/cpp/build/reference/utf-8-set-source-and-executable-character-sets-to-utf-8?view=vs-2017) rather than `/source-charset:utf-8` as it should also help with the [execution charset](https://docs.microsoft.com/en-us/cpp/build/reference/execution-charset-set-execution-character-set?view=vs-2017) which may be why you still get some weird characters.

At any rate one can only hope that, in future versions of Visual Studio, Microsoft will do what __all__ other smart text processors do and properly detect UTF-8 files even when they don't have a BOM, as this isn't exactly difficult to do. In 2019, developers should really set their applications to expect UTF-8 by default, rather than obsolete local copepages...

pbatard, 2019-03-12:
Should be fixed now. Let me know if you still see an issue.

closed 2019-03-12 by pbatard

kuh3h3, 2019-03-13:
thanks for quick patch.
it works well on vs 2010 .
there are no weird characters now.

but /utf8  not work on intel compiler.

from https://software.intel.com/en-us/forums/intel-c-compiler/topic/733662
improper UTF-8 characters handling by ICC on Windows

 -Qoption,cpp,--unicode_source_kind,"UTF-8"

compile ok, but  link failed

xilink: error #10014: problem during multi-file optimization compilation (code 1)
xilink: error #10014: problem during multi-file optimization compilation (code 1) 

from https://software.intel.com/en-us/forums/intel-c-compiler/topic/361884
xilink: error #10014: problem during multi-file optimization compilation 

/Qipo-

now link ok.

is this correct fix on intel compiler?

pbatard, 2019-03-13:
> is this correct fix on intel compiler?

I don't support intel compiler. Just like I don't support compilation using LLVM or anything but the Microsoft default compiler.

What you got is the best you're ever going to get from the official project. if you need more than this, then I suggest you fork your own project and apply the patches you need for your custom environment. I'm afraid I've got way too much on my plate than to have to worry about compatibility with the intel compiler, sorry.

commit 7e21f8a6 (2019-03-12): [misc] add /utf-8 to MSVC compiler options

* Closes #1277