pbatard/rufus #1278: Canceling an image download should clear the taskbar progress

ItielMaN, 2019-03-13:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
STR:
1. Using Rufus 3.5.1456 Alpha, use the new image download option
2. Select version, edition etc and start the download
3. Cancel download

AR:
The download is stopped but the taskbar progress is shown as paused (I think? see attached).

ER:
The taskbar progress is reset after canceling the download.

Log
---
```
Rufus x86 v3.5.1456 (Alpha) (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x040D (he-IL)
Will use default UI locale 0x040D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'he-IL'
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.05/Fido.ps1
File length: ‎29.6‎ KB
Successfully downloaded 'Fido.ps1'
Signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/he-IL/software-download/Windows10ISO
Querying https://www.microsoft.com/he-IL/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=9dbbeeb0-9fa4-4db8-b44f-db1007a20a2f&productEditionId=1060&sdVersion=2
Querying https://www.microsoft.com/he-IL/api/controls/contentinclude/html?pageId=cfa9e580-a81e-4a4b-a846-7b21bf4e2e5b&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=GetProductDownloadLinksBySku&sessionId=9dbbeeb0-9fa4-4db8-b44f-db1007a20a2f&skuId=7618&language=Hebrew&sdVersion=2
Downloading https://software-download.microsoft.com/pr/Win10_1809Oct_Hebrew_x64.iso?t=cfa3821f-a837-4c47-bf26-3f4e30f51366&e=1552603555&h=040fe3c1e47454c12048daf31756f8d3
File length: ‎4.4‎ GB
Download cancelled by user
```

ItielMaN, 2019-03-13:
[image]
Also, the progress bar in the app itself should probably reset (?).

pbatard, 2019-03-13:
I'm afraid this is intentional. User cancel translates to orange, which means intentional error (there is no definite color for paused, even if some application __might__ use the orange color too). Download error translates to red. This way, user initiated vs non initiated download interruptions are very apparent.

Also, not resetting is intentional. If I were resetting, users would be confused as to whether the download completed on cancel or error. This way, with progress bar clearly not complete, it is very explicit.

Closing as "100% intentional".

closed 2019-03-13 by pbatard