pbatard/rufus #1283: Fido.ps1 silently fails on fully upToDate Windows 7

lukaszgo1, 2019-03-16:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
With the Rufus 3.5 beta I've tried to download a Windows ISO on my fully uptodate Windows 7 pro x64 machine, but nothing happens. Status bar stays at Downloading Fido.ps1.
According to Fido readme it might be caused by  too old version of Powershell and log below seems to be confirming it. The same readme says, however that this fact would be communicated to the user.
I am not native English speaker but my understanding of the word 'anilla' is clean without updates which cannot be said about my Windows 7.

Log
---
```

```
Rufus x86 v3.5.1473 (Beta)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0415 (pl-PL)
Will use default UI locale 0x0415
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
NOTE: The Polish (Polski) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.06/Fido.ps1
File length: 27.2 KB
Successfully downloaded 'Fido.ps1'
Signature is valid ✓
Please Wait...
Checking for Rufus updates...
Checking release channel...
No new release version found.
Error: PowerShell 3.0 or later is required to run this script.
Method invocation failed because [System.String[]] doesn't contain a method nam
ed 'Contains'.
At C:\Users\Admin\AppData\Local\Temp\{D3F5BFFC-A7AF-402D-8943-8012E2E7C11C}.ps1
:335 char:28
+     if (-not $English.Contains <<<< ($Text)) {
    + CategoryInfo          : InvalidOperation: (Contains:String) [], ParentCo 
   ntainsErrorRecordException
    + FullyQualifiedErrorId : MethodNotFound
Exited download script with code: 0

pbatard, 2019-03-16:
Thanks for the report.

Actually, the script appears to be failing at the prompt that is supposed to tell you that PowerShell 3.0 is required and that would open a browser to where you can download it.

The problem however seems to be that PowerShell 2.0 doesn't even have the basic function other versions of PowerShell can use to locate a substring within a string.

I'll see what I can do to work around that...

pbatard, 2019-03-16:
If you try again, you should see that this is fixed now (you should now get a prompt telling you that PowerShell 3.0 is required and asking you to download it &mdash; remember that Windows 7 is version `6.1`).

I will close this issue, but if you still see a problem, you can still comment on it.

closed 2019-03-16 by pbatard