pbatard/rufus #1284: Rufus directly moves to status 'Ready' when used from a WinPE USB stick. Neither it generates a 'Log' button nor writes anything to the USB stick

sunayansaikia, 2019-03-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm running Rufus from a Win10 PE live USB stick (source usb drive - 1st one). I can access the target USB drive (2nd one) fine and so does Rufus. But when I try to create a bootable, Rufus directly moves to status 'Ready' (All options already tried). Neither it generates a 'Log' button (for me to debug) nor writes any files to the USB stick. What I saw in the UI is that it tries to delete and create partitions and then something happens, and it directly moves to 'Ready' state.

Log
---
*** NO logs generated
```
<FULL LOG>
```

pbatard, 2019-03-17:
1. If you see the Rufus UI then there __IS__ a log generated (and there IS a log button &mdash; it is __ALWAYS__ available), especially as you mention that you saw it delete and create partitions, that you can access by clicking the log button (last small button left of _START_) and I __ABSOLUTELY NEED TO SEE IT__! Next time, please try to read the checklist, as it tells you exactly how to access the log.

2. You may want to try with the 3.5 BETA which you can download [here](http://rufus.ie/downloads/)

sunayansaikia, 2019-03-17:
_I ABSOLUTELY NEED TO SEE IT!_

I did go through the checklist.
**Please check the screenshot attached.** 
@pbatard I think there is a bug in the code and it's not checking the prerequisites before launching the GUI.
And, moreover, there is some kind of **faulty exception handling** that's letting it declare the whole process successful (with status = 'Ready') even though the process didn't finish successfully.

[image]

pbatard, 2019-03-17:
That's exceedingly weird. It means all the Ui controls display expect the ones with the log button. That's  unheard off (at least, in millions and millions of downloads you are the very first person to report these controls not showing). Can you show me a screenshot of what you get when you click "Show advanced drive properties".

Also, even if the log button doesn't show, Rufus sends its log output to the debug facilities, so can you please install and run DebugView from [here](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview) and do what you were trying to do, then provide the log from DebugView?

I really can't help you without the log, and there are more than one way to get to it, even though your missing of the log button is exceedingly puzzling.

Also, I will need to know __excatly__ how you created your Win10 PE live USB stick and from what image (with a link to that image).

pbatard, 2019-03-17:
Oh, by the way, I just remembered that I also added a <kbd>Ctrl</kbd>-<kbd>L</kbd> shortcut to access the log, so you should be able to get to that data even without installing DebugView.

sunayansaikia, 2019-03-19:
@pbatard Thanks. I'm able to extract the log. It says'unable to locate FormatEx() in Fmifs.dll: [0x0000007F] The specified procedure could not be found'. 
I'm using Hiren Win10 PE.
I have read your article it says -> **_where you mentioned it's not a public API to format a drive, and instead you had to hijack an undocumented one called FormatEx?_**
So, how do I fix this issue? Get Fmifs.dll? Any other dependencies I need to get this working?

*LOG:*
Rufus x86 v3.4.1430
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: CoCreateInstance failed; hr = 8007007e
Localization set to 'en-US'
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (058F:6387)
Using autorun.inf label for drive H: 'HBCD_PE_x64'
Removing H: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1883, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {AC095035-A1F6-44A6-9138-9B703CB59FF3}
Max parts: 128, Start Offset: 17408, Usable = 15489530368 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {81ED3A19-ED85-446B-B33E-CD859FDAD7EE}
  Size: 14.4 GB (15488499200 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 4.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1709_English_x64.iso (4.4 GB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 14.4 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Unable to locate FormatEx() in Fmifs.dll: [0x0000007F] The specified procedure could not be found.
Format error: Success.

Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (058F:6387)
Using autorun.inf label for drive H: 'HBCD_PE_x64'
Removing H: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1883, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {11D60D55-DE1A-408C-AA03-CC1CBC7DB5B0}
Max parts: 128, Start Offset: 17408, Usable = 15489530368 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4A9FD826-A884-4C51-80B7-4DC25A23EDA8}
  Size: 14.4 GB (15487428096 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {F3713B4B-8E2C-44A0-A209-E3BB71DB48C9}
  Size: 512 KB (524288 bytes)
  Start Sector: 30250931, Attributes: 0x0000000000000000

pbatard, 2019-03-19:
Thanks. I'm going to assume that you are using `HBCD_PE_x64.iso` from [here](https://www.hirensbootcd.org/download/) with SHA-1 `6CBC2FFFB4D0563852A43F7E16A67581BD6BD809`.

By the looks of it, it appears that the people behind Hiren have forcibly removed Windows DLLs, such as the one that provides the __standard__ UI toolbar elements for the log and other buttons (it seems that the hash button which is a regular toolbar also doesn't appear in your version) as well as the one that provides formatting (it's not because it isn't a public API that it should be removed).

As I hope you can appreciate, I can provide no guarantee whatsoever that Rufus can run properly on versions of Windows where critical DLLs have been hacked away, and I would strongly encourage you to report that issue to the makers of Hiren, so that they stop being that cavalier about ditching DLLs that 
_they think_ are not needed...

The only thing I may try to do is report a formal error if the DLL is missing, as it doesn't seem Rufus is doing that at the moment, and I'll also see if I can figure out if I can detect when toolbar controls don't appear (which may be a lot more tricky to do).

Apart from that, I'm afraid that you are on your own. I simply can't support custom unofficial versions of Windows where DLLs appear to have been removed at random, and you should ask the people behind those versions for support.

sunayansaikia, 2019-03-19:
Many thanks! Yes, if it's the PE's problem, let it be. It's interesting that Hiren's CD bundles 'Rufus'. It may be that, it's never tested. 

Will check 'Windows to go' like you mentioned.

Anyway, please continue improving Rufus. This tool has helped me many times and thanks for coming up with such an open-source project like this.

closed 2019-03-19 by sunayansaikia

pbatard, 2019-03-19:
Thanks for understanding. I have just added a fix to Rufus to make sure it doesn't fail silently on that `[0x0000007F]` error, but obviously there's not much I can do to work around the missing DLLs.

In the long term, I may try to use [VDS](https://docs.microsoft.com/en-us/windows/desktop/vds/about-vds) for formatting instead of `fmifs.dll` which _might_ work with PE environments. It will be a while before I look into that though...

sunayansaikia, 2019-03-20:
Thanks again. I appreciate your efforts. Keep on developing and rocking. 👍