pbatard/rufus #1285: Cannot mount GUID volume

Siber92, 2019-03-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I'm getting a similar issue as these past threads
https://github.com/pbatard/rufus/issues/1016
https://github.com/pbatard/rufus/issues/476
That is "Cannot mount GUID volume" Using the USB 3 port.>

Log
---
```
<Rufus x86 v3.4.1430
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x041D
Will use default UI locale 0x041D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'sv-SE'
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x242DFE6F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 119.2 GB (128043643392 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 250085304, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_SV-SE_DV9'
  Size: 4.6 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: sv_windows_10_consumer_edition_version_1809_updated_sept_2018_x64_dvd_fb769d22.iso (4.6 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1\sources\install.wim:[0] (API)
Extracting: C:\Users\sibe1\AppData\Local\Temp\Ruf61AF.tmp (From [1].xml)
Closing: \\.\CDROM1\sources\install.wim
Will use 'Windows 10 Pro' (Build: 17763, Index 5) for Windows To Go

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive2':
? Partition 1 (offset: 32256, size: 119.2 GB)
? Partition 2 (offset: 128043675648, size: 31.5 KB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 976895 (1 block = 128 KB)
Bad Blocks: Using offset 125016 for fake device check
Bad Blocks: Writing test pattern 0x00
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{674b9271-4b28-11e9-a22e-8c89a5e7829c}\
F:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{674b9271-4b28-11e9-a22e-8c89a5e7829c}\, got \\?\Volume{674b9275-4b28-11e9-a22e-8c89a5e7829c}\
Could not remount \\?\Volume{674b9271-4b28-11e9-a22e-8c89a5e7829c}\ on F:\: [0x000000EA] Det finns mera data tillgängliga.
F:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{674b9271-4b28-11e9-a22e-8c89a5e7829c}\, got \\?\Volume{674b9275-4b28-11e9-a22e-8c89a5e7829c}\

Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000D4305
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 119.2 GB (128041579008 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 250083257, Boot: No>
```

Siber92, 2019-03-20:
Update: Seems that turning off "check for bad blocks" made it work.

pbatard, 2019-03-21:
Looks like a transient error that you figured out how to fix on your own. As such I will close this issue.

closed 2019-03-21 by pbatard