pbatard/rufus #1286: Rufus and universal usb installer both broke windows 7 its unusable anymore there are too much leftovers and errors for normal operations

loa11, 2019-03-20:
Rufus and universal usb installer both broke windows 7 its unusable anymore there are too much leftovers and errors for normal operations.

First thing that's broke is that there is broken performance  and registry hive both problems pointing to the usb stick which is carrying Windows 7 sp1 retail usb created with the Rufus.

Windows detected your registry file is still in use by other applications or services. The file will be unloaded now. The applications or services that hold your registry file may not function properly afterwards.  

 DETAIL - 
 15 user registry handles leaked from \Registry\User\S-1-5-21-3838845324-2666958239-4212638696-1001:
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\SmartCardRoot
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\Root
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\My
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\CA
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\trust
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\TrustedPeople
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Microsoft\SystemCertificates\Disallowed
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Policies\Microsoft\SystemCertificates
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Policies\Microsoft\SystemCertificates
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Policies\Microsoft\SystemCertificates
Process 460 (\Device\HarddiskVolume5\Windows\System32\lsass.exe) has opened key \REGISTRY\USER\S-1-5-21-3838845324-2666958239-4212638696-1001\Software\Policies\Microsoft\SystemCertificates

Event filter with query "SELECT * FROM __InstanceModificationEvent WITHIN 60 WHERE TargetInstance ISA "Win32_Processor" AND TargetInstance.LoadPercentage > 99" could not be reactivated in namespace "//./root/CIMV2" because of error 0x80041003. Events cannot be delivered through this filter until the problem is corrected.

WMI performance issue is somehow correctable but question is what else is broken and not so obvious like above two problems.

Now the question is from which version did you made some extreme changes making the rufus to be erroneous.
Seems windows is now thinking that some parts of usb installer is a root for windows.
Device\HarddiskVolume5 is usb stick.

pbatard, 2019-03-21:
Sorry, but it's not clear at all what you are asking and you also didn't bother to follow the check list, which makes it very difficult to try to help you. For instance, you don't mention the Windows 7 ISO you used to create the USB, and there are quite a few of them. Can you at least provide its SHA-1 so that we can validate that it is an official retail version?

You also seem to claim that 2 completely different and independent applications are somehow behaving in the same fashion and responsible for the same issue, which is a much extraordinary claim (I have ZERO relation with the developers of Universal USB Installer, and I certainly did not look at their source code, let alone reuse it for anything that Rufus does). Logically then, even if you may not want to hear it, the problem is most likely linked to your environment.

Especially, Rufus does __not__ access the USB drive after it has copied files over it. Especially, there is exactly __NOTHING__ in the code (which anybody can verify) that will access the registry hive residing on the USB. Therefore, whatever access you see there is coming from a different application, which you will need to identify.

On the other hand, so that it can __temporarily__ disable AutoPlay as well as read its users settings, Rufus does access the local registry hive on the system drive (i.e. NOT the USB drive, and unless Windows itself has a major bug, even a bug in the Rufus source would not result in the application erroneously accessing the USB registry hive instead of the local one), but it does so in as safe a manner as possible (especially AutoPlay is toggled through Local Group Policy settings, which is designed by Microsoft to be resilient and not leave the system in limbo in case of an issue), and none of the registry keys you list are even remotely susceptible to be accessed directly from our code. If anything, since they are certificate related, they might be invoked during digital signature verification (which is first performed by Windows when it launches Rufus, and which may also be requested by Rufus when it downloads an update), but you need to understand that, when that happens, it is Windows itself that is carrying out the process, not the application, and we can exert no control on how Windows __chooses__ to leverage the registry during that process.

Finally you are talking about "Windows 7 sp1 retail usb created with the Rufus", but you don't indicate whether you created an installation media or a Windows To Go media (bearing in mind that Windows To Go for Windows 7 is provides "as is" as it was only formally introduced by Microsoft for Windows 8, and the option is only being provided in Rufus as a courtesy because there doesn't seem to be anything in the Windows 7 ISOs preventing its usage, but that certainly does not mean that Microsoft did design Windows 7 to work in Windows To Go mode, and we offer no warranty whatsoever with regards to the ability of a Windows 7 Windows To Go drive of running as expected).

> Now the question is from which version did you made some extreme changes making the rufus to be erroneous.

I think the better question is what exactly makes you assert this, and declare that Rufus __must__ be the culprit. Be very mindful that "innocent until proven guilty" is not a paradigm that applies only to individuals, but it also applies to applications, and, unless you can produce __conclusive proof__ that Rufus is responsible for your registry issue (especially considering that you report certainly does not mention Rufus by name but instead states `other applications or services`, which can literally mean any of the dozens of applications and services that run on Windows at any one time), I will have no choice but to dismiss your report as __baseless accusation__, especially as you also provide no indication of where exactly the message about registry use originates from.

If you want to be taken seriously, and especially if you are inclined to point the finger at a specific application or say that is is "broken", please provide more comprehensive details to corroborate your claims.

pbatard, 2019-03-21:
The log from Rufus, which is also __explicitly__ requested in the check list would also help, as I have a feeling that you may be playing with the creation of a Windows To Go drive... Please provide a full log of the operation you were trying to complete so that I can at least have a better idea of what you were trying to accomplish and how.

loa11, 2019-03-21:
---------------------------------------------------
BUILD: 7601.17514.101119-1850
FILE: 7601.17514.101119-1850_x64fre_client_en-us_Retail_Ultimate-GRMCULXFRER_EN_DVD.iso
SIZE: 3,319,478,272 byte
SHA-1: 1693B6CB50B90D96FC3C04E4329604FEBA88CD51
MD5: 56A26636EC667799F5A7F42F142C772D
CRC: 8589EE18
----------------------------------------------------------------------

No other application or drivers were installed on the windows 7 pre or after the installation, those errors revealed themselves just after the first restarts. Also wmi errors were traced back to same volume5 which is ejected from the system right after the extraction was finished in the first turn of windows 7 installation process.
-----------------------------------------------

Here is what I did I had unformatted SSD kingston 240gb on the sat1 port, its set ans first hard drive, I created on the main pc bootable usb first with rufus and after that with universal usb installer as an bootable usb installer and not a  Windows To Go drive.

I prevented any windows update option in the installation process I didn't even made windows activation I skipped it.
After installation was successful, I restarted few times to observe the event logs and that the moment I saw the errors. I would think as everyone that such errors were caused by the bad hardware or by the bad installation but the culprit must be rufus and universal usb installer.why would I have hives on volume5 and windows thinking that its located in some parts on the usb stick? Think logically what could cause this?
I know that this version of windows is ok but ok you maybe right, I will test it with other version, do you have link to some retail windows 7 version that you know it is ok? 

Let me check the check marks again.

loa11, 2019-03-21:
Rufus x86 v3.4.1430
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'WDC WD10 01FALS-00E3A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3839, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00003481
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.3 GB (31470911488 bytes)
  Start Sector: 2048, Boot: No
Partition 2:
  Type: FAT32 (0x0b)
  Size: 100 MB (104857600 bytes)
  Start Sector: 61468672, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCULXFRER_EN_DVD'
  Size: 3.1 GB (Projected)
  Uses: EFI (win7_x64)
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 7601.17514.101119-1850_x64fre_client_en-us_Retail_Ultimate-GRMCULXFRER_EN_DVD.iso (3.1 GB)
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data

Format operation started
Requesting disk access...
Will use 'Q:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive8':
● Partition 1 (offset: 1048576, size: 29.3 GB)
● Partition 0 (offset: 31471960064, size: 100 MB)
Opened \\.\PhysicalDrive8 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 1 tracks (31.5 KB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{946d336a-699f-11e8-bd98-faeabaefeaaa}\
Disabling file indexing...
[... 924 lines cut ...]
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3839, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x27E8D1C6
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.4 GB (31583043072 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 61685694, Boot: No
Found non-USB removable device 'WDC WD10 01FALS-00E3A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
0 devices found
Found non-USB removable device 'WDC WD10 01FALS-00E3A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
0 devices found
Found non-USB removable device 'WDC WD10 01FALS-00E3A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
0 devices found
Found non-USB removable device 'WDC WD10 01FALS-00E3A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
0 devices found

loa11, 2019-03-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

loa11, 2019-03-21:
Sorry that this thread is in revers the check list as last loh as second last and issue as the first one.

pbatard, 2019-03-21:
> I created on the main pc bootable usb first with rufus and after that with universal usb installer as an bootable usb installer and not a Windows To Go drive.

Okay, so, twice, you installed Windows 7 on your Kingston SSD from a Windows 7 USB installer created from `7601.17514.101119-1850_x64fre_client_en-us_Retail_Ultimate-GRMCULXFRER_EN_DVD.iso` first using Rufus and then Universal USB Installer, and in both instances ran into the same error (reported in the event log if I understand you correctly and which doesn't seem to affect Windows 7 otherwise).

> but the culprit must be rufus and universal usb installer.

Again, if two completely independent applications, which copy all the files from the ISO __as is__ (the only trick to creating a Windows 7 installation flash drive is installing the MBR and PBR so that the drive can boot in BIOS mode, but that's about 1 KB of fixed data, that also comes from Microsoft and isn't altered), that assertion is __exceedingly__ unlikely.

> why would I have hives on volume5 and windows thinking that its located in some parts on the usb stick? 

This is not what your report states. Your report states that the process `lsass.exe` (Local Security Authority Subsystem Service) which is located on `\Device\HarddiskVolume5` in the directory `Windows\System32\` is accessing the system registry hive. And yeah, I made the same mistake as you when I read your report, in thinking it means that there's a registry hive residing on `\Device\HarddiskVolume5`, but that's not something that can actually be inferred from what the report states.

Now, if you look at the log you just pasted from Rufus, you'll notice that at no point does Rufus create a `Windows\System32\lsass.exe` file on your USB drive. It doesn't even create a `Windows\System32\` directory (and I am certainly not aware of the Windows installation process altering the USB drive which it uses as it source during installation, especially as it is designed to treat it in the exact same way as a __read-only__ DVD).

So the only thing I can conclude is that, despite what you are asserting, if there exists a `Windows\System32\lsass.exe` on `\Device\HarddiskVolume5` then `\Device\HarddiskVolume5` is __NOT__ your USB drive.

Most likely this is actually your system disk (`C:`) and you need to be mindful that Windows makes no guarantee that the system disk will always be `HarddiskVolume1`.

I guess one way you can find what drive `\Device\HarddiskVolume5` really is is to launch `diskpart` from a command prompt and then issue the command `list vol`. This should provide the list of volume numbers along with the drive letter that is attached to them.

Can you do that and report the drive letter you see for `Volume 5`?

loa11, 2019-03-21:
Look this is list volume without usb stick and just after I added back the usb stick.
-----------------------------------------------
Microsoft DiskPart version 6.1.7601
Copyright (C) 1999-2008 Microsoft Corporation.
On computer: DEJAN-PC

DISKPART> list volume

  Volume ###  Ltr  Label        Fs     Type        Size     Status     Info
  ----------  ---  -----------  -----  ----------  -------  ---------  --------
  Volume 0         System Rese  NTFS   Partition    100 MB  Healthy    System
  Volume 1     C                NTFS   Partition    223 GB  Healthy    Boot
  Volume 2     D   System Rese  NTFS   Partition    100 MB  Healthy
  Volume 3     E                NTFS   Partition     93 GB  Healthy
  Volume 4     F   Skladiste    NTFS   Partition    372 GB  Healthy

DISKPART> list volume

  Volume ###  Ltr  Label        Fs     Type        Size     Status     Info
  ----------  ---  -----------  -----  ----------  -------  ---------  --------
  Volume 0         System Rese  NTFS   Partition    100 MB  Healthy    System
  Volume 1     C                NTFS   Partition    223 GB  Healthy    Boot
  Volume 2     D   System Rese  NTFS   Partition    100 MB  Healthy
  Volume 3     E                NTFS   Partition     93 GB  Healthy
  Volume 4     F   Skladiste    NTFS   Partition    372 GB  Healthy
  Volume 5     G   GRMCULXFRER  NTFS   Removable     29 GB  Healthy
-----------------------------------------------------

Also look at some parts of windows registry :
---------------------------------------------------
Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Control\hivelist]
"\\REGISTRY\\MACHINE\\HARDWARE"=""
"\\REGISTRY\\MACHINE\\BCD00000000"="\\Device\\HarddiskVolume4\\Boot\\BCD"
"\\REGISTRY\\MACHINE\\SYSTEM"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\SYSTEM"
"\\REGISTRY\\MACHINE\\SOFTWARE"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\SOFTWARE"
"\\REGISTRY\\USER\\.DEFAULT"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\DEFAULT"
"\\REGISTRY\\MACHINE\\SECURITY"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\SECURITY"
"\\REGISTRY\\MACHINE\\SAM"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\SAM"
"\\REGISTRY\\USER\\S-1-5-20"="\\Device\\HarddiskVolume5\\Windows\\ServiceProfiles\\NetworkService\\NTUSER.DAT"
"\\REGISTRY\\USER\\S-1-5-19"="\\Device\\HarddiskVolume5\\Windows\\ServiceProfiles\\LocalService\\NTUSER.DAT"
"\\REGISTRY\\MACHINE\\COMPONENTS"="\\Device\\HarddiskVolume5\\Windows\\System32\\config\\COMPONENTS"

-----------------------------------
As a remedy I could edit in the registry all vlolume5 to volume1 and volume4 to volume0 but its awful that this happen.

loa11, 2019-03-21:
Also I need to find a way to change part in registry where volume5 is manifested as classes.

pbatard, 2019-03-21:
Well, I have no idea what created `Windows\System32` on your USB drive, but I can guarantee that it has nothing to do with Rufus. Rufus absolutely does not alter the installation files it copies from the ISO, and it doesn't create that directory either, so it has to come from either the Windows installation process or Windows itself after it is installed.

If your machine is UEFI based, I am pretty sure that you will see the exact same issue if you install Windows 7 outside of using Rufus or UUI by:
- Formatting your USB as a single FAT32 partition
- Extracting all the files from your ISO onto that partition using something like 7-zip

That's really all you need to do to install in UEFI mode, really (and as I mentioned, that's also pretty much all Rufus does behind the scenes), so, unless your machine is BIOS based, you should be able to remove Rufus from the equation altogether and confirm that this is a pure Microsoft issue.

If you do just that, you will have demonstrated that there is nothing that Rufus or UUI do that has any influence on the creation and later usage of `Windows\System32` from a removable flash drive. In other words, as much as I'd like to help, unless you can prove otherwise, you will have to ask Microsoft themselves what is going on with the creation of `Windows\System32` during a standard Windows 7 installation, because unless proven otherwise, I really can see no way for Rufus or UUI to be implicated in that, and, obviously, I don't provide support for Microsoft's weird installation behaviours.

loa11, 2019-03-21:
pbatard could you make rufus version which could force usb stick to always take the last volume position and to never take the first volumes, neither in boot process and neither in the windows installation process, as I see usb stick is switching volumes positions with the system drive.

pbatard, 2019-03-21:
Nope. The volume attribution is determined by Microsoft's internal Windows behaviour and is not something that can be altered. You get what Microsoft gives you and there is no API I know of that allows you to change the order of Volume attribution (you can change drive letters easily, but not volume numbers).

Also something I find very strange is that your `Boot\BCD` is on your `F:` drive.

All in all, it looks to me like you are carrying out a very nonstandard installation of Windows 7...

loa11, 2019-03-21:
In actuality boot is on volume0 and there is no trace on volume4 also on the usb stick which is volume5 there is no trace of Windows\System32 
G:\>dir
 Volume in drive G is GRMCULXFRER_EN_DVD
 Volume Serial Number is A03A-B5FC

 Directory of G:\

03/20/2019  07:46 PM               122 autorun.inf
03/20/2019  07:46 PM    <DIR>          boot
03/20/2019  07:46 PM           383,786 bootmgr
03/20/2019  07:46 PM           669,568 bootmgr.efi
03/20/2019  07:46 PM    <DIR>          efi
03/20/2019  07:46 PM           106,768 setup.exe
03/20/2019  07:50 PM    <DIR>          sources
03/20/2019  07:51 PM    <DIR>          support
03/20/2019  07:51 PM    <DIR>          upgrade

               4 File(s)      1,160,244 bytes
               6 Dir(s)  28,144,807,936 bytes free

The actual files were they should be but in the installation process the windows were feed with the wrong volume information.

This is some older motherboard that I wanted to use for some things Asus M2N68 with some older Amd x2 64 5600+. 
I think this motherboard doesn't have uefi.

I need to read all about were the installation volumes were located and how internally windows is getting itself volume position.

loa11, 2019-03-21:
I think for now I will burn the dvd and finds in some of my drawers the dvd drive itself because  I didn't used DVD for years.

pbatard, 2019-03-21:
> In actuality boot is on volume0

I'm just quoting what I see from what you copy/pasted:
```
"\\REGISTRY\\MACHINE\\BCD00000000"="\\Device\\HarddiskVolume4\\Boot\\BCD"
````

Now, one thing that could explain all the above is that this data comes from the initial pre-installation environment, which would have seen stuff such as boot records and registry hives residing on real or virtual drives that happen to share the same volume number as your USB drive has after installation.

But that's still a pure Microsoft issue. Neither Rufus nor UUI would have any kind of incidence on that, and I think you went with the wrong assumption that Rufus was doing something complex to create a working pre-installation environment (such as modifying files or creating registry hives), whereas nothing could be further from the truth and all that happens, really, apart from creating a couple of boot loaders that simply can not have any kind of incidence on what you report is Rufus is just going through a 1:1 file copy of the content that exists on the DVD. Therefore, as soon as the pre-installation launches, any issue that happens is 100% Microsoft's responsibility.

I will therefore close this issue, as it __is__ unrelated to Rufus and, regardless of how you installed Windows, you really should contact Microsoft support about it as they are the __ONLY__ people who can actually hep you there.

closed 2019-03-21 by pbatard

loa11, 2019-03-21:
I installed it from DVD, it populated the hives and volumes exactly, the only downside is if I would take the dvd drive from the system, because Microsoft decided that dvd drive should take volume 0 instead of the last volume in the bunch, hives and volumes would move for one number and I would face the same problem, but I will somehow leave the dvd inside of that small case I don't need it to be in the slot. I'm now talking with the creator of EasyBCD could he make complete tool to fix this mistake.
I would bet that very small percentage of PC's on our planet which are based on Microsoft systems have exact hives volumes pats.