pbatard/rufus #1289: During pendrive creation progress bar is update only visually, but actual control stays at 0%

lukaszgo1, 2019-03-25:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
When creating a bootable penndrive from an iso image Rufus probably displays a progress bar, but the progress bar control isn't updated. I said probably because I am blind therefore I cannot confirm that something in deed changes. I can however using object navigation in my screen reader NVDA acces progress bar control, and it is always at 0%.

lukaszgo1, 2019-03-25:
This worked in Rufus 2.18 and stopped in 3.0 version.

pbatard, 2019-03-25:
Yes, I was afraid I would get a comment like yours.

The issue is that, with the 3.x UI redesign, we merged the progress and info controls, which means that not only does the new "progress bar" display progress, but it also displays informational messages.

However, since Microsoft never designed win32 toolbars to display overlay text (if you try it, it will flicker wildly and look atrocious), and we also want to have an effect whereas the superimposed text colour in the complete and incomplete parts of the toolbar are different (so that it remains readable on both sides), we have to use custom paint messages to create the progressing rectangle, which of course means that screen readers will fail to register the control as an actual progress bar.

I can see a few solutions to try to work around that (e.g. adding a hidden toolbar, though I'm not sure it would be registered by screen readers), but I'm also seeing a lot of drawbacks/added complexity in doing just that, and they all seem like they would take a lot of time, which I don't really have, to implement.

Maybe when I manage to get higher priority stuff out of the way, and I manage to find time to spare, I may try to have another look at it, but, as much as I'd like to be able to state otherwise, I'm afraid that this is not something I am planning to revert or go out of my way to solve for the time being, so I will have to ask you to find other means to see if Rufus has completed its task.

For one thing, Rufus also sends its progress to the taskbar, and it does so using standard Windows APIs, so if your screen reader can read the taskbar progress (which it should, as, for instance, browsers tend to only display progress for downloads there by default), you should be able to obtain the progress there. Also, Rufus does produce a beep when it completes its task or when it fails, which I am also hoping screen readers can register with an indication of the originating application. While it is of course not as good as a progress, it should at least tell you when Rufus is done.

Satisfying the majority of users with a long awaited modernization of the interface is unfortunately not entirely compatible with satisfying the minority that might use screen readers. For instance, using overlay text on progress is something almost all modern applications will use and that users have come to expect (plus it does save on vertical space, which was my prime motivation to go that route). However, the problem is that the basic Win32 APIs that Rufus uses are too old to do that natively (hence the need to pain the control ourselves) and, while I did consider it when designing the new UI, I estimate that rewriting Rufus to use WPF, XAML or what have you, to use native modern controls, would have required me to double the time I spent on the UI redesign, which is already way longer than what I would have liked in the first place (it basically took me 3 full months to go through that, and it was a major ordeal to put everything else on standby just for the sake of improving the UI). That's not to say that I will not work on improving the application for visual impaired users, but I'm afraid I have to strike a balance somewhere, and trying to satisfy everyone can be very challenging. So really, the best I can say is: I'll see what I can do, but please don't expect anything fast...

lukaszgo1, 2019-03-25:
@pbatard wrote:
> which of course means that screen readers will fail to register the control as an actual progress bar.

As I've stated it is registered as a progress bar, the only problem is with the fact that the value never changes.

pbatard, 2019-03-25:
What I meant is, screen readers will fail to register the progress bar changes because we are drawing a custom progress over it. And of course, we don't want to tell the progress bar to update its standard progress value because otherwise we'd have 2 conflicting progresses being drawn together (including one without the overlay text) and flickering madly on screen.

closed 2019-06-14 by pbatard

lukaszgo1, 2019-07-30:
@pbatard I've tested your implementation and it works wonderfully. It is quite unusuall for an open source developer to consider accessibility related requests, so just wanted to say thanks.

pbatard, 2019-07-30:
Thanks for testing. I'm very happy to hear that the fix worked.

commit 24e8307d (2019-06-14): [ui] update progress bar position for screen readers

* Make sure to call the original winproc on PBM_SETPOS and friends.
* Closes #1289