pbatard/rufus #1292: Windows 7 ISO BIOS or UEFI

T15K6, 2019-03-26:
Hello, i use my USB flash drive with rufus and install Windows 7 x64 PRO iso 
use this options 
Partition scheme: MBR
Target System: BIOS or UEFI
my problem is i cant booth when i choose boot in UEFI i get FAIL
[FAIL] Could not locate  '\efi\boot\bootx64.efi' : [14] Not Found
here is log file
[rufus.log](https://github.com/pbatard/rufus/files/3009984/rufus.log)

pbatard, 2019-03-26:
Thanks for the report.

A few things first:
1. You have used the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode to enable dual BIOS + UEFI boot (because that's the only way you could have selected the options you describe) so you need to be mindful that cheat modes comes with no warranty and may have bugs.
2. There is no reason for you to use NTFS for your Windows installation media. There is no file larger than 4 GB on your ISO and, especially if you want a dual BIOS + UEFI bootable Windows 7 installation media, using FAT32 is the file system that will give you greater compatibility (if it was geared for BIOS-only installations that would be different but that doesn't seem to be your case). Especially the file system used by the installation media has no incidence on the file system that Windows will effectively use during installation, so, if your worry is that you should use NTFS on the installation media because you want NTFS on the version of Windows you install, that's a non issue, and there's no drawback that comes from using FAT32 here.

This being said, I think I have an idea about what's going on. One thing you should know is that Windows 7 x64 is only "half" compatible with UEFI, meaning that Microsoft did add UEFI compatibility, but they did it in a half-assed way because they didn't put the `\efi\boot\bootx64.efi` on the ISO itself, but instead they shoved it inside `\sources\install.wim`, so, to make a Win7 x64 compatible with UEFI boot, an extra operation is needed to extract that file from `\sources\install.wim` and copy it on the USB.

If you don't use <kbd>Alt</kbd>-<kbd>E</kbd> and choose _GPT_ for _UEFI (non-CSM)_ you will see the following being carried out at the end of the log:
```
(...)
Extracting: F:\upgrade\netfx\netfx.msp (18.3 MB)
Extracting: F:\upgrade\netfx\netfx1.cab (16.1 MB)
Extracting: F:\upgrade\netfx\netfxupdate.exe (72 KB)
Win7 EFI boot setup (This may take a while)...
Opening: F:\sources\install.wim:[1] (7-Zip)
Extracting: F:\efi\boot\bootx64.efi (From Windows\Boot\EFI\bootmgfw.efi)
Finalizing, please wait...
F:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...
(...)
```

However, when using <kbd>Alt</kbd>-<kbd>E</kbd>, there appears to be a bug in the logic I have for the extraction of `\efi\boot\bootx64.efi`, and that step does not occur, hence your error.

Unfortunately, this seems to happen for both NTFS or FAT32, so the only workaround I can see for your is either to disable <kbd>Alt</kbd>-<kbd>E</kbd> and create your bootable media for either BIOS or UEFI (but not both), or, using something like 7-zip to extract `Windows\Boot\EFI\bootmgfw.efi` from `sources\install.wim` and copy it as `\efi\boot\bootx64.efi` (which is the step that is currently missing in Rufus).

Of course, I'll see what I can do to fix that issue in a future version of Rufus (I'll see if I can do that in the upcoming 3.5 which I'm planning to release in a couple of days but I can't promise anything). Thanks for reporting it!

T15K6, 2019-03-26:
Thanks for help, i use costum install.wim its 13GB. i do what you say copy this file "bootmgfw.efi" from install.wim\Windows\Boot\EFI\ to USBdrive\efi\(create boot folder)\boot\ and rename bootmgfw.ef to bootx64.efi and its work! thank you <333333

commit 661ba7b6 (2019-03-27): [core] fix Win7 EFI boot file extraction when dual BIOS+UEFI is in use

* Closes #1292
* Also add "RISD" (Ricoh IDE SD Card reader) to list of card readers (#551)
* Also make running Fido script read-only
closed 2019-03-26 by T15K6