pbatard/rufus #1295: Error: Error while partitioning drive

JaxenSounds, 2019-04-03:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Nothing running is conflicting with Rufus (i read the FAQ)

Log
---
```
Rufus x86 v3.5.1497
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x789E7C51
Drive has a Zeroed Master Boot Record
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 4.0 GB (Projected)
  Note: File on disk is larger than reported ISO size by 526 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.0 GB)

Computing checksum for 'E:\Windows.iso'...
  MD5:	 81dad1e26ecf44b9a0b45ef7b7287d0f
  SHA1:	 28d85da550e3352146a1706183eab69159fbb443
  SHA256: 60bf9f7cf2391c7370d3cc69f01a211b4c4ada65f6774b3456cd4053362412aa

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
No partition to delete on disk '\\?\PhysicalDrive3'
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x789E7C51
Drive has a Zeroed Master Boot Record
```

pbatard, 2019-04-03:
I'm afraid this is answered in the FAQ:
https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied

There's nothing I can do for you here - you will need to find what application is preventing access.

closed 2019-04-03 by pbatard

najafnaqvi471, 2019-04-03:
hi frnds
 Error while partitioning drive
what is dis

pbatard, 2019-04-03:
Check your log.