pbatard/rufus #1296: French translation

Atelier-Made-in-iKi, 2019-04-05:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi
just a little correction: in french langage, in second listbox
"Veuillez sélectioner" is wrong
correct: "Veuillez sélectionner" 
thanks for this great software
Jean-Philippe

Log
---
```
Rufus x86 v3.5.1497 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
Found USB 2.0 device 'SanDisk Extreme SSD USB Device' (0781:558C)
Removing G: from the list: This is the disk from which Rufus is running!
Found USB 2.0 device 'WDC WD25 00BEVT-22ZCT0 USB Device' (04FC:0C25)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: FIXED, Disk size: 16 Go, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001FBFDE
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16007561216 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
```

pbatard, 2019-04-06:
Evidemment il fallait que je me vautre sur le ‘sélectionner’ qui est le plus proéminent…
Merci de l’avoir signalé !

closed 2019-04-06 by pbatard

Atelier-Made-in-iKi, 2019-04-06:
Et on peut pas le rater ;-)
Merci pour ce retour super rapide
Jean-Philippe

commit 1ca4c7ac (2019-04-06): [loc] fix typos in French translation

* Closes #1296
* Also add .github to the list of ignored AppVeyor directories