pbatard/rufus #1297: I would like to use rufus to always record the same ISO preset, this is possible?

Jotamel0, 2019-04-07:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

Jotamel0, 2019-04-07:
I would like to use rufus to always record the same ISO preset, this is possible

Jotamel0, 2019-04-07:
I've seen about command lines, but maybe there's another way

Jotamel0, 2019-04-07:
using rufus.ini maybe..

pbatard, 2019-04-07:
> I've seen about command lines, but maybe there's another way

No, there's only the commandline. I have no plans to enable any other way to do so at the moment.

pbatard, 2019-04-07:
As I believe I answered your issue, I will close it.

closed 2019-04-07 by pbatard