pbatard/rufus #1299: UEFI:NTFS "Could not locate '\efi\boot\bootx64.efi': [14] Not Found"

RufusUser, 2019-04-10:
This is my log if you need it

1 device found
Disk type: Removable, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 30448, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x001A3ECF
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 233.2 GB (250448163840 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 489158618, Boot: No
Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Using autorun.inf label for drive D: 'Little Big Usb'
1 device found
Disk type: Removable, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 30448, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x001A3ECF
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 233.2 GB (250448163840 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 489158618, Boot: No

At least i think that's my log

JonnyTech, 2019-04-10:
The log does not state your reported message of `UEFI:NTFS "Could not locate '\efi\boot\bootx64.efi': [14] Not Found"`. Can you re-run then press the log button in Rufus (next to the start button) to grab a log with the fail.

RufusUser, 2019-04-10:
[image]

JonnyTech, 2019-04-10:
Thanks but please include the full log of the disk creation from Rufus as that will show the files that were copied to the drive during its creation and any errors encountered.

RufusUser, 2019-04-10:
How?

pbatard, 2019-04-10:
@RufusUser, please don't ignore the check list. The log you provided __IS__ truncated,  which is explicitly mentioned you should not do in the check list.

You need to recreate your drive as you did before, and make sure you copy paste the __whole__ thing, starting with the line that has the Rufus version. Do not remove anything from your log and make sure you copy/paste the log after you re-created the drive in full.

It's only if you do that that we'll be able to help you.

pbatard, 2019-04-10:
Also, since the issue is most likely that the ISO you are trying to use does not have a `\efi\boot\bootx64.efi` file, are you absolutely sure that the ISO you are trying to boot is compatible with x64 UEFI systems? For instance, you cannot use a 32-bit Windows ISO and expect it to work with an x64 UEFI system (which is what you have).

RufusUser, 2019-04-10:
Ok i'll get a new iso and recreate then send you the log good?

pbatard, 2019-04-10:
You don't need to get a new ISO, unless you have found that the ISO you used was the wrong one (but in that case, and if you don't get the error from the title, you should just close the issue). You just need to re-do exactly what you did to create the USB that presented the problem, and provide the full log from Rufus then.

RufusUser, 2019-04-10:
Ok.  But my ISO is a x32 don't i need a x64 ISO?

pbatard, 2019-04-10:
Then that __IS__ your issue. You do need an x64 one, and if you use an x64 one then you will find that the error goes away.

I will therefore consider that this problem is resolved: Just get an x64 ISO. Your UEFI system is x64, you should not use a 32-bit ISO with it.

closed 2019-04-10 by pbatard

RufusUser, 2019-04-10:
What if it doesn't work do i still copy the log and send it to you?

pbatard, 2019-04-10:
If it doesn't work, then please open a new issue and make sure to follow the checklist (__DON'T__ remove it!!!), since the error you will get then will be different from the one above, and it will therefore be off-topic.

RufusUser, 2019-04-10:
Ok just to make sure i am tapping start this is my configuration

[image]