pbatard/rufus #1302: Why I can't flash a bootable USB on FAT32 with Win7 ISO?

Barabba11, 2019-04-14:
﻿Hi, I have an old laptop which supports boot from USB only if it is formatted on FAT32, the ISO is 2,25GB and I can't flash it with Rufus, only NTFS, why? It's less then 4GB! no .wim files that big!
If I can do it using software "ISO to USB" (ISOTOUSB.COM) why I can't do it with Rufus?
I tested latest and 2.18.
Thank you

pbatard, 2019-04-14:
> Hi, I have an old laptop which supports boot from USB only if it is formatted on FAT32

Is your laptop BIOS or UEFI based? You __MUST__ absolutely know this beforehand, and then use the proper settings in Rufus.

The only ever possibility for your laptop not to boot an USB Flash Drive unless it's formatted with FAT32 is if it's UEFI based and you are booting in pure UEFI mode (rather than UEFI-CSM).

Otherwise, if you have a BIOS based computer or you have UEFI-CSM enabled, I can absolutely guarantee that using NTFS is not an issue.

> only NTFS, why?

Because you selected BIOS for _Target Type_ in Rufus instead of UEFI, that's why.
Again, you __MUST__ select the _Target Type_ that matches the type of computer you want to boot. Selecting "BIOS" when you are trying to boot in UEFI mode will produce exactly the kind of issue you think you are seeing.

You may also want to read the following entry from the FAQ:
https://github.com/pbatard/rufus/wiki/FAQ#Help_I_created_a_Windows_installation_drive_but_it_doesnt_boot

Barabba11, 2019-04-14:
Thank you for kind and fast answer. The laptop is quite old, I don't believe bit can support uefi but I will try to boot with it. Do you have any iso as example for boot with is not a Windows iso of more then 2gb? Flash it only for a test takes time. Thanks.
Do you know the program I used instead? It's quite old too, I think it uses bios but with fat32. It worked

Barabba11, 2019-04-14:
I checked and it works with BOOTMGR and NTLDR boot mode, so it's bios based, not uefi.
Maybe you can introduce too fat32 in Rufus? Thanks

pbatard, 2019-04-15:
If it's BIOS based then NTFS will work. Unlike UEFI, BIOS is file system agnostic, which means that it's the USB drive that provides the file system driver (or it's early boot equivalent), and it would really take something extraordinary to prevent NTFS from working where FAT32 works, considering that the early driver comes from Microsoft and has been refined over 2 decades.

And indeed, now you indicate that NTFS __did__ boot, but that you got another error.

So can you please provide a few more details about your _"error like unit is unformatted"_? You will understand that vague error messages are not going to get people to go out of their way to help you out (especially as you chose to ignore the checklist and failed to provide a log, which may provide some clues).

For instance, are you aware that you may get messages from the Windows installer that may be similar to the one you got if you have multiple disks plugged in? When installing Windows, you should really make sure that you only have the target drive plugged in, and, if prompted, tell Windows that it should erase the disk. Have you tried that?

pbatard, 2019-04-16:
No update &rarr; Closed.

closed 2019-04-16 by pbatard

Barabba11, 2019-04-16:
Sorry to haven't replied on time, much work.
The error I got is the classic
"bad disk, replace it
Press CTRL+ALT+CANC"
That means it doesn't read the NTFS. I repeat it, same ISO file, I used  "ISO to USB" with FAT32, which is old software and doesnt support UEFI, it worked.
So where is the issue here?
If you can't fix I'll keep using  "ISO to USB", sorry for that (

pbatard, 2019-04-16:
> The error I got is the classic
> "bad disk, replace it
> Press CTRL+ALT+CANC"

Please be more precise. "Bad disk, replace it" is an approximation of the error message you got. I want the exact error.

Also when exactly does that error occur? At what stage of the boot or of the installation process?

Also, and this is probably the most important, are you using an official Microsoft ISO?
What is the SHA-1 of the ISO you use (you can compute the SHA-1 by clicking the `(✓)` button.

I'd also like you to copy paste the full log from Rufus of how you created your USB Flash drive with NTFS.

At this stage, I'm pretty sure we're comparing apples and oranges here, because there's no telling what bootloader ISOTOUSB installs to make the USB boot (and I am not interested in figuring that out), and since it doesn't seem to offer an NTFS option, we also cannot tell whether an NTFS formatted ISOTOUSB drive would work (which would demonstrate that NTFS or FAT32 has nothing to do with your issue).

Until I see strong evidence of the contrary, I see no indication that enabling FAT32 in Rufus would help your situation in any way. Most likely, you'll get the same error, and we will both have wasted our time on this whole thing, which I'd rather avoid...

Barabba11, 2019-04-17:
I understand you, and I don't want to take time from you.. 
At this moment I returned the laptop, repaired, so I can't make you that screeenshot, but I can tell you exactly how it worked.
Think you connect an HDD unformatted to a BIOS computer and you have a DVD with bootable disk inserted, now on BIOS you have the DVD boot priority and it boots, but you put the HDD as priority, here you have that message! SO it "boots" on the HDD but couldn't find nothing there.
In my case it was booting from the HDD (corrupted) and i choosed on bios the USB, so I?m sure it was there the problem. Probably the sentence was "Not a system disk, replace it and reboot pressing CTRL+ALT+CANC", I'm sure 90%, but can't guarantee. I understand this is an important question.
Now I downloaded this ISO, which is original Windows:
https://windowspro.ru/engine/download.php?id=3
and this old version of ISO to USB (2013) https://drive.google.com/open?id=1h1WmApF1zZ_4_E93EmwnQfXXXdF0JXlG
The ISO worked perfectly with it! I tried to give you all material I used..

Barabba11, 2019-04-17:
Here are some infos about the file structure of the USB with ISO
https://drive.google.com/open?id=19A8PoW9FNiR6VY_U8vLt-9OJrK5uBm5_
https://drive.google.com/open?id=1mYaoggc2BTNsEP_YrSEA6CpqIK50HsoR

pbatard, 2019-04-17:
> Now I downloaded this ISO, which is original Windows:
https://windowspro.ru/engine/download.php?id=3

Microsoft does not provide their ISOs as torrents, so I have my doubts that it's as original as you say. I'm downloading it as we speak, and will check its SHA-1 against known SHA-1 for official Russian versions (if it is a genuine Microsoft ISO, then its SHA-1 will be found [here](https://msdn.rg-adguard.net/public.php)).

Also, if you are using UEFI (as is your case) and you want to use FAT32 (and your ISO does not contain a >4GB file), then you don't need to use Rufus. Just format your target drive as FAT32 and extract all the ISO files onto that drive (using 7-zip or whatever you want). __That's it.__ So can you please try that and see what happens?

If you get the same issue, then it will demonstrate that this whole thing has nothing to do with Rufus.

pbatard, 2019-04-17:
Actually, since I appear to be conflating two issues (that's the trouble when you're trying to help multiple people at the same time), you're not the guy who's booting in UEFI mode but in BIOS mode, which is confirmed with the message you get, which is a BIOS only message, so you can ignore the test I asked you to do, since it'd only work for UEFI.

I'll still want to see the log of how exactly you created your drive for NTFS in Rufus before I spend any more time on this issue though...

pbatard, 2019-04-17:
Interesting - Your ISO (SHA-1: `b77b1e90b1419bdb7172262f5830088ae1e4e821`) is actually [an official one](https://msdn.rg-adguard.net/public.php?seach=b77b1e90b1419bdb7172262f5830088ae1e4e821)...

Barabba11, 2019-04-17:
I can redo the flash with Rufus, using the same parameters and send you the log.. but I don't see really big sense on it, you know, you have the image too. I used MBR, BIOS, and it forced me use NTFS.. nothing else I touched

Barabba11, 2019-04-17:
maybe for you is more interesting to see the structure of the actual flash which is working.. just tell me the sectors I need to dump and I?ll send you. I don't have big experience on find out partition table, etc.

pbatard, 2019-04-17:
> but I don't see really big sense on it

I do. The log provides some data about your environment which I need to see. Besides, you are supposed to __ALWAYS__ provide a log when creating an issue, which you chose not to so. Please don't ignore the check-list. I've been bitten way too many times by stuff that would have been __OBVIOUS__ had the user provided the log when requested, but who decided not to do so. I want to see your log, period.

pbatard, 2019-04-17:
I'm not interested in dumps. I'm not interested in what ISOTOUSB does. I have exactly zero interest in trying to replicate what other tools are doing, especially if (until proven otherwise), it may only help a single user. My time is NOT infinite.

pbatard, 2019-04-17:
For what is worth, I have just tested a bootable USB created by Rufus 3.5 using MBR + NTFS and your ISO, and found no issue. The flash drive booted into the Windows installer without problem every time. This was tested on real BIOS based hardware with 2 different flash drives, and also I created one of the drives from Windows 10 and another from Windows 7.

Therefore, I am pretty sure that whatever problem you are seeing is environmental and has little to do with Rufus. All the more reasons to want to see your log.

Barabba11, 2019-04-18:
Hi, this is my log
[log-rufus.txt](https://github.com/pbatard/rufus/files/3094689/log-rufus.txt)
Today I tested to another old laptop and it wasn't working, after "press a key to boot from disk" i had black screen. I've tried then to flash with ISO to USB and I got a curson on top left blinking for some seconds, then black screen. Anyway this is not an important test.
What I wanted to say you, very important.. and I forgot, when Rufus flashed usb was put in that laptop I had a "press a key to boot from CD", and after the error.
I've no clue where that sentence is written, in in NTFS partition or somewhere else (I think so)

pbatard, 2019-04-18:
Thanks for the log.

The only "odd" things I see in there are the following:

```
Extracting: E:\autorun.inf (43 bytes)
  Unable to create file: [0x00000005] Accesso negato.
  NOTE: This is usually caused by a poorly designed security solution. See https://goo.gl/QTobxX.
  This file will be skipped for now, but you should really look into using a *SMARTER* antivirus solution.
```

```
Unable to create E:autorun.inf
NOTE: This may be caused by a poorly designed security solution. See https://goo.gl/QTobxX.
```

This leaves me to think that you do have software on the machine you used Rufus on that actively prevents files from being created or written, and I am of course wondering if that software might also alter bootloaders, such as the ones that Windows installs on the NTFS (or FAT32) partitions...

At any rate, you should work on eliminating that issue, as it may very well be linked to the one you face later.

> I had a "press a key to boot from CD",

Yes, that means that the bootloader in the MBR executed, and after that the Microsoft bootloaders take over.

At this stage, I'm pretty sure your issue is either an incompatibility between the default Microsoft bootloaders and your machine, or the security (or other) software you use that altered some important elements such as boot records.

Now, I have no idea what ISO to USB does to bypass these issues, but as I mentioned, I have no interest to find that out if the issue boils down to poorly designed security solution, as I genuinely don't have time to add workarounds for that in Rufus.

I'm afraid that you will have to figure out what exactly causes these issues in your environment, because, as I also pointed out, I saw none of these when testing (and other people also don't report a similar problem when using NTFS) which pretty much indicates that this is not something that should be addressed in Rufus.

Barabba11, 2019-04-18:
that problem is from Avira Antivirus and it regards only the file autorun.inf only. I've the same problem with ISO to USB, don't worry nothing is affected, I used the same computer for both solutions.
If that message is on MBR it means before NTFS and so I guess I?m right thinking that that very old bios read only files from FAT32.
I've not the skills to tell you exactly what wasn't wrong with Rufus, in my opinion the NTFS only support, I told you and prove the fact that this ISO can be flashed in FAT32 and it boots regulary, there are not .wim files inside >4GB and I don't see reasons to avoid FAT32 from Rufus. I kindly asked you to introduce again the FAT32 on rufus if the iso doesn't contain files >4GB and I think this issue will be overcomed in Rufus too..
but if you don't want to do that I can't force you, I'm just sorry for that, if the case I'll use ISO to USB

pbatard, 2019-04-18:
> I?m right thinking that that very old bios read only files from FAT32.

No. As I explained, BIOS is file system agnostic. The file system handling is provided by the bootloader, not the BIOS, which means that there's no reason for FAT32 to work where NTFS wouldn't.

> I don't see reasons to avoid FAT32 from Rufus

Because:
* There does not exist any evidence that FAT32 will ever work where NTFS doesn't when using BIOS (see what I explained above). And if you understand how BIOS works, and how Microsoft does both the handling of FAT32 and NTFS in their Windows BIOS installer, it makes a lot of logical sense that there isn't.
* It avoids people who might end up with a >4GB file and are using BIOS wonder why they suddenly aren't able to select FAT32. I get enough of that from people using UEFI, and, again, unless you actually have evidence that, in the same conditions, FAT32 works where NTFS doesn't (which would be quite something, and which you'd have to report to Microsoft because they are the ones who perform file system access for either FAT32 or NTFS and they have been doing so for 20 years with the Windows installer), all you have right now is wishful thinking, not proof.

> FAT32 is much more largely supported then NTFS.

No. Not with Windows installers, which is what matters here. Please don't conflate out-of-scope matters with in-scope one. The only thing that matters is the Windows BIOS installer, and, IIRC, ever since Windows 2000, it has supported __boths__ FAT32 and NTFS interchangeably, which means it shouldn't matter whether you use one or the other. Again, BIOS is completely file system agnostic.

> in my opinion the NTFS only support,

You gave your opinion, and, as someone with __some__ amount of expertise on the matter I gave you mine. If you want to dispute this further, then I will ask you to provide hard technical evidence, because, in my expert opinion, using FAT32 instead of NTFS will not change a damn thing and we will both have completely wasted time on this. As a matter of fact, at this stage, unless you can come with technical evidence to prove what you assert, I will estimate that I wasted enough time on what looks like a purely environmental matter on your end (since myself and about every other people don't seem to run into the issue you have), and stop replying to this issue.

Barabba11, 2019-04-18:
ok thanks, it's clear, driver for ntfs is in bootloader, so this iso has a buggy bootloader for ntfs