pbatard/rufus #1304: Modifying RUFUS's GUI, and adding additional features.

AdamKearn, 2019-04-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Issue description
-----------------
Hello, I am building RUFUS from source with visual studio.
I want to change the GUI but I can't find where the correct file is.
Can someone please tell me what script is used for the UI so I can modify it.
^  (including the file path in soloution explorer)  ^

FYI:  I am developing my own operating system I am trying to modify RUFUS so that it will just display a small box allowing the user to burn a USB.  I would also like to incorporate an auto download thing so it will automatically download the ISO form my website and store it in the "windows temp" folder or something. 

Thanks in advance  :)
Adam.

AdamKearn, 2019-04-17:
Second thoughts, would it be easier if I made my own project (windows form application) and just includ the code that burns the image to the disk/usb?
If so could you point out what scripts I would need to port over to get it to work.

I have designed a mockup of what I am aiming to curate below.
[image]

Thanks for your help guys!  Any help is much appreciated.
Adam.

pbatard, 2019-04-17:
Hi Adam,

There is no real UI/formatting breakdown in Rufus (it's all pretty much all tied together), so if you expect to find a nice formatting + partitioning backend, that services a UI frontend, you're going to be very disappointed...

Moreover, we're not using any "modern" means of producing the UI, but using the basic win32 way of creating a UI through a `.rc` file. So, to answer your first question, the file you are looking for is [rufus.rc](https://github.com/pbatard/rufus/blob/master/src/rufus.rc).

You can open and modify that file in Visual Studio through the UI editor, and you will find it under the `Resource Files` subdirectory in the rufus sub-project.

Now, considering that you seem to be unfamiliar with `.rc` files, I would indeed encourage you not to bother with trying to modify the `.rc` (especially as you'll find that you're in for a world of hurt if you want to support localization and other stuff), and maybe use what you are more familiar with to design your own app.

In that case, you will need to familiarize yourself with what Rufus does in its main message loop, especially when the user presses the <kbd>START</kbd> button, by having a look at [this](https://github.com/pbatard/rufus/blob/0ad3f8c1d38b84ac0857226a6c3ac45c4afaa84a/src/rufus.c#L1863-L2741).

But, mostly, what you are really after figuring out is what happens in `FormatThread()` which is the core of the partitioning & formatting process, by having a look [here](https://github.com/pbatard/rufus/blob/0ad3f8c1d38b84ac0857226a6c3ac45c4afaa84a/src/format.c#L2063-L2533) (knowing that this thread will call on other functions all over the place).

For the ISO file extraction (which depends on libcdio), you may also want to have a look [here](https://github.com/pbatard/rufus/blob/0ad3f8c1d38b84ac0857226a6c3ac45c4afaa84a/src/iso.c#L682-L973).

I'm afraid that, if you were expecting some nice logical breakdown of operations, you're going to be sorely disappointed, but that's just how things are: Rufus is designed to make something that is more complex than people imagine look __deceptively__ simple, even though there are tons of things that need to be handled in the background to make that happen, and there are about 20 different cases that need to be addressed during formatting (e.g., regular FAT vs FAT32, ISO9660 vs UDF, Windows install vs Windows To Go, BIOS vs UEFI, Syslinux vs GRUB, modifying of labels that may exist in conf files), and that's not even counting all the many quirks that this or that configuration specifically needs, and that must be enacted upon for the drive to boot.

Alternatively, you may also want to look at how the endless people have managed to integrate Rufus into their custom GUI installer, as per https://github.com/endlessm/rufus, as they've been doing something similar to what you are trying to accomplish.

pbatard, 2019-04-18:
Since there has been no further comments, I will consider that the issue is closed.

closed 2019-04-18 by pbatard