pbatard/rufus #1305: How to Change/Update UI?

Kuldeep-kd, 2019-04-17:
I wanted to do some tweaks to the UI, but it seems I can`t find any Forms in the whole solution of rufus. Also there is no Wiki containing UI Editing information. Please let me know is there any guide for the same.

pbatard, 2019-04-17:
Rufus is a plain old win32 application, so there are no forms. And there is no Wiki regarding UI modifications, because I just don't have the time for that. If you want to edit the UI, you are supposed to know your way aroun `.rc` files and the best you'll get is the information I am providing in #1305.

Since you are mostly asking the same thing as #1305, I will close this issue as duplicate.

closed 2019-04-17 by pbatard

Kuldeep-kd, 2019-04-17:
I just was unable to find where to edit, rufus.rc is just what I wanted.

pbatard, 2019-04-17:
Okay then. Btw, the issue I meant to reference was #1304, not #1305 (which is this issue). I corrected that above now.