pbatard/rufus #1307: any idea about this?

zhz1237ok, 2019-04-19:
UltraISO has a feature that it can divide USB into 2 parts: 

one for Windows PE(about 300M) 

another for users to store their files(like windows ISO files)
[image]
so users would not delete Windows PE files accidently

and the users'files part can be format freely and dont need to worry about the windows PE part

dont want consider it?

pbatard, 2019-04-19:
> so users would not delete Windows PE files accidently

1. It's very difficult for users of modern Windows to delete files accidentally. System files are protected and require taking ownership before you can delete them, so, I am going to dispute that this is an actual issue. Do you have __actual__ references to people managing to delete system files.

2. This would only apply to Windows To Go drives, since people don't need to store data when creating a Windows installation drive.

3. You need to understand that the purpose of an application is __NOT__ to emulate everything other (somewhat) similar applications might do. If that were the case, things would be pretty bland and you'd never see any improvements - just a rehash of the same boring old features. Every developer or group of developer have their own idea of what features will benefit users best. What you suggest is not one these features.

> most people in china uses Windows PE(inside it contains lots of tools like windowsNT、ghost、CGI tools、deskgnius) to install windows

Then it looks to me like most people in China are not using the __official__ means of installing Windows, and are instead using "unofficial" images. This is not something want to support, because there's no telling how these "unofficial" images have been modified.

If I wanted to provide people with malware, the first thing I'd do would be to create an "unofficial" Windows installation image, with some attractive features (e.g. paying software that people might be able to get for free) but in which I would also inject my malware. This way, when people install my image, I'd have complete control over their Windows installation, as opposed to trying to find a vector to install malware on a Windows platform that was installed from the official media (which, by the way, Microsoft makes easy to obtain).

So, as you may understand, I have no intention to go out of my way to support installation images that are very dubious of origin, and might contain illegitimate software, if not worse.

> dont want consider it?

No.

You are trying to make me address an issue that, all things considered (rather than your limited view of it) would only benefit a __very small__ number of people, and what's more, people who are using __unofficial__ Windows installation media.

If you want this, please use UltraISO.

closed 2019-04-19 by pbatard

zhz1237ok, 2019-04-19:
~~windows PE(actually is a win10 "lite" it cant be used like win10) is a useful tool and yes, it's unofficial and some of them may contains malware

but, it's not the system we to use , it is created to intall the WIN10

because the official means to install WIN10 is not good enough,thats why people figered out a better one, and **windows PE only intalled in USB to help us to intall win10 in desk**

so it doesnt matter if it is official or not , the tools inside it can be official(like windowsNT)
**and we install official image of WIN10**

and it can do lots of things official cant(for example i can redivide my desk umlimitedly after i installed Win10[the operation is not running in win10, it running in USB，you can even call it bios plus])~~

i'm not good at english, if you have time, please watch this, this is how we install official WIN10 image normally:
https://www.youtube.com/watch?v=5CxazAromNA

system files can be deleted because it is "PE", PE files exposed to any users and deletable
and you cant arrest IDM's creator because one of IDM's user downloaded illegal things

pbatard, 2019-04-19:
> because the official means to install WIN10 is not good enough,thats why people figered out a better one

Then please contact Microsoft if you think that the official means of installing Windows 10 is "not good enough". Or create your own application to better handle your __unofficial__ images.

I can and only ever will support __OFFICIAL__ Microsoft ISOs. If you want to use an unofficial ISO, you are 100% on your own.

zhz1237ok, 2019-04-19:
> > because the official means to install WIN10 is not good enough,thats why people figered out a better one
> 
> Then please contact Microsoft if you think that the official means of installing Windows 10 is "not good enough" because that's how we derive the installation process used by Rufus (a single partition with all the installation files). Or create your own application to better handle your **unofficial** images.
> 
> I can and only ever will support **OFFICIAL** Microsoft ISOs. If you want to use an unofficial ISO, you are 100% on your own.

do you have used rufus to install Ubuntun? before your intalling, you just got into a ready-made Ubuntun system, right? thats the same thing PE do, a win10 interface to better intall WIN10

pbatard, 2019-04-19:
You do understand that the installation process is mostly a __one time usage__. In other words, people will boot the USB once, install Windows, and be done with it. So the idea that files that are used for a one-time installation must be protected against user deletion is a bit far fetched.

Now, if you are reusing the same installation drive to install Windows over and over again then, first of all, I do hope you have legitimate licenses to do all these installations, and second, you are doing SysAdmin operations, and, contrary to what many people seem to believe, Rufus is not a SysAdmin tool. Instead it is geared towards people performing a one time installation of their OS, so if you need more than this, I will kindly ask that you use something else than Rufus to accomplish your task.

Also, if you want to run Windows from USB (as opposed to using it for a one-time installation, as per Rufus' design), then you can create a Windows To Go drive (which Rufus supports) and where system files are protected from deletion by the OS, just like on a regular Windows installation.

zhz1237ok, 2019-04-19:
> You do understand that the installation process is mostly a **one time usage**. In other words, people will boot the USB once, install Windows, and be done with it. So the idea that files that are used for a one-time installation must be protected against user deletion is a bit far fetched.
> 
> Now, if you are reusing the same installation drive to install Windows over and over again then, first of all, I do hope you have legitimate licenses to do all these installations, and second, you are doing SysAdmin operations, and, contrary to what many people seem to believe, Rufus is not a SysAdmin tool. Instead it is geared towards people performing a one time installation of their OS, so if you need more than this, I will kindly ask that you use something else than Rufus to accomplish your task.
> 
> Also, if you want to run Windows from USB (as opposed to using it for a one-time installation, as per Rufus' design), then you can create a Windows To Go drive (which Rufus supports) and where system files are protected from deletion by the OS, just like on a regular Windows installation.
> 
> > Microsoft thought it is useless
> 
> I'm afraid that they're not alone there. So do we.

PE is more than a installation process, it contains many, yes, admin tools to help people even after intalled the system, for example:  KB4490481 after intalled this patch,  i cant enter my WIN10, and I cant move my important data, 
and i just boot the USB, go into PE, there, i can edit my desk, i can move my data out and even use [Dism++](https://www.chuyu.me/zh-Hans/index.html) tools to remove this patch(after failed with WINRE) to enter my WIN10

pbatard, 2019-04-19:
Again, it looks to me like what you really want is Windows running from USB, a.k.a. [Windows To Go](https://en.wikipedia.org/wiki/Windows_To_Go). Please look into using Windows To Go. Unless you select Windows To Go (which is Microsoft's __official__ solution to what you seek), Rufus will __NOT__ create installation media from Windows installation ISOs that are aimed at anything else but to install Windows. And again, if you are unhappy with that, I will kindly ask that you use a different utility (if UltraISO does what you want, why not use it?) or fork your own competing version of Rufus.

zhz1237ok, 2019-04-19:
> Again, it looks to me like what you really want is Windows running from USB, a.k.a. [Windows To Go](https://en.wikipedia.org/wiki/Windows_To_Go). Please look into using Windows To Go. Unless you select Windows To Go (which is Microsoft's **official** solution to what you seek), Rufus will **NOT** create installation media from Windows installation ISOs that are aimed at anything else but to install Windows in a one time fashion. And again, if you are unhappy with that, I will kindly ask that you use a different utility (if UltraISO does what you want, why not use it?) or fork your own competing version of Rufus.

acctually its not a windows to go, because PE cant be used like normal WIN10 , it only 300M, and cant connect to internet etc.

what i want to say is that, if it possible to divide USB into 2 parts before install things like PE (because i used rufus to install PE in to my USB, PE files like a mess, and i'm not only use my USB to install PE, but also store software files, so, is it possible to pre-divide USB and rufus only install ISO into one part, and i can format another part with exFAT)

pbatard, 2019-04-19:
https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want

zhz1237ok, 2019-04-19:
> https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want

I know that, but, its really a good idea to pre-divide, i just put this idea here, i dont request you guys to do

have a nice day, and thanks to tolerate my bad english

pbatard, 2019-04-19:
> but, its really a good idea to pre-divide

I disagree. It adds complexity. I will confuse users. It will be time consumming to maintain, test and support. And, most importantly, as opposed to what you appear to believe, our view is that it will only help a __very small minority of users__ (who of course, may see it otherwise, but I would surmise that most Chinese people who install Windows don't go around looking for means to run `dism++` while doing so, and actually go with the __official, unmodified__ content that was put forth by Microsoft (since it's Microsoft's job to put content that does work, and fix it if that ever isn't the case &mdash; if that is your issue, then, again, it's something you need to take to Microsoft, as they do welcome feedback!).

> thanks to tolerate my bad english

It may surprise non-native English speaker (which I am too), but most English speakers don't care if your English isn't perfect, especially on technical issue trackers. As long as you can make yourself understood, we couldn't care less if your grammar or spelling is correct. So please don't worry about your English.

zhz1237ok, 2019-04-19:
p.s. by using PE，we only need to download new ISO to replace old one(16G USB can contains 3 different win10 version ISOs) to install WIN10 rather than format USB everytimes

and in old times, this pre-divide function acctually called "隐藏分区(hidden partition/protected partition)", its invisible to user, so you can only see a empty user part of USB(which we can store official ISO and other things), but recently win10 can show the hidden part of USB, but still, it is helpfu to isolate the PE files

pbatard, 2019-04-19:
>  by using PE，we only need to download new ISO to replace old one

But you'll need to copy it to the USB, which takes about the same time as Rufus will take to recreate the USB from that ISO. All in all, if you are using USB 3.0 media, you're talking about sparing 1 to 2 mins... whereas you probably spent 30 mins or more waiting for your ISO to download. You'll need to find a more compelling argument, because that's not one.

Also, if your plan is to somehow add multiboot, please be informed that Rufus will also never be designed to help with multiboot, so there again, you're on your own: https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support.

> and in old times

Pre Windows 1703 I think, as I believe that finally added multiple partition mounting from multiple media. Yet another reason why adding what you suggest would be more complex than you think because, since Windows 7 or 8 cannot see two partitions from removable USB media at once (unless that media has the FIXED attribute, which is very rare), we'd have to mount/unmount each partition or indeed play with attributes, which is a major pain in the ass. All this to help a __very small amount of people__ who __erroneously__ think having access the PE is something everybody else will want, or even need.

Again, I can only advise you to be very cautious about imaging that what __you__ see as extremely useful or desirable is something that a vast majority of people will see that way as well. It's very easy to think that your own view will apply to other people, when it really won't.