pbatard/rufus #1308: Replace Manjaro default install bootx64.efi with ubuntu's bootx64.efi

mlyxshi, 2019-04-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Under GPT+UEFI

Case 1:
From FAQ, I know that the default Manjaro installation bootx64.efi(Manjaro:/efi/boot/bootx64.efi) file does not support Fat32 module. So after I use Rufus ISO mode to create bootable usb, I add the default Ubuntu installation bootx64.efi (Ubuntu:/efi/boot/bootx64.efi) (It support many FileSystem including Fat32) to the directory  Manjaro:/efi/boot.  I choose boot from the ubuntu efi file.  In this way, I succeed to boot the Manjaro installation procedure(If boot from default Manjaro efi, grub prints unknow filesystem)

> This can happen if the people who created the ISO chose not to add FAT32 module support in their GRUB EFI bootloader (for instance, the Manjaro distro maintainers are currently doing just that, which is VERY BAD PRACTICE), because they are relying a bit too much on the kludge that is ISOHybrid. Of course, since Rufus tries to use as much of the original ISO as it can, if the distro maintainers decide to use a crippled bootloader and completely ignore the de-facto way of booting an UEFI ISO, which is to copy all its content, as-is, onto a FAT32 drive, then you will encounter issues...

Case 2:
If i use Rufus, everything is fine. However, if I use WinRAR to extract the Manjaro ISO file to Fat32 usb, add the Ubuntu efi file, and boot from Ubuntu's efi. From the grub picture, Maybe some mistake of load the grub.cfg? I don't know the reason.

In conclusion
Ubuntu:   
- Simply extract ISO to Fat32                                 Yes
- Rufus   iso/dd                                                        Yes

Windows
- Simply extract ISO to Fat32                                 Yes
- Rufus    iso/dd                                                       Yes

Manjaro 
- Simply extract ISO to Fat32                                 No
- Simply extract ISO to Fat32   +Ubuntu efi          No
- Rufus     iso                                                            No
- Rufus     iso + Ubuntu efi                                      Yes
- Rufus     dd                                                            Yes

[image]

[image]

I want to know the difference between extract ISO with WinRAR  and Rufus iso mode.(the usb disk seems that it has same files under Windows)

I open Rufus log, it says that " find ios9660 image and extract with Joliet extension". Is the extension the difference? What's the reason that "Simply extract ISO to Fat32   +Ubuntu efi" fail, however "Rufus  iso mode+Ubuntu efi" succeed?

pbatard, 2019-04-25:
Your problem is most likely that you didn't set your FAT32 label to be same one as the ISO.

For instance, if you use `manjaro-kde-17.1.12-stable-x86_64.iso` then you **must** set your FAT32 label to `MJR17112`, otherwise the init script will not know how to find the installation media.

You will also see the label explicitly mentioned in `/boot/grub/kernels.cfg` on the ISO, at line 37:
```
    for kopt in x86_64 $kopts  misobasedir=manjaro misolabel=MJR17112 quiet systemd.show_status=1; do
```

Basically, it means the parameter `misolabel=MJR17112` will be passed to the Linux kernel, which can then be used by the init script to detect the installation media.

Obviously, when you create the FAT32 drive in Rufus, it detects the label from the ISO and uses that for FAT32 (it may also alter the GRUB config files to alter the name if needed, but that's not needed for Manjaro).

Now, the puzzling part is that if this is your issue, then you should still see the GUI kernel selection as with your Rufus + Ubuntu GRUB test and get an error further down the line about trying to locate `disk-by-label/MJR17112` which should point to the issue above being the problem.

So, if you don't see that, then I can see 2 other possibilities for your error:
* WinRAR doesn't properly extract ISO files
* You didn't partition or format your drive properly when you created your FAT32 media.

So, let me provide you with an example that I have validated using `manjaro-kde-17.1.12-stable-x86_64.iso`:

1. Format the drive as GPT/FAT32 and make sure it is assigned the label: `MJR17112`. Note that I used Rufus to do that, but it shouldn't matter. Especially, I selected _Non bootable_ for _Boot selection_ and did not have the Manjaro ISO selected.
2. Use [7-zip](https://www.7-zip.org/) (__NOT__ WinRAR) to extract the ISO onto the FAT32 drive. As opposed to WinRAR, 7-zip is Open Source, and, from having actually looked at its source, I can definitely vouch that it's not going to screw things up when extracting content. I cannot say the same thing about WinRAR.
3. Extract `BOOTx64.EFI` and `grubx64.efi` from the latest Ubuntu (`ubuntu-19.04-desktop-amd64.iso`) into your `/efi/boot` directory.

If you do just that, you'll find that Manjaro boot. Thus, at this stage, you'll have to figure out which of WinRAR, partitioning/formatting process or your bootloader copying is the issue.

Oh, and, since Manjaro thought it was a [great idea to completely delete their previous bug tracker without notice](https://forum.manjaro.org/t/so-what-the-heck-happened-to-the-bug-tracker/83789), where I logged a request for them to add the `fat` module back into their GRUB bootloader, I've pretty much described [the same successful test](https://gitlab.manjaro.org/tools/development-tools/manjaro-tools/issues/306) in the new issue I re-created. Unfortunately, it's become pretty clear that the Manjaro maintainers [don't like to hear that their distro is trailing in user-friendliness when it comes to creating bootable media, even when you point them to how easily it could be resolved](https://forum.manjaro.org/t/fix-manjaros-grub/83876)...

mlyxshi, 2019-04-26:
I find two reasons:
- I don't change the disk volume label name for Manjaro and Archlinux
- WinRAR does not extract accurately(7-zip extracts very well)
This surprises me a lot because I used WinRAR to simply extract (Windows10 and Ubuntu)ISO to FAT32 disk, There is no mistake and installation boots successfully. Therefore, I thought WinRAR has a good ability to exact ISO files.

However, Using WinRAR to extract Archlinux and Manjaro will lead to mistakes. 

Thank you very much, and I learn a lot about the principle behind Rufus and Grub.

pbatard, 2019-04-26:
Happy to hear you managed to solve your issues.

With regards to WinRAR vs 7-zip, and considering that you are in the issue tracker for an Open Source software, I will say this: Never use closed source software (even if it's free) when you can use Open Source software to accomplish the exact same thing... 😉

Oh, and feel free to ask the Manjaro maintainers about fixing their GRUB bootloader. As their bug tracker indicates, they won't listen to me, but maybe they'll listen to you.

I'll close this issue now.

closed 2019-04-26 by pbatard