pbatard/rufus #1309: is there a rufus for linux

chemel9, 2019-04-26:
hi i have a doubt ...
if rufus is open source , why is only available for windows? why linux no?

you can make a flatpak or snap for rufus, is very easy.
in this way, all users can use rufus in linux

gacelperfinian, 2019-04-26:
The author has already explained [here](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS) why there will be no support on other operating systems. 
TLDR: the program relies on Windows-specific functions and capabilities and there are already tools on other operating systems to write a bootable USB Disk.

(@pbatard: please close this issue)

pbatard, 2019-04-26:
Indeed this questions is asked on a very regular basis, and if OP had followed the checklist they would have found it answered in the FAQ. Closing.

closed 2019-04-26 by pbatard