pbatard/rufus #1311: Error finding / downloading syslinux files

jharnal, 2019-05-04:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

jharnal, 2019-05-04:
Rufus x86 v3.5.1497 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because the remote download script can not be accessed.
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC5A27575
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8016302080 bytes)
  Start Sector: 2168, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20181203 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 19.04 amd64'
  Size: 2.0 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-19.04-desktop-amd64.iso (2.0 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20181203/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - cancelling
Checking for Rufus updates...
Checking release channel...
Unable to send request: This system's SSL library is too old to be able to access this website.
Updates: Unable to access version data
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
Found USB 2.0 device 'TOSHIBA MK5055GSX USB Device' (152D:2329)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC5A27575
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8016302080 bytes)
  Start Sector: 2168, Boot: Yes
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC5A27575
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8016302080 bytes)
  Start Sector: 2168, Boot: Yes
Checking for Rufus updates...
Checking release channel...
Unable to send request: This system's SSL library is too old to be able to access this website.
Updates: Unable to access version data
Checking for Rufus updates...
Checking release channel...
Unable to send request: This system's SSL library is too old to be able to access this website.
Updates: Unable to access version data

pbatard, 2019-05-04:
>   - [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

I don't believe you did. Please search for the error message you got from the log:

```
This system's SSL library is too old to be able to access this website.
```

You'll find duplicate issues such as #1232 that answer your question.

Since this is a duplicate from an existing issue, I will close this one.

closed 2019-05-04 by pbatard