pbatard/rufus #1314: Cannot disable secure boot in BIOS, MBR is the only way?

isometriq, 2019-05-12:
Issue description
-----------------
If I cant disable secure boot in the bios (for booting Windows 10 installer with GPT and UEFI), do I need to use MBR ?

pbatard, 2019-05-12:
> If I cant disable secure boot in the bios

Until proven otherwise, you always can.

Unless you have stumbled on a very mythical creature, there isn't a single x86 board that I know of where you cannot disable Secure Boot.

The reason for that is that, a few years ago, Microsoft did try to force some manufacturers to produce ARM-based hardware where Secure Boot could not be disabled, but the backlash from consumer about this was such that it ensured that x86 motherboard manufacturers would think twice about attempting the same thing.

I am therefore not aware of a single x86 based computer where Secure Boot cannot be disabled. Possibly, it might not be as obvious to do as checking an option that says `Secure Boot: Disabled` but I'm pretty confident that if you look hard enough, you will find how to disable it.

Now, if you do believe that you have found a motherboard where Secure Boot cannot be disabled, then myself (and probably a lot of other people) will be __exceedingly__ interested to get the full make and model, because, assuming that you're not trying to install Windows on one of these old ARM based machines where Secure Boot is locked (and I say ARM, not ARM64) that would be big news.

So, can you tell us what machine you are using, that you think prevents Secure Boot from being disabled?

Yewenlamb, 2019-05-13:
The option in a lot of modern hardware I am seeing is only able to be disabled after a password has been set in the BIOS, it will then allow you to edit the Secure Boot / Legacy / CSM settings.

pbatard, 2019-05-13:
I haven't really seen that myself (can you please provide actual examples of hardware where you have seen that?) as I don't recall ever seeing a motherboard where the options provided when a password is applied and the ones where password has been set would differ. You have to realize that changing the set of options that are proposed with and without BIOS password requires additional code and testing, which equates $$$ and which hardware manufacturers are unlikely to want to pay unless they have a very good reason to do so.

Even for enterprise customers, I'm not sure I see the point of adding fine-grained level of password protection (leaving some options accessible while protecting others) as, in all the corporate environments I've been confronted to, it was an all or nothing affair: either you trust the employee with hardware access, in which case you let them access all firmware settings, or you don't, in which case you use a blanket password protection that doesn't grant them any access at all.

Still, I'm willing to stand corrected and I'd be interested to have examples of motherboards where this fine-grained level of password protection is applied.

At any rate, if your accessing to the Secure Boot options is password protected (be it with a blanket or fine-grained password), and you don't have that password, then __clearly__, you should NOT be entitled to bypass boot restrictions such as trying to boot non Secure Boot validated media.

Also, to answer OP's original question, since the whole point of Secure Boot is to prevent the boot of non-signed bootloaders, it should be rather obvious that (provided that the people who designed the UEFI firmware do take security seriously), the minute you have a machine where Secure Boot is enabled, then Legacy Boot __MUST__ be disabled, else Secure Boot becomes rather useless.

So your only option, __really__, if Secure Boot is enforced and you want to boot unsigned media is to disable Secure Boot, as there won't be a "Legacy loophole" that you can exploit (and I am using "exploit" deliberately here, as anything that can bypass Secure Boot when it is enforced is indeed an exploit that needs to be addressed).

isometriq, 2019-05-14:
It was an Acer laptop, probably a few years old only. I will get the model number.

In the BIOS I was able to switch between Legacy and UEFI. After selecting UEFI, the secure boot option was grayed and could not be changed. There was no BIOS password set or whatsover.

My key was using GPT and the BIOS did not want to boot it, saying it was in conflict with security policy. And Legacy would not want to boot the GPT partition. So I recreated the key on a MBR partition, then I was able to install in Legacy mode.

pbatard, 2019-05-15:
https://itsfoss.com/disable-secure-boot-in-acer/

Apparently you need to set a supervisor password for the Secure Boot toggle to become available.

As I mentioned, there is currently no x86 based UEFI hardware that I know of where Secure Boot cannot be disabled by the user, one way or another. But of course, I never said that some manufacturers might not make this disabling non-intuitive...

isometriq, 2019-05-16:
Yes, the BIOS interface looked like the one on the screenshots. I guess I didn't search well enough. I don't have this laptop with me as it's not mine, so I won't be able to confirm the procedure right away.

pbatard, 2019-05-17:
Okay. I'm going to close this issue since I believe I have demonstrated that Secure Boot can be disabled on your platform. If you still believe you have found a computer where Secure Boot cannot be disabled, feel free to add additional comments.

closed 2019-05-17 by pbatard