pbatard/rufus #1315: Can't compile 'Rufus' with Visual Studio 2017

mbluecoder, 2019-05-14:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I can not compile your source code. I'm using Visual Studio 2017. Everything builds, except the 'rufus' project. So 10 build, and one does not. I'm getting errors like:

- CLIENT_ID is undefined,
- Many syntax errors,
- One error is in this line (process.c):
   - PF_TYPE_DECL(NTAPI, NSTATUS, NtOpenProcess, (PHANDLE, ACCESS_MASK, POBJECT_ATTRIBUTES, CLIENT_ID*))
     It does not like the * in the above line.
- A lot of errors about missing characters, like ';', '*', '{', etc.

Any help is appreciated! Thanks.

Log
---
```
Can not post log as it is on another machine, and a security violation to copy it.
```

pbatard, 2019-05-14:
> Can not post log as it is on another machine, and a security violation to copy it.

Well that's gonna be tough to troubelshoot, especially as AppVeyor demonstrates that VS2017 don't have any build issues and therefore that the problem is 100% on your end.

[Here](https://ci.appveyor.com/project/pbatard/rufus/build/job/bpg6sxob5dwne0it) is a log that demonstrates that the VS2017 compilation does work (same as what I get on my machine).

As much as I'd like to help, I'm afraid my time is way too limited to help you figure out what it is exactly with your environment that prevents Rufus from compiling. The best I can say is that, it is most certainly not a general issue with the project itself.

mbluecoder, 2019-05-14:
Thank you for the fast response. I'm trying update the SDK now.

It is strange that all other projects build fine, but not the Rufus one.

pbatard, 2019-05-14:
Can you try on a vanilla install of VS2017? By the looks of it, you may be using a corporate version that might have been tweaked somehow.

Also, I'm afraid I'm not aware of anybody having reported the kind of compilation errors you report. The last VS2017 compilation issue I know of had to do with compiling on platforms with non western languages, but that was fixed in 7e21f8a60c8d508fef62a13c32e5cb1849dd0757 and the error you would have gotten then are different from the ones you have.

mbluecoder, 2019-05-14:
Yes. I had to manually download the source code from GitHub, copy then unzip the file on a secure machine (no internet access). I will try at home tonight to build there.  I'll let you know.  -Michael

mbluecoder, 2019-05-15:
OK - I tried compiling at home and it worked perfectly.  So I compared differences. At work I WAS using VS2017 v15.3.3. I upgraded it to v15.9.11, along with adding some 'individual components' during the setup, and it works fine. I wanted to put that out there for anyone having this problem.

pbatard, 2019-05-15:
Okay. I'll close this issue then.

closed 2019-05-15 by pbatard