pbatard/rufus #1316: Western Digital WDBKXH5000ABK-01 not recognized as 'Device'

penalvch, 2019-05-19:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
What is expected is when using latest Windows 10 x64 with latest Rufus (see log) with a Western Digitial My Passport WDBKXH5000ABK-01:
https://support-en.wd.com/app/products/product-detail/p/1481#WD_specifications

 is that Rufus recognizes it as a 'Device' and formats it in a 'It just works' fashion.

What happens instead is that the 'Device' drop down does not show it as an available device. Using the Rufus 'Portable' and non-'Portable', with and without 'Run as Administrator' does not change anything.

However, the Windows 10 Media Creation Tool properly recognizes the 'Device' with no problem, and one may use it to make a working Windows 10 USB bootable device (i.e. not a 'Device' hardware failure):
https://support.microsoft.com/en-us/help/15088/windows-10-create-installation-media

Log
---
```
Rufus x86 v3.5.1497 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'WD My Passport 0748 USB Device' (1058:0748)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.

```

penalvch, 2019-05-19:
WORKAROUND: Use Universal USB Installer 1.9.8.3 (not latest version but what had local copy ready to roll):
https://www.pendrivelinux.com/universal-usb-installer-easy-as-1-2-3/

pbatard, 2019-05-19:
It's an external hard drive, not a flash drive, so it is not listed in Rufus __by design__.

The log tells you exactly what you need to do to get the drive listed, as well as the FAQ (which you say you looked at as part of the check list): [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected).

Please understand that different software have different design philosophies.

Next time, please try to do your howework. Or, at the very least, read what the log you copy/pasted tells you.

closed 2019-05-19 by pbatard

penalvch, 2019-05-19:
pbatard, thanks for the quick response.

While it is understood 'by design' its not working by default, alternative USB installers seem to have figured out that problem without troubling the user with checkboxes after the fact of things not working as folks normally expect.

Hence, instead of having clunky and unnecessary checkboxes, why not have this autosensed at program launch to offload the 'user didn't click the box they should have already known about' like alternative USB programs have seemed to have already figured out and implemented?

I'll leave the design decisions to those who commit, but using Rufus at this point is simply offputting given its present design.

pbatard, 2019-05-19:
This is a very deliberate design decision. You may not agree with it, but personally, I'd much rather inconvenience a few people, by not letting them erase the data they want, than inconvenience others by allowing them to erase data they don't want to erase.

My prime priority with Rufus is to avoid data loss, always. As such, preventing people from losing data off an important backup HDD because they plugged the flash drive they wanted to format and didn't pay attention to the drive that was selected by Rufus is what I will go with, regardless of how many people try to tell me that _"Well, if those other people lose their data this way, they had it coming anyway"_ or _"but other software doesn't care! Why should you?"_.

If you disagree with this, then I will kindly ask you that you use different software than Rufus because, clearly, it is not designed for you.

penalvch, 2019-05-19:
pbatard, speaking to only the workflow for Universal USB Installer, it solves the same problem that Rufus does in a more intuitive, 'give it to me immediately when it makes sense' fashion.

For Universal USB Installer the 'Device' piece is dealt with by always presenting a checkbox for 'Show all Drives' to the immediate right of their 'Device' drop down. Hence, nobody would be confused, or has to consult documentation, because they see immediately the 'Show all Drives' checkbox and in turn deduce this is likely the reason their device isn't presenting.

However, with Rufus, one has to toggle a 'Show advanced drive properties' to reveal this option, or as you noted earlier, check documentation to then know to toggle the same.

Hence, for those folks testing Rufus quickly for viability, and who are simply just trying to accomplish a goal quickly, the conclusion of Rufus is offputting in comparison to viable alternatives remains.

[image]

pbatard, 2019-05-19:
The "advanced" options of Rufus are very much hidden __by design__, precisely because we consider that the alternative used by Universal USB Installer and other software is __too risky__ when it comes to unexpected data loss.

Again, please understand that different software have different design philosophies and that asking them to all use the same user interface choices is ridiculous.

We consider that the UUI option will lead to unwanted data loss from users, therefore we __deliberately chose__ not to go with that. Preservation of data has higher precedence than "trying to accomplish a goal quickly", since doing things "quickly" is the shortest way to having some people unexpectedly lose data.

Please stop trying to think that we designed the software the way it is designed simply because we haven't looked at what other similar software is doing. The way the UI is designed, and which elements should be displayed or hidden by default was carefully debated according to the expected ramifications of doing so for the users.

penalvch, 2019-05-19:
pbatard, thanks for the continued project insight.

Given it, the project accomplished the goal intended of reducing the probability of data loss.

However, making folks click a drop down to click the exact same checkbox presented immediately by other software doesn't reduce data loss. If someone is going to make a mistake of deleting their data by clicking a checkbox, that is going to happen no matter if they have to click the drop down or not.

Speaking only to USB Universal Installer, while this may already be 'by design', have been well debated, considered, etc. it does have a one last 'Are you sure you want to delete this dialog' to avoid said data loss.

Hence, they solved the same problem in a more intuitive, less clunky way, with no additional probability of data loss (i.e. Rufus is offputting to use 'as designed'.)

[image]

pbatard, 2019-05-19:
> However, making folks click a drop down to click the exact same checkbox presented immediately by other software doesn't reduce data loss.

So you designed software that does something similar to what Rufus does and are willing to provide user reports to back that up? Because unless you can make the claim that you are in a genuine position to venture what you say above as a fact, I'm afraid it's hard to see it as anything but a convenient __unsubstantiated__ claim...

The thing you need to understand, I already do get more reports than I'd like of _"I erased the wrong drive by mistake"_ with Rufus, even with the list HDD options hidden by design.  So, I'm sorry, I will continue to consider that I'd much rather inconvenience a few people, by not letting them erase the data they want straight away, than inconvenience others by allowing them to erase data they don't want to erase.

Oh, and Rufus also does have the "Are you sure you want to delete this drive" dialog confirmation as well.

Now, considering that I'm not planning to spend all day going in circles and about a design decision that's here __for the benefit of the majority of Rufus users__, simply because a __minority__ feels it inconveniences them, I'm going to lock this thread.

Again, if you don't like the way Rufus does things, please use Universal USB Installer or some other software and please understand that, as much chagrin as it may cause you, I am NOT designing Rufus for the minority of users who try to scream loudest so that their "vision" gets implemented, but for the majority of users.